espressif/idf-component-manager #2: [BUG] button component broken for C3 and S3 (and H2?)

chegewara, 2021-10-21:
2 issues:
1) component suppose to be for all chip models, but it is missing declaration for C3 and S3:
```
#if CONFIG_IDF_TARGET_ESP32
#define ADC_BUTTON_WIDTH       ADC_WIDTH_BIT_12
#elif CONFIG_IDF_TARGET_ESP32S2
#define ADC_BUTTON_WIDTH       ADC_WIDTH_BIT_13
#endif
```
and error:
```
../managed_components/espressif__button/button_adc.c: In function 'button_adc_init':
../managed_components/espressif__button/button_adc.c:104:27: error: 'ADC_BUTTON_WIDTH' undeclared (first use in this function); did you mean 'ADC_BUTTON_ATTEN'?
         adc1_config_width(ADC_BUTTON_WIDTH);
                           ^~~~~~~~~~~~~~~~
                           ADC_BUTTON_ATTEN
../managed_components/espressif__button/button_adc.c:104:27: note: each undeclared identifier is reported only once for each function it appears in
[2/12] Performing build step for 'bootloader'
[1/1] cd /home/chegewara/esp/master/examples/get-started/hello_world/build/bootloader/esp-idf/esptool_py && /home/chegewara/.espressif/python_env/idf4.4_py3.8_env/bin/python /home/chegewara/esp/master/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x0 /home/chegewara/esp/master/examples/get-started/hello_world/build/bootloader/bootloader.bin
Bootloader binary size 0x4ef0 bytes. 0x3110 bytes (38%) free.
[3/10] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/hello_world_main.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```
2) please update documentation, `iot_button_create` has been renamed to `button_create`
https://docs.espressif.com/projects/espressif-esp-iot-solution/en/latest/input_device/button.html

chegewara, 2021-10-21:
3) button hold event is called 2 times every time:
```
I (4701) TAG: BUTTON_LONG_PRESS_HOLD
I (4701) TAG: BUTTON_LONG_PRESS_HOLD
I (4711) TAG: BUTTON_LONG_PRESS_HOLD
I (4711) TAG: BUTTON_LONG_PRESS_HOLD
I (4721) TAG: BUTTON_LONG_PRESS_HOLD
I (4721) TAG: BUTTON_LONG_PRESS_HOLD
I (4731) TAG: BUTTON_LONG_PRESS_HOLD
I (4731) TAG: BUTTON_LONG_PRESS_HOLD
I (4741) TAG: BUTTON_LONG_PRESS_HOLD
I (4741) TAG: BUTTON_LONG_PRESS_HOLD
```

tore-espressif, 2021-11-16:
Hello @chegewara ,
I uploaded version 2.0.0 of button component from esp-iot-solution master branch: https://components.espressif.com/component/espressif/button

The bugs you mentioned should be fixed there.
We are working on a more transparent connection between our component registry and source code (typically on Github).
You should be able to see `homepage` link in https://components.espressif.com/component/espressif/button now!

chegewara, 2021-11-16:
Great, thanks.

homepage is working there and i will test component and report.

chegewara, 2021-11-16:
Seems to be ok now.

closed 2021-11-16 by chegewara