espressif/idf-component-manager #4: lack of documentation for library developers

trombik, 2022-02-09:
### The Component Manager version

N/A

### ESP-IDF Version

N/A

### python Version

N/A

### Operating System

N/A

### Browser (for https://components.espressif.com  Issues)

N/A

### Description

I could not find documentation for library developers.

## contexts

I am a maintainer of `esp-idf` components at https://github.com/esp-idf-lib. 

once @pedrominatel said espressif would help us but i've gotten none so far.

https://github.com/UncleRus/esp-idf-lib/issues/260#issuecomment-977882894

I am trying to understand how the component registry works, and evaluate how it is useful for us.

## what I need to know

* how the registry works; the registration process, quality assurance (if any), verification of ownership, acceptable use policy, name space policy, etc.
* technical details: how to publish beta, snapshot, and releases, reference manual of metadata, integrity checks (SHA sum, or cryptographic signatures)

### To Reproduce

```cpp
N/A
```

### Expected behaviour

espressif provides documentation for library developers.

### I have checked existing issues and online Documentation

- [X] I confirm I have checked existing issues and online Documentation.

VojtechBartoska, 2022-02-09:
Hello @trombik,

first of all thank you for raising an issue and your interest in using Component Manager.

**To clarify current situation, Component Manager was released recently and at this stage, it's only available for uploading internal (Espressif) components.** 

Due to this fact there is not option to upload 3rd party components, there is no documentation and so on.

To give you an update we are now working on making upload available for everyone, when it's ready and released. It will contain all necessary information.

We are looking forward for having [esp-idf-lib](https://github.com/UncleRus/esp-idf-lib) in Component Manager service as @pedrominatel mentioned. :) I will keep this issue open so it will be reminder for us to track upload of esp-idf-lib libraries if you will be interested.

I hope it's clear now. If you have more questions, don't hesitate to ask.

trombik, 2022-02-09:
@VojtechBartoska sounds great. thank you.

kumekay, 2022-10-25:
@trombik We are in preview for 3rd party contributors, if you are interested, please fill the form https://components.espressif.com/preview/

trombik, 2022-10-26:
@kumekay thanks. just signed up. i'll ask questions later.

gojimmypi, 2023-05-29:
Hi @kumekay the 3rd party preview signup link is currently 404. 

I'd like to add the wolfSSL component to the Espressif component registry. Per a discussion with @igrr and noted in https://github.com/espressif/esp-wolfssl/issues/16#issuecomment-1506605363 we'd like to move the esp-wolfssl repo and have a fresh, regularly maintained wolfSSL library.

Please let me know how best to proceed.

Thank you.

kumekay, 2023-05-29:
Hello @gojimmypi 

We have already removed the preview signup form and planning to release signup through Github in just a few days (I'm pretty sure it's going to happen this week)

To solve it even sooner, please get in touch with me at sergei.silnov@espressif.com

gojimmypi, 2023-05-29:
Hi @kumekay that's awesome, thank you! email sent.

gojimmypi, 2023-06-07:
Following up on this issue: the ESP Registry is really quite nice, the [latest docs](https://docs.espressif.com/projects/idf-component-manager/en/latest/guides/packaging_components.html) are very clear.

I have wolfSSL published at https://components.espressif.com/components/wolfssl/wolfssl

Thank you @kumekay for your help. 

Heads up to others to beware of unicode or other non-UTF8 chars in `README.md` files. 

This `iconv` command may be helpful to clean files:

```
    cat ./README_REGISTRY_PREPEND.md  ./README.md > ./NEW_README.md
    THIS_ENCODING=$(file -b --mime-encoding ./NEW_README.md)
    echo "Found encoding: $THIS_ENCODING"

    iconv  --to-code=UTF-8//ignore  --output=./README.md  "./NEW_README.md"
```

trombik, 2023-07-10:
Hi,

When sign-in by GitHub OAuth, I've got:

> User with name "trombik" already exists, probably it was created by another method. Please contact support.

Would you have a look?

trombik, 2023-07-24:
@kumekay thanks, successfully logged in. does the registry support organizations? like GitHub organization. Have a look at our use case at: https://github.com/UncleRus/esp-idf-lib/discussions/543

trombik, 2023-07-24:
it's not clear to me but i guess i need my namespace?

```
Selected profile "default" from the idf_component_manager.yml config file
Saving archive to "/usr/home/trombik/github/trombik/esp-adf-component-ai-thinker-esp32-a1s/dist/adf-ai-thinker-esp32-a1s_0.0.1.tgz"
Uploading archive /usr/home/trombik/github/trombik/esp-adf-component-ai-thinker-esp32-a1s/dist/adf-ai-thinker-esp32-a1s_0.0.1.tgz
100%|██████████| 7.29k/7.29k [00:00<00:00, 33.7kB/s]
ERROR: You are not allowed to perform this action.
```

trombik, 2023-07-25:
I need `trombik` namespace just like GitHub. Also, as a maintainer of `esp-idf-lib`, we will need another for our libraries. It would be nice to have a doc explaining how one can obtain and manage namespaces. In addition, how toke ACL works.

kumekay, 2023-07-25:
Hi @trombik 
I apologize for the confusion with your login through GitHub because we only had a few users before we introduced login through GitHub, we didn't invest enough efforts to the process of migration existing accounts.

Now you should see both `trombik` and `esp-idf-lib` namespaces in your profile. 

Documentation about namespace management as well as namespace management itself is coming a bit later this August.

Note: the old token, you used before, continues to work until you revoke it, but it won't be visible in the panel.

trombik, 2023-07-28:
@kumekay thanks a lot. I'll give it a try.

kumekay, 2024-02-27:
Just a small update regarding this issue. You can manage permissions for components and namespaces on the profile page of the registry: https://components.espressif.com/users/me

As for now, namespaces not matching GitHub username are created on request via [components@espressif.com](mailto:components@espressif.com)

kumekay, 2024-12-05:
We are continuously updating the documentation for the Component Manager here: https://docs.espressif.com/projects/idf-component-manager/en/latest/.

Additionally, we have launched a new series of blog posts with tutorials:
https://developer.espressif.com/tags/component/

While we will keep improving the documentation, I believe the main part of this task is now complete. You are welcome to share any topics you think are still missing.

closed 2024-12-05 by kumekay