espressif/idf-component-manager #6: Allow to download private components using HTTP Basic Authentication

harlem88, 2022-04-05:
### Related area

missing support

### Is your feature request related to a problem?

We use a components hosted on private GitLab. To download the component we use `Project Access Token` via HTTP Basic Authentication.
We try to use in a `git manifest property` a URL with basic authentication, like 
`git: https://git:${PROJ_ACC_TOKEN}@git.my_git.com/my_repo.git`, 
but the env `PROJ_ACC_TOKEN` is not resolved by idf-manager.

### Describe the solution you'd like

It would be awesome if the component manager either `manifest properties` for username and password or URL Basic Authentication can refer to environment variables.

### Describe alternatives you've considered

We could use `ssh` to download the component, but not all repositories are under our control, and we receive only the Project Access Token and it could be difficult to insert it in a CI workflow.

### Additional context

If it is possible to do this, maybe you can add an explanation somewhere.

``` yaml
## IDF Component Manager Manifest File
dependencies:
  my_repo:
      git: https://git:${PROJ_ACC_TOKEN}@git.my_git.com/my_repo.git
```

### I have checked existing list of Feature requests and I have provided enough information.

- [X] I confirm I have checked existing list of Feature requests and I have provided enough information.

kumekay, 2022-04-05:
@harlem88 Antonio, thank you for the feature request, 

Now we don't support the expansion of environment variables in the manifest file.

So far we suggested using git over ssh, but as you mentioned it won't work in your case,
as a workaround for your CI  job maybe you can perform variable substitution in the shell?

I.e. using `sed`:

```
sed -i 's/${PROJ_ACC_TOKEN}/'"${PROJ_ACC_TOKEN}"'/' idf_component.yml
```

harlem88, 2022-04-06:
@kumekay Sergei thank you for the answer.

But just out of curiosity, do you have any timeline for this feature? Or have you chosen not to support it?

kumekay, 2022-04-06:
@harlem88 Unfortunately I cannot give you any estimation now,
We may support it, however, it won't be trivial because now manifest is used as the source of truth to decide when it's time to solve dependencies and potential dependency on the environment may break the flow in a few places, so it should be introduced carefully

Let's keep this issue open

closed 2022-04-21 by espressif-bot

FrancescoVaiani, 2022-05-03:
@kumekay I see this issue is closed now. Is there an ETA on a release that includes the corresponding commit?

kumekay, 2022-05-03:
@FrancescoVaiani We are planning to release a new version this week

kumekay, 2022-05-19:
@FrancescoVaiani  https://pypi.org/project/idf-component-manager/1.1.0/ it's finally here