espressif/idf-component-manager #9: Directory does not exist when adding dependency to renamed main component

jonathanfoster, 2022-10-24:
### The Component Manager version

v1.1.3

### ESP-IDF Version

v5.0

### python Version

3.10

### Operating System

macOS 12.6

### Browser (for https://components.espressif.com  Issues)

Brave

### Description

I'm trying to add a dependency to a main component that has been renamed `src` and I receive the following error:
```shell
$ idf.py add-dependency --component src lvgl/lvgl>=8.*
ERROR: Directory "/Users/$USER/src/project/components/src" does not exist! Please specify a valid component under /Users/$USER/src/project
```
It appears the components path is added whenever the component is not `main`. https://github.com/espressif/idf-component-manager/blob/b1813aa551a66989e3969dbb82281a7add2a8b75/idf_component_manager/core.py#L106

### To Reproduce

```cpp
1. Create a new project
2. Rename `main` component to `src`.

set(EXTRA_COMPONENT_DIRS "src")
```
3. Add dependency
 
idf.py add-dependency --component src lvgl/lvgl>=8.*
```
```

### Expected behaviour

I expect the component to be added to the main `src` component.

### I have checked existing issues and online Documentation

- [X] I confirm I have checked existing issues and online Documentation.

kumekay, 2022-10-25:
Hi @jonathanfoster, thank you for the issue.

It is expected behavior, the `add-dependency` command doesn't work with components in the non-default path specified in EXTRA_COMPONENT_DIRS. We will update the help output of the command.

However, it just adds a line to the manifest file, 
so create `src/idf_component.yml` file with the content:

```
dependencies:
  lvgl/lvgl: ">=8.*"
```

The component will be downloaded on the next run of CMake, for example on `idf.py reconfigure`

jonathanfoster, 2022-10-25:
@kumekay Thanks for the heads up. Updating the help output to specify the constraint is a good start. 

I would also consider going a step further and adding an option to support a renamed main component (e.g. `--main`). The level of effort would be minimal and it wouldn't be a breaking change to the API. And since this behavior is supported by IDF, it would make component manager more cohesive with the rest of the framework.

This may seem like a small issue, and I'm sure it is when you look at IDF as a whole, but these small issues can compound and result in a poor experience for developers. It's these type of small details that really set you apart.

commit 54239d58 (2023-05-17): Description:
Added path as an option to support renamed main component, added tests, refactored code

Closes #9
Closes #10
closed 2023-05-17 by espressif-bot