espressif/idf-component-manager #12: Further documentation of manifest format requirements

ljden, 2022-11-01:
### Related area

Documentation

### Is your feature request related to a problem?

It is unclear how to specify a specific version of a component.
E.G. what of the following are valid and, if so, what do they mean:
```
version: "==1.0.0"
version: "~1.0.0"
version: "^1.0.0"
```

### Describe the solution you'd like

More details specifying what constitutes valid/invalid syntax. Versioning is just an example

### I have checked existing list of Feature requests and I have provided enough information.

- [X] I confirm I have checked existing list of Feature requests and I have provided enough information.

kumekay, 2022-11-01:
@ljden Thank you for the issue, 

While the proper documentation is on its way you can find some notes about the version specification format in the wiki:
https://github.com/espressif/idf-component-manager/wiki#requirement-specifications-format

I'll keep this issue open before we update it

ljden, 2022-11-02:
Thanks, that clears some of the things up.
How does versioning requirements work on the component side? Is that pulled from the `idf_components.yml` in the component root? This is in regards to an internally developed component hosted on a git repo
Can you specify a branch/tag and what are the requirements?
Is there a way to specify that you always want the latest version (ie check on every build)?
Can you specify both a version number and a (non-master) branch?

kumekay, 2022-11-02:
> Can you specify a branch/tag and what are the requirements?
> Can you specify both a version number and a (non-master) branch?

For git components, semantic version requirements don't make much sense, so the `version` field is handled differently. You can only specify some git references as version: branch, tag, or commit id:

```
git-only-cmp:
    path: git-only-cmp
    git: https://github.com/espressif/example_components.git
    # branch
    version: main 
    # Or commit
    # version: 6eb8d4a725643d657bd016e07b3260f8303eabf2
    # Or tag
    # version: v1.1.1
```
  

> Is there a way to specify that you always want the latest version (ie check on every build)?

Component manager runs on every CMake execution and only updates if there were changes in the manifest. The best way to trigger desired behavior is to to drop lock file on every compilation:

```
rm dependencies.lock && idf.py build
```

If you want to develop a component locally and always have the most recent version in your test project using local dependency is a more convenient way. In this way, it won't be copied to your project and used from the original location:

```
dependencies:
  my_component:
    path: ../../some_path/my_component
```

ljden, 2022-11-02:
Ah thanks for clearing that up.
I was experimenting with git components as a substitute to git submodules so that common dependencies aren't duplicated.
I had an issue building for linux (host tests) when including components, I've opened an issue on esp-idf, any ideas as to what might be the cause and if its an esp-idf issue or an issue with the component manager? https://github.com/espressif/esp-idf/issues/10088
If you think it might be the component manager I can open a separate issue for further discussion

ljden, 2022-11-15:
@kumekay Is there any documentation on what rules are available/valid?

kumekay, 2022-11-15:
@ljden now rules can only filter based on idf version and target, 

```
 rules: # will add "optional_component" only when all if clauses are True
      - if: "idf_version >=3.3,<5.0" # supports all SimpleSpec grammars (https://python-semanticversion.readthedocs.io/en/latest/reference.html#semantic_version.SimpleSpec)
      - if: "target in [esp32, esp32c3]" # supports boolean operator ==, !=, in, not in.
```

nicola-lunghi, 2023-01-06:
> git-only-cmp:
>     path: git-only-cmp
>     git: https://github.com/espressif/example_components.git
>     # branch
>     version: main 
>     # Or commit
>     # version: 6eb8d4a725643d657bd016e07b3260f8303eabf2
>     # Or tag
>     # version: v1.1.1

Hi,
can you add this to the official documentation/example idf_component file??

Thanks,
Nick

kumekay, 2023-01-06:
Hi @nicola-lunghi

This January is the month of documentation for us, and we are working on the new documentation site about the component manager, similar to the one for [ESP-IDF](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/index.html).

All the details about the manifest format will be covered there. I prefer to keep idf_component.yml simple and short, with the link to complete documentation. 

We are planning to publish the first docs there next week. So it's pretty close. I'll update this issue as soon as it goes live.

ljden, 2023-01-11:
@kumekay not sure if this is a problem with unclear documentation, or a bug, but it appears that:
"1.0.0-beta" does not satisfy `version: "1.0.*"` or `version: "~1.0"`
Is this expected? If so please ensure the documentation is clearer, if not I can open an issue

kumekay, 2023-01-11:
Hi @ljden 

By default, component manager is set to only download stable versions of dependencies. However, if you would like to include pre-release versions of a specific dependency in your project, it can be specified in the manifest file by adding the `pre_release: true`

```
  namespace/pre_release_component:
    version: "~1.0"
    pre_release: true # Allow downloading of pre-release versions
```

ljden, 2023-01-12:
Is that documented anywhere? 
Also, it didn't work - any ideas?
```yaml
  espressif/esp-idf-cxx:
    version: "~1.0"
    pre_release: true
```
```make
CMake Error at /home/vbox/esp/esp-idf/tools/cmake/component.cmake:249 (message):
  ERROR: 4 problems were found in the manifest file
  /home/vbox/sensor_manager/idf_component.yml:

  Invalid manifest format

  Invalid dependency format

  Dependency version spec format is invalid

  Unknown keys in dependency details: pre_release
```

kumekay, 2023-01-12:
@ljden 

`pre_release` flag was added to the component manager in version 1.2, however with any version you can specify dependency as: 

```    
  espressif/esp-idf-cxx:
    version: "~1.0.0-alpha"
```

If version specification includes pre-release specifier, pre-release versions are always included.

I've also found a confusing issue in behavior of the `pre_release` flag, if it is used with specifier like `~1.0` version `1.0.0-beta` still won't be installed with `pre_release: true`, because `~1.0` effectively translates to `>=1.0.0,<2.0.0`, but `1.0.0-beta` is less than `1.0.0`. We will address this issue.

igrr, 2023-01-12:
The other thing is that there should be a top-level `dependencies:` dictionary missing. If it's missing, i think we get errors such as posted by @ljden. I think it was implicitly assumed in @kumekay 's comment above:

```yaml
dependencies:
  espressif/esp-idf-cxx:
    version: "~1.0.0-alpha"
```

ljden, 2023-01-12:
@igrr yeah the rest of the file was assumed/well formed

Right, so would this match the beta or is it the same as `~1.0`?
```yaml
  version: "1.0.*"
  pre_release: true
```
Also, how does the `~` work with prereleases? Is "~1.0.0-beta" effectively `>=1.0.0-beta, <1.1.0` or `>=1.0.0-beta, <1.0.0`?
And is `~1.0.3` equivalent to `>=1.0.3, <1.1.0` or `>=1.0.0, <1.1.0` 

Thanks again for all the help

kumekay, 2023-01-12:
@ljden 

`1.0.*` is the same as `~1.0.0`, so it won't match beta, 
but `~1.0.0-beta` converts to `>=1.0.0-beta, <1.1.0` so it should work in your case. 

And thank you for your questions! They help us understand what we need to catch up in the documentation.

ljden, 2023-01-12:
Thanks, and what about my last example? Basically does the `~` ignore the patch number or does it take it as the lower bound and the `~` is just shorthand for go from my specified lower patch to any higher patch for this major+minor version?

kumekay, 2023-01-12:
@ljden I'm sorry, missed it, `~1.0.3` is equivalent to `>=1.0.3, <1.1.0`

ljden, 2023-01-13:
Thanks - I've got one more that is really getting into the corner cases but I'm just curious. Less so documentation and more so implementation.
You say that technically `1.0.0-beta < 1.0.0` and that `~1.0.0` is equivalent to `>=1.0.0 && <1.1.0`. I would assume it is not meant to, but will `1.1.0-beta` satisfy `~1.0.0`?

kumekay, 2023-01-13:
@ljden 

No, `~1.0.0` won't satisfy `1.1.0-beta`. 

BTW, you can experiment with different specs running `python` in your ESP-IDF shell and entering this snippet:

```
from idf_component_tools.semver import SimpleSpec, Version
SimpleSpec("~1.0.0").match(Version("1.1.0-beta"))
```

One more note on behavior of `>=1.0.0`. In the component manager 1.1.4 and earlier it would satisfy `1.1.0-beta`. It was a bug, and starting 1.2.0 `1.1.0-beta` will only satisfy `>=1.0.0` if `pre_release: true` is passed.

kumekay, 2023-03-09:
Described in https://github.com/espressif/idf-component-manager/commit/a7d6b43d9c0ce50a50fa416de51204976f9efa99

closed 2023-03-09 by kumekay