espressif/idf-component-manager #15: idf.py fullclean shows ERROR in remove_managed_components (IDFGH-9024)

AxelLin, 2022-12-25:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-dev-2509-gcfef24863f

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32C3DevKit-M1

### Power Supply used.

USB

### What is the expected behavior?

"idf.py fullclean" should work without any error.

### What is the actual behavior?

After upgrade to v5.1-dev-2509-gcfef24863f, now everytime executing "idf.py fullclean" shows below error:
I didn't change anythin in managed_components folder, and this error does not happen before upgrading esp-idf.

$ idf.py fullclean
Executing action: fullclean
Executing action: remove_managed_components
ERROR:
        Some components (joltwallet__littlefs) in the "managed_components" directory were modified         on the disk since the last run of the CMake. Content of this directory is managed automatically.

        If you want to keep the changes, you can move the directory with the component to the "components"         directory of your project.

        I.E. for "joltwallet__littlefs" run:
        mv /home/axel/esp/esp-idf-dev/apps/myproj/managed_components/joltwallet__littlefs /home/axel/esp/esp-idf-dev/apps/myproj/components/joltwallet__littlefs

        Or, if you want to discard the changes remove the ".component_hash" file from the component's directory.

        I.E. for "joltwallet__littlefs" run:
        rm /home/axel/esp/esp-idf-dev/apps/myproj/managed_components/joltwallet__littlefs/.component_hash

### Steps to reproduce.

idf.py build ; idf.py fullclean

AxelLin, 2022-12-28:
Hi @kumekay 
https://github.com/espressif/idf-component-manager/issues/15#issuecomment-1364633783 does not work, I got another error:
'''
-- Building ESP-IDF components for target esp32c3
CMake Error at /home/axel/esp/esp-idf/tools/cmake/build.cmake:542 (message):
  usage: __main__.py [-h] [--project_dir PROJECT_DIR]

                     [--interface_version {0,1}]
                     {prepare_dependencies,inject_requirements,inject_requrements}
                     ...

  __main__.py: error: argument --interface_version: invalid choice: 2 (choose
  from 0, 1)

Call Stack (most recent call first):
  /home/axel/esp/esp-idf/tools/cmake/project.cmake:447 (idf_build_process)
  CMakeLists.txt:26 (project)

-- Configuring incomplete, errors occurred!
'''
I give up using master for now (hope to get it fix soon).

kumekay, 2022-12-28:
@AxelLin Yes, unfortunately, after https://github.com/espressif/esp-idf/commit/17224f325450a001c4744064470cb6813b1c8514 master branch requires component-manager 1.2+

You can use the commit of ESP-IDF before this. 

The solution to the problem is not trivial and takes time. We will release it by next week.

kumekay, 2023-01-11:
@AxelLin, I apologize for the delay in addressing this issue. We are committed to releasing a solution by tomorrow.

kumekay, 2023-01-13:
@AxelLin 

Before we release, would you mind trying the version of the component manager from the branch `bugfix/default_excludes` to confirm that it resolves the issue for you?

You can install it with this command:

```
pip install git+https://github.com/espressif/idf-component-manager.git@bugfix/default_excludes -U
```

AxelLin, 2023-01-14:
> @AxelLin
>
> Before we release, would you mind trying the version of the component manager from the branch `bugfix/default_excludes` to confirm that it resolves the issue for you?
>
> You can install it [...]

Test with v5.1-dev-2889-gdb9586c53f with above pip install command, I still got below error:
```
$ idf.py fullclean build
Executing action: fullclean
Executing action: remove_managed_components
ERROR: 
        Some components (joltwallet__littlefs) in the "managed_components" directory were modified         on the disk since the last run of the CMake. Content of this directory is managed automatically.

        If you want to keep the changes, you can move the directory with the component to the "components"         directory of your project.

        I.E. for "joltwallet__littlefs" run:
        mv /home/axel/esp/esp-idf-dev/apps/wifi-sensor/managed_components/joltwallet__littlefs /home/axel/esp/esp-idf-dev/apps/wifi-sensor/components/joltwallet__littlefs

        Or, if you want to discard the changes remove the ".component_hash" file from the component's directory.

        I.E. for "joltwallet__littlefs" run:
        rm /home/axel/esp/esp-idf-dev/apps/wifi-sensor/managed_components/joltwallet__littlefs/.component_hash
```

kumekay, 2023-01-17:
@AxelLin 

I apologize for not mentioning it in my previous comment, but before installing from git, it is necessary to remove the previous version from PyPI.

 However, I am pleased to inform you that a new bugfix release has been made available on PyPI. You can update to v1.2.2 either by running `./install.sh` from the ESP-IDF directory or by running `pip install -U idf-component-manager` within the ESP-IDF environment.

AxelLin, 2023-01-18:
Hi @kumekay 
It works now, thanks a lot.

AxelLin, 2023-08-25:
@kumekay 

I'm using v5.1.1-1-gd3c99ed3b8 to build for esp32c3.

$ idf.py fullclean build
Executing action: fullclean
Executing action: remove_managed_components
ERROR: Some components (joltwallet__littlefs) in the "managed_components" directory were modified on the disk since the last run of the CMake. Content of this directory is managed automatically.
If you want to keep the changes, you can move the directory with the component to the "components"directory of your project.
I.E. for "joltwallet__littlefs" run:
mv /home/axel/esp/esp-idf-dev/apps/myproj/managed_components/joltwallet__littlefs /home/axel/esp/esp-idf-dev/apps/myproj/components/joltwallet__littlefs
Or, if you want to discard the changes remove the ".component_hash" file from the component's directory.
I.E. for "joltwallet__littlefs" run:
rm /home/axel/esp/esp-idf-dev/apps/myproj/managed_components/joltwallet__littlefs/.component_hash
axel@alpha:~/esp/esp-idf-dev/apps/myproj$ cat /home/axel/esp/esp-idf-dev/apps/myproj/managed_components/joltwallet__littlefs/.component_hash
2e392b279f0bc282a9dfb66608de4b5351d5ab4249254cbb5ebae09099c045ec

But I didn't touch anything in managed_components folder.

kumekay, 2023-08-25:
Hello @AxelLin  

It looks like it's the same problem as in https://github.com/espressif/idf-component-manager/issues/40 

For now, you can use `IDF_COMPONENT_OVERWRITE_MANAGED_COMPONENTS=1` as a quick workaround.

AxelLin, 2023-09-07:
@kumekay 
When would you fix this?

kumekay, 2023-09-08:
Hello @AxelLin 

We are still looking for the best approach to dealing with this problem maintaining guarantees on reproducibility that component manager promises. 

We will address it in the coming release

kumekay, 2023-09-15:
Hello @AxelLin the files that littlefs generates during build are already gitignored and can be excluded to resolve the problem.

Unfortunately, we can only do it for coming ESP-IDF version 5.2. For existing versions of IDF, we can exclude filters to the idf_component.yml of the littlefs. So for this particular component, will do our best to fix it next week.

AxelLin, 2023-12-16:
Hi @kumekay 

This is probably off-topic from the origianl issue, I'm wondering if you can help to answer it:
(I asked the question in ep32 forum https://www.esp32.com/viewtopic.php?f=13&t=37005, but unforunately no response there.)

While debugging, sometimes I would like to add some debug code in the managed component files,
but it seems the idf.py build always build with a clean cloned managed component files.
Is it possible to make some local changes in the managed component files and build with the local changes?
This would be helpful to debug issues in the managed component files.

kumekay, 2023-12-17:
Hi @AxelLin 
You can move a component from `managed_components` directory of your project to `components` directory and modify it there. Then IDF will use the component from `components` directory and component manager won't make a copy in the `managed_components`

hfudev, 2024-07-10:
related to https://github.com/espressif/idf-component-manager/commit/193974bf24f92e540c0f03c96b008905683ead40

hfudev, 2024-08-13:
Version 2.0.0 has been released, and I think 193974bf24f92e540c0f03c96b008905683ead40 solves your original problem.

closing the issue.

closed 2024-08-13 by hfudev