espressif/idf-component-manager #16: Support conditional dependence

leeebo, 2023-01-03:
### Related area

dependence

### Is your feature request related to a problem?

I want to develop an example supports both IDF5.0 and IDF4.4.  But `led_strip` > 2.0.0 only works in IDF5.0, and `led_strip` ~ 1.1.0 only works for IDF4.4. For different IDF versions, I need to specify different version of `led_strip` also, any method to achieve this? 

### Describe the solution you'd like

Support conditional dependence like :

```
dependencies:
  led_strip : "~1.1.0"
    if: "idf: ==4.4"
  led_strip : "~2.0.0"
    if: "idf: ==5.0"
```

### I have checked existing list of Feature requests and I have provided enough information.

- [X] I confirm I have checked existing list of Feature requests and I have provided enough information.

igrr, 2023-01-03:
In theory, conditional dependencies are possible now (e.g. https://github.com/espressif/esp-rainmaker/blob/fe94cc6994c4d012bad3ce04d6ae07b616894714/components/esp_rainmaker/idf_component.yml#L5-L6), but in the specific case, I could not get the following to work correctly:
```yaml
dependencies:
  espressif/led_strip:
    version: "1.*"
    rules:
      - if: "idf_version <5.0"
  espressif/led_strip:
    version: "2.*"
    rules:
      - if: "idf_version >=5.0"
```
With IDF v5.x, this correctly downloads led_strip=2.0.0. But with IDF v4.x, it doesn't download anything.

kumekay, 2023-01-03:
@leeebo  Thank you for the question, 
as @igrr already mentioned, conditional dependencies are supported with the `rules` block.

However, the same dependency cannot be listed twice - duplicated keys are not allowed in the YAML file. We may overcome this limitation at some point, but the current syntax of the manifest cannot handle it.

Could you please describe why you need if rules in this particular case?

`espressif/led_strip=2.0` depends on idf>=5.0, so it will never be installed in a project with idf 4.4. 

So if dependency is defined like: `espressif/led_strip: "*"`, it will install version 2.0 with IDF 5.0+ and 1.0 for IDF 4.4

The only case where you can end up with the wrong version is if you switch from ESP-IDF 4.4 to 5.0 in the project directory, keeping the `dependencies.lock` file. It will happen, because led_strip=1.0 doesn't specify supported IDF versions.

A little bit hacky, but a working solution in this particular case of a component from the default `espressif` namespace is to specify dependency once with the namespace and the second time without it:

```
dependencies:
  espressif/led_strip:
    version: "1.*"
    rules:
      - if: "idf_version <5.0"
  led_strip:
    version: "2.*"
    rules:
      - if: "idf_version >=5.0"
```

One more note, there is no led_strip=1.1.0, only 1.1.0-alpha. By default component manager doesn't install pre-release version, so you may want to add `pre_release: true` to the dependency.

leeebo, 2023-01-04:
`Could you please describe why you need if rules in this particular case?`
I just want my examples using led_strip to work both on IDF4.x and IDF5.x

@kumekay Thanks for your explanations, I think both methods you offered can solve my problem

kumekay, 2023-01-06:
@leeebo It's great! In future versions, we will support alternative syntax for dependencies, so it will be possible to list the same component twice with different conditions.

hfudev, 2023-08-28:
Hi @leeebo The feature is implemented in https://github.com/espressif/idf-component-manager/commit/c29b269f65b61a83f18c368740dbd5a71169eba1, and will be released together with version 1.4.0.

Your current use case could be implemented as

```yaml
dependencies:
  led_strip:
    matches:
      - if: "idf_version==4.4"
        version: "~1.1.0"
      - if: "idf_version==5.0"
        version: "~2.0.0"
```

For detailed documentation, you may refer to https://docs.espressif.com/projects/idf-component-manager/en/latest/reference/manifest_file.html#matches-and-rules.