espressif/idf-component-manager #18: Compile Fails: Some components (espressif/fmt) in the "managed_components" directory were modified

sukesh-ak, 2023-01-09:


kumekay, 2023-01-09:
Hello @sukesh-ak,

I noticed that your issue seems to be similar to the one discussed in issue !15. We are currently working on a solution.

As a temporary workaround for ESP-IDF 5.0, you may try downgrading idf-component-manager to version 1.1.4. To do this, please run the following command within your ESP-IDF environment (after running export.ps1):

```
pip install idf-component-manager==1.1.4
```

I hope this helps.

sukesh-ak, 2023-01-10:
Thanks. I looked into issues on IDF repo so didn’t see this. 

It compiles after deleting that folder `rm -rf managed_components`. 

I am using Windows so for now its ok. Some of the people who are using my project reported this issue so reported it here.

kumekay, 2023-01-18:
@sukesh-ak 
A new bugfix release with the fix has been made available on PyPI. You can update to v1.2.2 either by running ./install.sh from the ESP-IDF directory or by running `pip install -U idf-component-manager` within the ESP-IDF environment (after sourcing `export.sh`).

If you need any help, please don't hesitate to reopen this issue.

closed 2023-01-18 by kumekay

sukesh-ak, 2023-01-18:
> @sukesh-ak A new bugfix release with the fix has been made available on PyPI. You can update to v1.2.2 either by running ./install.sh from the ESP-IDF directory or by running `pip install -U [...]

Thank you for the update. Will test once I get a chance.