espressif/idf-component-manager #20: idf.py fullclean does not remove dependencies.lock

ljden, 2023-01-17:
### The Component Manager version

v1.1.4

### ESP-IDF Version

v5.0

### python Version

3.10.6

### Operating System

Linux

### Description

`idf.py fullclean` does not remove the `dependecies.lock` file. This seems to mean that if the `idf_components.yml` file has been modified (eg from git to path) the component manager will not update the dependency and will use the existing `dependencies.lock` file to resolve dependencies

### To Reproduce

```cpp
`idf.py fullclean && ls -al dependencies.lock`
```

### Expected behaviour

`idf.py fullclean` removes all files created in the build system including the `dependecies.lock`

### I have checked existing issues and online Documentation

- [X] I confirm I have checked existing issues and online Documentation.

kumekay, 2023-01-18:
Hello @ljden, thank you for reaching out with your question.

The `idf.py fullclean` behavior is intentional and designed to ensure repeatable builds. The `dependencies.lock` file is treated similarly to the sdkconfig file and we recommend keeping track of it with version control.

We will add a new idf.py subcommand to upgrade dependencies in the upcoming component manager release, to handle this case. However, implementation of the command will be similar to running `rm dependencies.lock && idf.py reconfigure`

ljden, 2023-01-19:
right - not what I was expecting but if there's thought behind it ok.
Re dependencies.lock being version controlled - that doesn't seem to make sense as the file is both auto generated and unmodifiable by the user. Also pretty sure esp-idf recommend you don't source control sdkconfig and instead have a defaults file you do track but I could be remembering incorrectly.
Thanks for the clarifications!

closed 2023-01-19 by ljden