espressif/idf-component-manager #23: Unable to override managed component with mock

ljden, 2023-02-07:
### The Component Manager version

v1.1.4

### ESP-IDF Version

v5.0

### python Version

3.10.6

### Operating System

Linux

### Description

Having issues building a host test for a component that uses the component manager to manage a dependency, that then also needs to mock said dependency for the host test.

```make
-- building ESP EVENT MOCKS
-- building FREERTOS MOCKS (only task, event-groups and queue)
-- building esp-idf-cxx MOCKS (I2C only)
CMake Error at /home/vbox/esp/esp-idf/tools/cmake/component.cmake:249 (message):
  ERROR: Cannot process component requirements.  Multiple candidates to
  satisfy project requirements:

    requirement: "esp-idf-cxx" candidates: "esp-idf-cxx, espressif__esp-idf-cxx"

Call Stack (most recent call first):
  /home/vbox/esp/esp-idf/tools/cmake/build.cmake:553 (__component_get_requirements)
  /home/vbox/esp/esp-idf/tools/cmake/project.cmake:440 (idf_build_process)
  CMakeLists.txt:15 (project)
```

See https://github.com/espressif/esp-idf/issues/10599#issue-1552329641 for more details

### To Reproduce

```cpp
vbox@vbox:~/comp/host_test$ cat comp/host_test/CMakeLists.txt
cmake_minimum_required(VERSION 3.16)
set(CMAKE_CXX_STANDARD 20)
set(CMAKE_CXX_STANDARD_REQUIRED ON)

include($ENV{IDF_PATH}/tools/cmake/project.cmake)
set(COMPONENTS main)

idf_build_set_property(COMPILE_DEFINITIONS "-DconfigTICK_RATE_HZ=1000" APPEND)

list(APPEND EXTRA_COMPONENT_DIRS "$ENV{IDF_PATH}/tools/mocks/freertos/")
list(APPEND EXTRA_COMPONENT_DIRS "$ENV{IDF_PATH}/tools/mocks/esp_event/")
...
list(APPEND EXTRA_COMPONENT_DIRS "./mock_comps/mocks/esp-idf-cxx")

project(sensor_manager_host_test)
vbox@vbox:~/comp/host_test$ cat main/CMakeLists.txt
idf_component_register(SRCS "test_comp.cpp"
    WHOLE_ARCHIVE
    PRIV_REQUIRES comp cmock
    )

find_package(Catch2 3 REQUIRED)
find_package(trompeloeil REQUIRED)

target_link_libraries(${COMPONENT_LIB}
  PRIVATE
  Catch2::Catch2WithMain
  trompeloeil::trompeloeil
  )
vbox@vbox:~/comp/host_test$ cat main/idf_component.yml
dependencies:
  idf: "5.0.*"
  comp:
    path: "../../"
vbox@vbox:~/comp/host_test$ cat ../idf_component.yml
version: "0.0.1"
description: component
dependencies:
  idf: "5.0.*"
  espressif/esp-idf-cxx: "~1.0.0-beta"
...
vbox@vbox:~/sensor_manager/host_test$ tree mock_comps/
mock_comps/
└── mocks
    ├── esp-idf-cxx
    │   ├── CMakeLists.txt
    │   └── include
    │       └── i2c_cxx.hpp
    └── README.md

```make
vbox@vbox:~/comp/host_test$ rm -rf dependencies.lock managed_components/ sdkconfig build/ && idf.py build
...
-- building ESP EVENT MOCKS
-- building FREERTOS MOCKS (only task, event-groups and queue)
-- building esp-idf-cxx MOCKS (I2C only)
CMake Error at /home/vbox/esp/esp-idf/tools/cmake/component.cmake:249 (message):
  ERROR: Cannot process component requirements.  Multiple candidates to
  satisfy project requirements:

    requirement: "esp-idf-cxx" candidates: "espressif__esp-idf-cxx, esp-idf-cxx"

Call Stack (most recent call first):
  /home/vbox/esp/esp-idf/tools/cmake/build.cmake:553 (__component_get_requirements)
  /home/vbox/esp/esp-idf/tools/cmake/project.cmake:440 (idf_build_process)
  CMakeLists.txt:15 (project)
```
```

### Expected behaviour

Able to mock a component that is managed by idf_component.yml

### I have checked existing issues and online Documentation

- [X] I confirm I have checked existing issues and online Documentation.

kumekay, 2023-02-08:
@ljden, thank you for reporting the issue.

The mocking support and component manager were developed independently, and we will investigate the issue to find a solution.

kumekay, 2025-10-06:
@ljden Do you still have this issue? As for now we are not able to reproduce it.