espressif/idf-component-manager #24: Need solutions for `Can't reliably evaluate relative path without context `

leeebo, 2023-02-13:
### Related area

overwrite_path

### Is your feature request related to a problem?

We have 2 components (**A** and **B**) and 1 examples (**E**) in the same repo:

In **E**'s `yml`,  we set **B** as the dependency, and give it a `overwrite_path`

In **B**'s `yml`,  we set **A** as the dependency, and **also** give it a `overwrite_path`

Then the  `Can't reliably evaluate relative path without context` will happends...

We need the `overwrite_path` works as normal (infer the path?), because we always want build with all the local code in the development stage...

### Describe the solution you'd like

Save the path context of each component, Infer the `overwrite_path`

### I have checked existing list of Feature requests and I have provided enough information.

- [X] I confirm I have checked existing list of Feature requests and I have provided enough information.

kumekay, 2023-02-13:
@leeebo Thank you for the issue, 

`Can't reliably evaluate relative path without context` is not intended to be a user-facing error, so it's a bug. We will take a look and will fix it in the next release.

kumekay, 2023-02-15:
Fixed in 6cc9347dee2b6dd5d61b4cc23e843b3fe921ab65

closed 2023-02-15 by kumekay

leeebo, 2023-02-22:
@kumekay Thanks, as the next version 1.3.0 not published yet, any method for me to replace the version installed in ESP-IDF venv with the repo's latest code?

kumekay, 2023-02-22:
@leeebo 

After you activate your ESP-IDF env by opening ESP-IDF shell on Windows or sourcing export.sh/fish/ps1:

```
pip uninstall -y idf-component-manager
pip install git+https://github.com/espressif/idf-component-manager
```