espressif/idf-component-manager #25: Erroneous "Multiple Candidates" error from espressif/qrcode

Panometric, 2023-02-17:
### The Component Manager version

1.2.2

### ESP-IDF Version

4.4

### python Version

3.9.1

### Operating System

Widnows 10

### Description

Cannot build wifi_prov_mgr example:
```
Processing 2 dependencies:
[1/2] espressif/qrcode (0.1.0)
[2/2] idf (4.4.4)
-- DEBUG: Use esp-modbus component folder: C:/Users/Mike/esp/esp-idf/components/freemodbus.
CMake Error at C:/Users/Mike/esp/esp-idf/tools/cmake/component.cmake:237 (message):
  ERROR: Cannot process component requirements.  Multiple candidates to
  satisfy project requirements:

    requirement: "qrcode" candidates: "qrcode, espressif__qrcode"

Call Stack (most recent call first):
  C:/Users/Mike/esp/esp-idf/tools/cmake/build.cmake:486 (__component_get_requirements)
  C:/Users/Mike/esp/esp-idf/tools/cmake/project.cmake:384 (idf_build_process)
  CMakeLists.txt:6 (project)
```

### To Reproduce

```cpp
idf.py menuconfig
```

### Expected behaviour

There is only one qrcode at https://components.espressif.com/components/espressif/qrcode
It should just download it and continue building. 

### I have checked existing issues and online Documentation

- [X] I confirm I have checked existing issues and online Documentation.

kumekay, 2023-02-17:
@Panometric thank you for the issue, 

I've just tried, but didn't manage to reproduce your problem. 
Could you please provide more details?
Do you have EXTRA_COMPONENT_DIRS configured in your project? Could you please share the content of your `main/idf_component.yml`?

Could you please also try to reproduce the problem on an empty project by running:

```
idf.py create-project tstqrcode
cd tstqrcode
idf.py add-dependency qrcode
idf.py reconfigure
```

Panometric, 2023-02-17:
@kumekay Thanks for the quick reply. I agree, your tstqrcode project works fine. 

My EXTRA_COMPONENT_DIRS  is set in CMakeLists.txt. And that appears to be a problem. 
```
set(EXTRA_COMPONENT_DIRS $ENV{IDF_PATH}/examples/common_components/qrcode)
```

 My file is simply this:
```
dependencies:
  qrcode: "^0.1.0"
  ## Required IDF version
  idf:
    version: ">=4.1.0"
```

I added the idf_component.yml because the example would not compile, the qrcode was missing. 

I can see why I should not define it twice, but it's an easy mistake to make and the error did not lead me to understand where it was coming from.

closed 2023-02-17 by Panometric