espressif/idf-component-manager #26: How to delete a component?

dizcza, 2023-03-06:
### Related area

It's missing in the docs

### Is your feature request related to a problem?

Guys,
I'm crying to be unable to delete one of my managed components (`espressif/esp_websocket_client`).

```
$ cat dependencies.lock 
dependencies:
  espressif/esp_websocket_client:
    component_hash: 21b3c7f85d6266d39d58c371d1143f1dbc75970d39208f5e5df6eb66106dfa04
    source:
      service_url: https://api.components.espressif.com/
      type: service
    version: 0.0.3
  idf:
    component_hash: null
    source:
      type: idf
    version: 5.0.0
manifest_hash: 01c5a924ab8f6483a4738a7a686bacf04143e42b141c3c185526c67f7705a9d0
target: esp32
version: 1.0.0
```

Here is what I tried
```
$ python -m idf_component_manager --name espressif/esp_websocket_client
usage: __main__.py [-h] [--path PATH] [--namespace NAMESPACE] [--service-profile SERVICE_PROFILE] [--name NAME]
                   [--archive ARCHIVE] [--job JOB] [--version VERSION] [--skip-pre-release] [--check-only]
                   [--allow-existing] [--example EXAMPLE]
                   {pack-component,upload-component,upload-component-status,create-project-from-example,delete-version}
__main__.py: error: the following arguments are required: command
(esp) dizcza@ubu:~/Projects/Embedded/eclipse-workspace/esp32-sdpsensor$ python -m idf_component_manager --name espressif/esp_websocket_client delete-version
WARNING: Deprecated! New CLI command: "compote component delete"
Traceback (most recent call last):
  File "/home/dizcza/miniconda3/envs/esp-vscode/lib/python3.9/runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/home/dizcza/miniconda3/envs/esp-vscode/lib/python3.9/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/home/dizcza/.espressif/python_env/idf5.0_py3.9_env/lib/python3.9/site-packages/idf_component_manager/__main__.py", line 101, in <module>
    main()
  File "/home/dizcza/.espressif/python_env/idf5.0_py3.9_env/lib/python3.9/site-packages/idf_component_manager/__main__.py", line 88, in main
    check_required_args(args, ['name', 'version'])
  File "/home/dizcza/.espressif/python_env/idf5.0_py3.9_env/lib/python3.9/site-packages/idf_component_manager/__main__.py", line 31, in check_required_args
    raise ValueError('--{} is required'.format(_f.replace('_', '-')))
ValueError: --version is required
(esp) dizcza@ubu:~/Projects/Embedded/eclipse-workspace/esp32-sdpsensor$ cat dependencies.lock 
dependencies:
  espressif/esp_websocket_client:
    component_hash: 21b3c7f85d6266d39d58c371d1143f1dbc75970d39208f5e5df6eb66106dfa04
    source:
      service_url: https://api.components.espressif.com/
      type: service
    version: 0.0.3
  idf:
    component_hash: null
    source:
      type: idf
    version: 5.0.0
manifest_hash: 01c5a924ab8f6483a4738a7a686bacf04143e42b141c3c185526c67f7705a9d0
target: esp32
version: 1.0.0
(esp) dizcza@ubu:~/Projects/Embedded/eclipse-workspace/esp32-sdpsensor$ python -m idf_component_manager --name espressif/esp_websocket_client delete-version 0.0.3
usage: __main__.py [-h] [--path PATH] [--namespace NAMESPACE] [--service-profile SERVICE_PROFILE] [--name NAME]
                   [--archive ARCHIVE] [--job JOB] [--version VERSION] [--skip-pre-release] [--check-only]
                   [--allow-existing] [--example EXAMPLE]
                   {pack-component,upload-component,upload-component-status,create-project-from-example,delete-version}
__main__.py: error: unrecognized arguments: 0.0.3
(esp) dizcza@ubu:~/Projects/Embedded/eclipse-workspace/esp32-sdpsensor$ python -m idf_component_manager --name espressif/esp_websocket_client delete-version --version 0.0.3
WARNING: Deprecated! New CLI command: "compote component delete"
ERROR: Failed to get API Token from the config file
(esp) dizcza@ubu:~/Projects/Embedded/eclipse-workspace/esp32-sdpsensor$ compote component delete
Usage: compote component delete [OPTIONS]
Try 'compote component delete -h' for help.

Error: Missing option '--name'.
(esp) dizcza@ubu:~/Projects/Embedded/eclipse-workspace/esp32-sdpsensor$ compote component delete espressif/esp_websocket_client
Usage: compote component delete [OPTIONS]
Try 'compote component delete -h' for help.

Error: Missing option '--name'.
(esp) dizcza@ubu:~/Projects/Embedded/eclipse-workspace/esp32-sdpsensor$ compote component delete --name espressif/esp_websocket_client
Usage: compote component delete [OPTIONS]
Try 'compote component delete -h' for help.

Error: Missing option '--version'.
(esp) dizcza@ubu:~/Projects/Embedded/eclipse-workspace/esp32-sdpsensor$ compote component delete --name espressif/esp_websocket_client --version 0.0.3
ERROR: Failed to get API Token from the config file
```

Also tried deleted the corresponding lines in `dependencies.lock`, removing the entire `managed_components` folder but after I run `idf.py reconfigure`, the `espressif/esp_websocket_client` is *still* back again and again.

Silly me.

### Describe the solution you'd like

A user-friendly command line to remove a managed component.

### Additional context

I did check issues and the docs https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/tools/idf-component-manager.html.

I'm wondering: has no one requested/experienced this before me?... Weird.

### I have checked existing list of Feature requests and I have provided enough information.

- [X] I confirm I have checked existing list of Feature requests and I have provided enough information.

kumekay, 2023-03-07:
Hi @dizcza 
Thank you for your question, 
If I understand correctly, you are trying to remove a dependency of your project installed by the component manager and stored in `managed_component`. Unfortunately, there is no CLI command to delete a dependency. You must manually remove `espressif/esp_websocket_client` (or `esp_websocket_client`) row from the `idf_component.yml` manifest file `dependencies` section. I guess in your `main` directory (or it may be in some other component either in the `components` subdirectory or one of the paths specified in EXTRA_COMPONENT_DIRS if you use them)

`delete-version` or `compote component delete` command is made for component developers to delete a component version from the registry. 

Last few weeks, we worked hard to cover an obvious documentation gap, and it will be published very soon. We will cover this topic too.

dizcza, 2023-03-07:
Ah, so there it is. Because I tried removing lines in the `dependencies.lock` located in the root directory. And it had no effects.

All right. Your solution worked out.

closed 2023-03-07 by dizcza