espressif/idf-component-manager #28: Authentication Token

mayankbytebeam, 2023-03-21:
### The Component Manager version

v1.2.2

### ESP-IDF Version

v5.0

### python Version

3.8.7

### Operating System

Windows 10

### Browser (for https://components.espressif.com  Issues)

Edge

### Description

can someone please let me know, which authentication token are you referring in docs. Either it is not clear, or I don't know.

https://docs.espressif.com/projects/idf-component-manager/en/latest/guides/packaging_components.html#authentication

PS: I want to publish my idf component to idf component registery.

### To Reproduce

```cpp
Upload a component without setting the token.
```

### Expected behaviour

ERROR: Failed to get API Token from the config file

### Additional info

NA

### I have checked existing issues and online Documentation

- [X] I confirm I have checked existing issues and online Documentation.

kumekay, 2023-03-21:
@mayankbytebeam Thank you for opening the issue. This documentation site only covers the coming 1.3.0 release. Commands for sign-up and token management are not merged yet with the main branch. I'll keep this issue open and keep you updated.

mayankbytebeam, 2023-03-21:
Thanks for the information. 

When are you going to roll out the next 1.3.0 release?
Also, how to publish the package with the 1.2.0 release?

mayankbytebeam, 2023-03-23:
@kumekay We are about to release our idf component and We need to publish the idf component to the registery as soon as possible. can you please help us with the current release (1.2.0)?

kumekay, 2023-03-23:
@mayankbytebeam Could you, please send me the link to your project and desired namespace (company name) to sergei.silnov@espressif.com

mayankbytebeam, 2023-03-23:
@kumekay please head over to https://github.com/bytebeamio/bytebeam-esp-idf-sdk

I think we will be releasing next week, so for the same I will connect with you over mail.  Apart from this, I am curious to know about the update process as we are planning for the weekly bug fixes, sdk improvements updates after the initial release.

Thanks for the help, we really appreciate it.

kumekay, 2024-02-28:
Since the Bytebeam SDK is already in the registry https://components.espressif.com/components/bytebeamio/bytebeam-esp-idf-sdk I'm closing this

closed 2024-02-28 by kumekay