espressif/idf-component-manager #29: Recreate a build from dependencies.lock file

ljden, 2023-04-11:
### Related area

repeatable builds

### Is your feature request related to a problem?

I need to keep a copy of the `managed_components` directory to recreate a previous build. If I build with an old `dependencies.lock` and no `managed_components` the idf manager will pull the latest components and update the lock file. There should be some way to recreate a build from a lock file without having the lock file be updated

### Describe the solution you'd like

I would like to be able to recreate a previous build using only the `dependencies.lock` file, and not need to keep a copy of the `managed_components` directory

### Additional context

I couldn't see there being an option for this, if it is there and I have missed it let me know

### I have checked existing list of Feature requests and I have provided enough information.

- [X] I confirm I have checked existing list of Feature requests and I have provided enough information.

ljden, 2023-04-12:
I would presume the solver is rerunning as the `managed_components` have been removed/not stored.
The stated reason is "Manifest hash changed, solving dependencies."

Personally I think that you should be able to recreate a build without the `managed_components` as they can be re-pulled from their remote sources.

kumekay, 2023-04-19:
@ljden 

You are right in stating that `dependencies.lock` should be enough to reproduce the build, and it's an expected behavior. 

I did an experiment to reproduce your situation.
However, I wasn't able to reproduce the issue. When I move the project directory, the manifest hash stays the same. 

Here are my steps. Could you please check that problem exists in your case for a minimal project? 

Few more questions about your environment:

- Does the IDF version stay the same?
- Do you have dependencies in your manifest file comping from a git repo?

1. Create a project: 
```
cd ~/t
idf.py create-project test
cd test
idf.py add-dependency example/cmp
idf.py reconfigure
```

Output:

```
Dependencies lock doesn't exist, solving dependencies.
.........Updating lock file at /Users/ku/t/test/dependencies.lock
Processing 2 dependencies:
[1/2] example/cmp (3.3.8)
[2/2] idf (5.1.0)
```

And the lock file:
```
dependencies:
  example/cmp:
    component_hash: 8c5891c21ecdebd1f545252144a5bf541071dce316f3e12eb288a5bcb506badf
    source:
      service_url: https://api.components.espressif.com/
      type: service
    version: 3.3.8
  idf:
    component_hash: null
    source:
      type: idf
    version: 5.1.0
manifest_hash: ff7214c6404c4f2e8356e01390590a4f0e4021c877b84ae0c41a743026e919f7
target: esp32
version: 1.0.0
```

Then move the project:

```
mv ~/t/test ~/t/test2
cd ~/t/test2
rm -rf build managed_components
idf.py reconfigure
```

But the solver is not triggered:

```
-- Building ESP-IDF components for target esp32
Processing 2 dependencies:
[1/2] example/cmp (3.3.8)
[2/2] idf (5.1.0)
```

The lock file is untouched.
```
manifest_hash: ff7214c6404c4f2e8356e01390590a4f0e4021c877b84ae0c41a743026e919f7
```

ljden, 2023-04-19:
The issue is more so when a component dependency has been updated.

Ie:
* add git dependency
* build
* add new commit to remote git dependency
* remove managed_components
* rebuild

I would expect the component manager to try and pull the same commit hash as in the lock file, rather than pulling the latest master commit.

Predominantly our dependencies come from git sources (internally developed) and idf version stays the same

kumekay, 2023-04-20:
@ljden Thank you for the details, I was able to reproduce it, and it's of course a bug. We will fix it in the next release. I'll update this issue as soon as is it will be merged into the main branch.

kumekay, 2023-06-30:
@ljden this should be fixed in https://github.com/espressif/idf-component-manager/releases/tag/v1.3.0

https://docs.espressif.com/projects/idf-component-manager/en/latest/guides/updating_component_manager.html

closed 2023-06-30 by kumekay