espressif/idf-component-manager #31: Update to v5.2-dev broke build

higaski, 2023-05-05:
### The Component Manager version

v1.2.2

### ESP-IDF Version

v5.2-dev

### python Version

3.11.3

### Operating System

Arch

### Description

I've just tried to update to ESP-IDF tag [v5.2-dev](https://github.com/espressif/esp-idf/releases/tag/v5.2-dev). For some reason this broke my entire build system. Pretty much any idf.py command I try to run gives me a huge traceback.

```sh
CMake Error at /home/vinci/esp/esp-idf/tools/cmake/build.cmake:539 (message):
  Traceback (most recent call last):

    File "<frozen runpy>", line 198, in _run_module_as_main
    File "<frozen runpy>", line 88, in _run_code
    File "/home/vinci/.espressif/python_env/idf5.2_py3.11_env/lib/python3.11/site-packages/idf_component_manager/prepare_components/__main__.py", line 6, in <module>
      main()
    File "/home/vinci/.espressif/python_env/idf5.2_py3.11_env/lib/python3.11/site-packages/idf_component_manager/prepare_components/prepare.py", line 124, in main
      args.func(args)
    File "/home/vinci/.espressif/python_env/idf5.2_py3.11_env/lib/python3.11/site-packages/idf_component_manager/prepare_components/prepare.py", line 31, in prepare_dep_dirs
      ).prepare_dep_dirs(
        ^^^^^^^^^^^^^^^^^
    File "/home/vinci/.espressif/python_env/idf5.2_py3.11_env/lib/python3.11/site-packages/idf_component_manager/core.py", line 62, in wrapper
      return func(self, *args, **kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^
    File "/home/vinci/.espressif/python_env/idf5.2_py3.11_env/lib/python3.11/site-packages/idf_component_manager/core.py", line 438, in prepare_dep_dirs
      downloaded_component_paths, downloaded_component_version_dict = download_project_dependencies(
                                                                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    File "/home/vinci/.espressif/python_env/idf5.2_py3.11_env/lib/python3.11/site-packages/idf_component_manager/dependencies.py", line 111, in download_project_dependencies
      solution = solver.solve()
                 ^^^^^^^^^^^^^^
    File "/home/vinci/.espressif/python_env/idf5.2_py3.11_env/lib/python3.11/site-packages/idf_component_manager/version_solver/version_solver.py", line 34, in solve
      self.solve_manifest(manifest)
    File "/home/vinci/.espressif/python_env/idf5.2_py3.11_env/lib/python3.11/site-packages/idf_component_manager/version_solver/version_solver.py", line 53, in solve_manifest
      self.solve_component(requirement)
    File "/home/vinci/.espressif/python_env/idf5.2_py3.11_env/lib/python3.11/site-packages/idf_component_manager/version_solver/version_solver.py", line 56, in solve_component
      cmp_with_versions = requirement.source.versions(
                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    File "/home/vinci/.espressif/python_env/idf5.2_py3.11_env/lib/python3.11/site-packages/idf_component_tools/sources/web_service.py", line 143, in versions
      cmp_with_versions = self.api_client.versions(component_name=name, spec=spec)
                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    File "/home/vinci/.espressif/python_env/idf5.2_py3.11_env/lib/python3.11/site-packages/idf_component_tools/api_client.py", line 285, in wrapper
      return f(self, request=request, *args, **kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    File "/home/vinci/.espressif/python_env/idf5.2_py3.11_env/lib/python3.11/site-packages/idf_component_tools/api_client.py", line 305, in versions
      body = request(
             ^^^^^^^^
    File "/home/vinci/.espressif/python_env/idf5.2_py3.11_env/lib/python3.11/site-packages/idf_component_tools/api_client.py", line 282, in request
      return self._base_request(
             ^^^^^^^^^^^^^^^^^^^
    File "/home/vinci/.espressif/python_env/idf5.2_py3.11_env/lib/python3.11/site-packages/idf_component_tools/api_client.py", line 213, in _base_request
      response = session.request(
                 ^^^^^^^^^^^^^^^^
    File "/home/vinci/.espressif/python_env/idf5.2_py3.11_env/lib/python3.11/site-packages/requests/sessions.py", line 587, in request
      resp = self.send(prep, **send_kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    File "/home/vinci/.espressif/python_env/idf5.2_py3.11_env/lib/python3.11/site-packages/requests/sessions.py", line 701, in send
      r = adapter.send(request, **kwargs)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    File "/home/vinci/.espressif/python_env/idf5.2_py3.11_env/lib/python3.11/site-packages/cachecontrol/adapter.py", line 48, in send
      cached_response = self.controller.cached_request(request)
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    File "/home/vinci/.espressif/python_env/idf5.2_py3.11_env/lib/python3.11/site-packages/cachecontrol/controller.py", line 155, in cached_request
      resp = self.serializer.loads(request, cache_data, body_file)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    File "/home/vinci/.espressif/python_env/idf5.2_py3.11_env/lib/python3.11/site-packages/cachecontrol/serialize.py", line 95, in loads
      return getattr(self, "_loads_v{}".format(ver))(request, data, body_file)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    File "/home/vinci/.espressif/python_env/idf5.2_py3.11_env/lib/python3.11/site-packages/cachecontrol/serialize.py", line 190, in _loads_v4
      return self.prepare_response(request, cached, body_file)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    File "/home/vinci/.espressif/python_env/idf5.2_py3.11_env/lib/python3.11/site-packages/cachecontrol/serialize.py", line 141, in prepare_response
      return HTTPResponse(body=body, preload_content=False, **cached["response"])
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

  TypeError: HTTPResponse.__init__() got an unexpected keyword argument
  'strict'

Call Stack (most recent call first):
  /home/vinci/esp/esp-idf/tools/cmake/project.cmake:464 (idf_build_process)
  CMakeLists.txt:22 (project)
```

I'm sorry that I have so little information, I'd like to describe the issue further, but I don't even know where to start... The TypeError at the end kinda looks like it has something to do with some Python HTTP package...?

### To Reproduce

```cpp
idf.py *
```

### Expected behaviour

.

### I have checked existing issues and online Documentation

- [X] I confirm I have checked existing issues and online Documentation.

kumekay, 2023-05-05:
@higaski Hello,
Thank you for the issue.
The problem is related to a breaking charge in underlying library. We already aware of this problem and working on the fix

kumekay, 2023-05-05:
@higaski 

You may try this:

In your ESP-IDF environment (after `source path/to/idf/export.sh`):

```
pip install -U "urllib3<2"
```

higaski, 2023-05-05:
Yep thanks again, this did the trick!

shenguoqing1, 2023-05-07:
How to solve it on Windows system

kumekay, 2023-05-08:
> How to solve it on Windows system

Hello @shenguoqing1 
On windows you may open `ESP-IDF PowerShell Environment` or `ESP-IDF Command Prompt (cmd.exe)` from the Start menu and run the same command:

```
pip install -U "urllib3<2"
```

In addition, if you are using ESP-IDF 5.0 or newer it is enough to re-run the install script (`install.ps1` on windows) and the ESP-IDF environment should be fixed.

arturdearaujo, 2023-05-17:
How to solve for VS Code on Windows?

whatdtech, 2023-05-21:
Navigate to esp-idf installation folder which you have selected during installation using vscode and execute `python -m pip install -U "urllib3<2"` and it works. For me installation folder is **python_env\idf5.2_py3.8_env\Scripts**

kumekay, 2023-05-21:
> How to solve for VS Code on Windows?

@arturdearaujo 
In VS Code, open command pallete with Ctrl+Shift+P and run terminal using `ESP-IDF: Open ESP-IDF terminal`. In terminal window run `pip install -U "urllib3<2"`

closed 2023-05-26 by kumekay