espressif/idf-component-manager #32: pyproject.toml contains unused dependencies.

fpgmaas, 2023-05-13:
### The Component Manager version

main

### ESP-IDF Version

NA

### python Version

3.10

### Operating System

macOS

### Browser (for https://components.espressif.com  Issues)

NA

### Description

It seems the `pyproject.toml` file contains some unused dependencies.

```shell
deptry . --extend-exclude 'integration_tests'
Scanning 82 files...

docs/en/conf.py:8:0: DEP001 'conf_common' imported but missing from the dependency definitions
idf_component_manager/utils.py:6:0: DEP003 'packaging' imported but it is a transitive dependency
idf_component_tools/file_cache.py:79:12: DEP001 '_winreg' imported but missing from the dependency definitions
idf_component_tools/file_cache.py:91:8: DEP001 'win32com' imported but missing from the dependency definitions
idf_component_tools/file_cache.py:108:20: DEP001 'win32api' imported but missing from the dependency definitions
idf_component_tools/file_cache.py:142:8: DEP001 'com' imported but missing from the dependency definitions
idf_component_tools/file_cache.py:143:8: DEP001 'com' imported but missing from the dependency definitions
idf_component_tools/file_cache.py:173:12: DEP001 'win32com' imported but missing from the dependency definitions
idf_component_tools/file_cache.py:183:20: DEP001 'com' imported but missing from the dependency definitions
idf_component_tools/manifest/constants.py:39:8: DEP001 'idf_py_actions' imported but missing from the dependency definitions
idf_component_tools/sources/git.py:21:4: DEP001 'urlparse' imported but missing from the dependency definitions
idf_component_tools/sources/web_service.py:29:4: DEP001 'urlparse' imported but missing from the dependency definitions
pyproject.toml: DEP002 'colorama' defined as a dependency but not used in the codebase
pyproject.toml: DEP002 'future' defined as a dependency but not used in the codebase
Found 14 dependency issues.

For more information, see the documentation: https://fpgmaas.github.io/deptry/
```

### To Reproduce

```cpp
poetry add --group dev deptry
deptry . --extend-exclude 'integration_tests'
```

### Expected behaviour

The pyproject.toml should not contain any unused dependencies and there should be no modules imported that are not in the project's dependencies.

### Additional info

I believe many of the issues can be resolved by:

- replacing `urlparse` with `urllib.parse`
- Adding `pywin32` as a dependency.
- Removing the dependency `colorama`
- Removing the dependency 'future'

Disclaimer: I am the author of deptry. I see this as nice way to contribute to other projects while simultaneously testing deptry on a variety of projects and gather some feedback.

### I have checked existing issues and online Documentation

- [X] I confirm I have checked existing issues and online Documentation.

kumekay, 2023-06-02:
Hello @fpgmaas

Thank you so much for making deptry and running it on our codebase!

We forgot to remove `future` and add `packaging` during migration to poetry, so the report is useful. To avoid it in the future we decided to add it to our pre-commit hooks and CI workflow.

While we use poetry for our development environment, we still need to support py2.7 for some time, so for package building, we still need `setup.py`. As a result, dependencies are listed twice, and it's easy to make a mistake.

kumekay, 2023-06-09:
Implemented in 6d0039dc9c8cb3ec146035bfee678366ab9db07d

closed 2023-06-09 by kumekay