espressif/idf-component-manager #33: Managed component examples are missing

SimonMerrett, 2023-05-24:
### The Component Manager version

 v1.2.2

### ESP-IDF Version

v5.02

### python Version

3.8.7

### Operating System

Windows 10

### Browser (for https://components.espressif.com  Issues)

Chrome

### Description

I have manged to follow the process to install a managed component in a vscode extension version of ESP-IDF. To start with, I opened the hello_world example project, opened a terminal and used `add-dependency` for `esp_modem` to create manifest. Then reconfigured the project and the managed component folder appeared in the project folder structure, with most of what I expected from esp_modem there. But none of the example projects from the esp_modem repository are there. 

Where are the example projects for esp_modem? Or how do I get them so that they can be used as the basis of a future project? Do they just need to be manually copy-pasted into a new project and then add esp_modem as a dependency to that project? 

The current documentation seems to be ambiguous as to whether it is addressing developers who want to get their components included in the registry and people who are trying to use the registry components. So far, the only mention of examples seems to be for component developers, rather than users. Does the esp_modem component repo examples folder need to be updated so that the component manager recognises its examples folder as containing examples (saw docs about cmake.txt needing to link to the examples folder to look for them?)?

Thanks for any clarity you can give me.

### To Reproduce

```cpp
See description
```

### Expected behaviour

I should be able to see examples when using esp_modem as an installed component from the component manager.

### I have checked existing issues and online Documentation

- [X] I confirm I have checked existing issues and online Documentation.

kumekay, 2023-05-25:
Hell @SimonMerrett thank you for your questions.

You can find example projects for esp modem on https://components.espressif.com/components/espressif/esp_modem. Check the right-hand menu.

To start a new project from an example you can run a command like:

```idf.py create-project-from-example "espressif/esp_modem^1.0.0:ap_to_pppos"```

You can find it for every example on the component page.

Initially, we made a decision to not include examples of the component downloaded by the component manager to the `managed_components` directory of the project. Because example projects are not required to build the project. About one month ago we reconsidered this question and now examples are not excluded from a component. However, we won't reprocess component versions uploaded before it, to make sure that the component hash sum won't change for users who downloaded it before.

So for the current version of the esp_modem the best way is to download them from the project page. After the next release of the esp_modem examples will be included in the downloaded component.

SimonMerrett, 2023-05-26:
@kumekay thanks so much for the explanation. It all makes sense now. I think that it could be made clearer in the readme that the registry website holds the commands to make examples. We are so used to thinking github is the centre of the universe that we don't expect to find stuff somewhere else! Even a short section like:

> The IDF Component manager is a tool that downloads dependencies for any [ESP-IDF](https://www.espressif.com/en/products/sdks/esp-idf) CMake project. It makes sure that the right versions of all components required for a successful build of your project are in place. The download happens automatically during a run of CMake. It can source components either from [the component registry](https://components.espressif.com/) or from a git repository. _**If you want to create a project from a component example, check out the commands on the right hand side of a component's registry page**_

closed 2023-05-26 by SimonMerrett

kumekay, 2023-05-26:
@SimonMerrett Sure, we are planning a noticeable documentation update and will add more information for users of components.