espressif/idf-component-manager #34: Check in poetry.lock into source control

corpix, 2023-05-28:
### Related area

git

### Is your feature request related to a problem?

There is no lock file in source control. This makes packaging this tool difficult (I do update this environment https://github.com/mirrexagon/nixpkgs-esp-dev)

### Describe the solution you'd like

I see in `.gitignore` says `poetry.lock` was removed because of tests. Maybe this could be done during test environment preparation?

### I have checked existing list of Feature requests and I have provided enough information.

- [X] I confirm I have checked existing list of Feature requests and I have provided enough information.

corpix, 2023-06-26:
Thanks for the reply!
I think I could, will try, thank you for the explanation!

kumekay, 2024-04-22:
@corpix In a coming release of 2.0.0 we will remove setup.py, but still won't add a lock file. What is the main problem of packaging without a lock file?

kumekay, 2024-05-20:
@corpix I would like to close this issue now, however you are welcome to reopen if you want to discuss this topic further.

closed 2024-05-20 by kumekay