espressif/idf-component-manager #37: Stuck at Building Process Due to IDF Component Dependency

alibukharai, 2023-06-20:
### The Component Manager version

1.2.3

### ESP-IDF Version

ESP-IDF v5.0.2-376-g24b9d38a24

### python Version

Python 3.9.12

### Operating System

Description:	Ubuntu 22.04.2 LTS  Release:	22.04

### Browser (for https://components.espressif.com  Issues)

Chrome

### Description

The current version of ESP-IDF (5.0) is causing issues with building the [ESP-Box](https://github.com/espressif/esp-box/tree/master/examples/factory_demo) examples. The building process gets stuck and does not progress further. 

### To Reproduce

```cpp
just simply build factory demo in [ESP-Box](https://github.com/espressif/esp-box/tree/master/examples/factory_demo) repository.
```

### Expected behaviour

(base) ➜  factory_demo git:(feature/chat_gpt) idf.py --version                          
/home/ali/esp/esp-idf_5.0/tools/check_python_dependencies.py:12: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  import pkg_resources
ESP-IDF v5.0.2-376-g24b9d38a24
(base) ➜  factory_demo git:(feature/chat_gpt) idf.py build flash monitor
/home/ali/esp/esp-idf_5.0/tools/check_python_dependencies.py:12: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  import pkg_resources
Executing action: all (aliases: build)
Running cmake in directory /home/ali/work/learning/esp-box/examples/factory_demo/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DIDF_TARGET=esp32s3 -DCCACHE_ENABLE=0 /home/ali/work/learning/esp-box/examples/factory_demo"...
-- Found Git: /usr/bin/git (found version "2.34.1") 
-- Component directory /home/ali/esp/esp-idf_5.0/components/esp_coex does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/ali/esp/esp-idf_5.0/components/tinyusb does not contain a CMakeLists.txt file. No component will be added
-- The C compiler identification is GNU 11.2.0
-- The CXX compiler identification is GNU 11.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/ali/.espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /home/ali/.espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /home/ali/.espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32s3
Solving dependencies requirements

> No logs below this

### I have checked existing issues and online Documentation

- [X] I confirm I have checked existing issues and online Documentation.

alibukharai, 2023-06-20:
This works for me
1- Deleting this folder .espressif/python_env/idf5.0_py (your version)
2- And reinstalling IDF ./install.sh

kumekay, 2023-06-20:
@alibukharai 

This issue is probably related to CacheControl library, but it's hard to reproduce it.

If you still have an environment with the problem, could you please try this:

1. run `pip list` inside ESP-IDF environment. 
2. Check content of `~/.cache/Espressif/ComponentManager/.api_client` 
3. Try to drop this directory first before doing anything else

kumekay, 2023-06-30:
@alibukharai Fixed in component manager v1.3.0

closed 2023-06-30 by kumekay