espressif/idf-component-manager #38: Manifest File version field broken

higaski, 2023-07-04:
### The Component Manager version

v1.3.1

### ESP-IDF Version

ESP-IDF v5.2-dev-1349-gcf7e743a9b

### python Version

Python 3.11.3

### Operating System

Arch

### Description

It is no longer possible to use the version field when referencing git branches or tags.

### To Reproduce

```cpp
Add the following dependency to any project:

  dcc:
    version: v0.27.1 # branches don't work either
    git: https://github.com/ZIMO-Elektronik/DCC.git
```

### Expected behaviour

The component manager should check out the specified tag. Valid tags **may** contain prefixes (e.g. v1.0.0).

### I have checked existing issues and online Documentation

- [X] I confirm I have checked existing issues and online Documentation.

higaski, 2023-07-04:
I just realized that ESP-IDF always comes with the latest version of the component manager... this is bad.
How can I downgrade to whatever version branches and tags are still working?

hfudev, 2023-07-05:
Hi @higaski I can reproduce this issue locally. We're preparing for a 1.3.2 bugfix version very soon. Sorry for the inconvenience.

> How can I downgrade to whatever version branches and tags are still working?

Starting from ESP-IDF 5.0, the `install.sh` will install the latest versions that match our internal maintained constraint file for all python packages. The `install.sh` will first download the branch corresponding constraint file (for example, the current master one) from https://dl.espressif.com/dl/esp-idf/espidf.constraints.v5.2.txt and place it under `~/.espressif`, then pass this constraint file while doing a pip install for each required dependency.

The constraint version spec (2023-07-05) for `idf-component-manager` is `~=1.2`. So you may also install the `1.2.3` version with `pip install "idf-component-manager==1.2.3`. That will work as a temporary workaround.

commit 252cb109 (2023-07-05): fix: git source dependencies with `version` field work again

closes #38
closed 2023-07-05 by espressif-bot