espressif/idf-component-manager #39: Component manager checks out non-existing SHA

higaski, 2023-07-09:
### The Component Manager version

IDF component manager v1.3.2

### ESP-IDF Version

ESP-IDF v5.2-dev-1113-g28c643a56d

### python Version

Python 3.11.3

### Operating System

Arch

### Description

The component manager seems to can get "stuck" on some specific SHA hash when trying to checkout a git branch. I've tried clearing the component managers cache but that didn't solve the issue. Neither did adding another commit to the dependency...

```sh
-- Building ESP-IDF components for target esp32s3
Processing 7 dependencies:
[1/7] bblanchon/arduinojson (6.21.2)
[2/7] dcc (f5114f028552f6128ac4a8baa86ce395bbbd8b93)
[3/7] espressif/esp_tinyusb (1.3.0)
[4/7] espressif/mdns (1.2.0)
[5/7] espressif/tinyusb (0.14.3)
[6/7] idf (5.2.0)
[7/7] mdu (1b601b8dd00ec7d8a409b41bd2b2ec35dbc3785e)  // <- This ain't an existing sha
CMake Error at /home/vinci/esp/esp-idf/tools/cmake/build.cmake:544 (message):
  WARNING: From https://github.com/ZIMO-Elektronik/MDU

   + 01815ef...a637011 packet     -> packet  (forced update)

  

  ERROR: Branch "1b601b8dd00ec7d8a409b41bd2b2ec35dbc3785e" doesn't exist in
  repo "https://github.com/ZIMO-Elektronik/MDU.git"

Call Stack (most recent call first):
  /home/vinci/esp/esp-idf/tools/cmake/project.cmake:547 (idf_build_process)
  CMakeLists.txt:21 (project)
```

### To Reproduce

```cpp
On my system it's currently enough to add the following dependency:

mdu:
  version: packet
  git: https://github.com/ZIMO-Elektronik/MDU.git
```

### Expected behaviour

Check out the latest commit when specifying a git branch.

### I have checked existing issues and online Documentation

- [X] I confirm I have checked existing issues and online Documentation.

higaski, 2023-07-09:
The issue disappeared after some time... not sure why though. Could this be a GitHub API thing?

higaski, 2023-07-11:
I have force pushed this commit once yes. So https://github.com/ZIMO-Elektronik/MDU/commit/1b601b8dd00ec7d8a409b41bd2b2ec35dbc3785e must refer to the old version which got overwritten. The question is though, why does the component manager keep referring to it?

kumekay, 2023-07-11:
It's intentional behavior. The goal of the component manager to make reproducible builds possible. For git sourced components it stores the commit id in dependencies lock, to be sure the same commit will be fetched next time. 

So it will only try to update once dependencies.lock is removed or `idf.py update-dependencies                                                                                                                                                ` is executed.

We will document this behavior.

higaski, 2023-07-11:
Oh ok thanks for clearing that up. Personally I find that a bit unintuitive, specially since this is not what CMake does.

E.g. if I would like to add a dependency to an [fmt](https://github.com/fmtlib/fmt) branch called `text` I could use something along the lines of
```cmake
FetchContent_Declare(
  fmt
  GIT_REPOSITORY https://github.com/fmtlib/fmt.git
  GIT_TAG text)
```

This would get me the latest commit from this branch, always. If I would be interested in checking out a specific commit I'd explicitly state that by using a tag, or, in case the maintainers of the dependency wouldn't care about versioning at all, a specific hash.

I understand that you want to give users the best possible and simplest experience, but all these "implicit" assumptions that the whole ESP-IDF framework now makes on how it should be used (e.g. how component dependencies are handled) feel like footguns.