espressif/idf-component-manager #41: Build with component manager fails because the API address cannot be resolved (IDFGH-10883)

M-Bab, 2023-08-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.4

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

I expected to be able to build an idf project with idf.py and added managed components.

### What is the actual behavior?

On some systems the build fails because the domain name of `api.components.espressif.com` cannot be resolved. I have actually no idea why this only fails on our CI machines (see log below) because the address can not be resolved anywhere:

```
> drill -Q api.components.espressif.com
> drill -Q api.components.espressif.com @8.8.8.8
> drill -Q api.components.espressif.com @9.9.9.9
> drill -Q components.espressif.com
16.162.213.162
18.166.114.217
> drill -Q components.espressif.com @8.8.8.8
16.162.213.162
18.166.114.217
> drill -Q components.espressif.com @9.9.9.9
16.162.213.162
18.166.114.217
```

Unlike `api.components.espressif.com` the parent address `components.espressif.com` can be resolved.

Our CI docker works with ESP-IDF version v4.4.3 locally I work with v4.4.4 and I am aware that v4.4.5 is available. But neither should this have a large impact on the managed components nor did I find any note about changes there in the release notes.

### Steps to reproduce.

1. Try to build a project with idf.py and managed components enabled. But the managed components are not available locally or cached.

### Build or installation Logs.

```plain
$ echo Building APP
Building APP
$ idf.py build
-- Found Git: /usr/bin/git (found version "2.25.1") 
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/builder/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /home/builder/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /home/builder/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
[... 92 lines cut ...]
    File "/home/builder/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages/idf_component_manager/version_solver/version_solver.py", line 29, in solve
      self.solve_manifest(manifest)
    File "/home/builder/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages/idf_component_manager/version_solver/version_solver.py", line 48, in solve_manifest
      self.solve_component(requirement)
    File "/home/builder/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages/idf_component_manager/version_solver/version_solver.py", line 51, in solve_component
      cmp_with_versions = requirement.source.versions(
    File "/home/builder/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages/idf_component_tools/sources/web_service.py", line 95, in versions
      cmp_with_versions = self.api_client.versions(name, spec, target)
    File "/home/builder/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages/idf_component_tools/api_client.py", line 196, in versions
      body = self._base_request(
    File "/home/builder/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages/idf_component_tools/api_client.py", line 178, in _base_request
      raise APIClientError('HTTP request error')
  idf_component_tools.api_client_errors.APIClientError: HTTP request error
Call Stack (most recent call first):
  /home/builder/esp/esp-idf/tools/cmake/project.cmake:384 (idf_build_process)
  CMakeLists.txt:6 (project)
-- Configuring incomplete, errors occurred!
See also "/builds/hf/gui-embedded/build/CMakeFiles/CMakeOutput.log".
Executing action: all (aliases: build)
Running cmake in directory /builds/hf/gui-embedded/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DIDF_TARGET=esp32 -DCCACHE_ENABLE=0 /builds/hf/gui-embedded"...
cmake failed with exit code 1
Cleaning up project directory and file based variables 00:01
ERROR: Job failed: exit code 1
```

kumekay, 2023-08-16:
@M-Bab Thank you for opening the issue, it was caused by a mistake in DNS configuration,
It should be fixed now.

```
$ nslookup api.components.espressif.com                                                                                                                                         
Server:		8.8.8.8
Address:	8.8.8.8#53

Non-authoritative answer:
Name:	api.components.espressif.com
Address: 18.166.114.217
Name:	api.components.espressif.com
Address: 16.162.213.162
```

closed 2023-08-16 by kumekay

M-Bab, 2023-08-21:
Thanks for the fast reaction! Works well again.