espressif/idf-component-manager #43: Managed Components not part of project in separate directory?

gojimmypi, 2023-10-12:
### Related area

component directory

### Is your feature request related to a problem?

This is similar to https://github.com/espressif/idf-component-manager/issues/13 and https://github.com/espressif/idf-component-manager/issues/40 but I would like to be able to manage components in a shared project directory.

### Describe the solution you'd like

I'd like to be able to `idf.py add-dependency` or perhaps have new `idf.py install-component` into a directory that is not part of a project, but is otherwise shared by possibly multiple projects or examples.

I didn't see anything mentioned in the [IDF Component Manager Documentation](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/tools/idf-component-manager.html). 

Note I was able to use `add-dependency`, but needed to create a hacky `main` directory:

```
gojimmypi:/mnt/c/homekit/esp32-homekit-demo
$ idf.py add-dependency "wolfssl/wolfssl^5.6.0-stable"
/mnt/c/SysGCC/esp32/esp-idf/v5.1/tools/check_python_dependencies.py:12: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  import pkg_resources
Executing action: add-dependency
ERROR: Directory "/mnt/c/homekit/esp32-homekit-demo/main" does not exist! Please specify a valid component under /mnt/c/homekit/esp32-homekit-demo or try to use --path
gojimmypi:/mnt/c/homekit/esp32-homekit-demo
$ mkdir main
gojimmypi:/mnt/c/homekit/esp32-homekit-demo
$ idf.py add-dependency "wolfssl/wolfssl^5.6.0-stable"
/mnt/c/SysGCC/esp32/esp-idf/v5.1/tools/check_python_dependencies.py:12: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  import pkg_resources
Executing action: add-dependency
Created "/mnt/c/homekit/esp32-homekit-demo/main/idf_component.yml"
Successfully added dependency "wolfssl/wolfssl^5.6.0-stable" to component "main"
Done
```

Any guidance as to Best Practices would also be appreciated.

### Describe alternatives you've considered

Install the component to each example. 

Install the component to the `HomeKit` submodule. (see below)

### Additional context

This is specifically in reference to @AchimPieters using [wolfSSL](https://components.espressif.com/components/wolfssl/wolfssl) as a managed component in the [esp32-homekit](https://github.com/AchimPieters/esp32-homekit-demo) project.

In particular, note there's a shared solution [`components` directory](https://github.com/AchimPieters/esp32-homekit-demo/tree/main/components) that's used by the [examples](https://github.com/AchimPieters/esp32-homekit-demo/tree/main/examples).

I have somewhat of a workaround as described in https://github.com/wolfSSL/wolfssl/issues/5909#issuecomment-1758318130. However it's not a very graceful solution in that I copied the managed component source into a [new](https://github.com/gojimmypi/esp32-homekit-demo/tree/main/components), non-managed component directory.

### I have checked existing list of Feature requests and I have provided enough information.

- [X] I confirm I have checked existing list of Feature requests and I have provided enough information.

kumekay, 2023-10-12:
Hello @gojimmypi 
Thank you for raising the topic,
The proposal of `install-component` sounds sensible, but it's very hard to implement because components are not simply downloaded and unpacked. As the very first step the full tree of dependencies is built. For this the context of the project (IDF version, current target) is required to choose the right versions of all dependencies. Even if you add only one dependency it may have intricate web of sub-dependencies. 

At the same, what is stopping you from adding the dependency from the component registry to all the examples?

gojimmypi, 2023-10-12:
Hi @kumekay -

Thanks for your feedback!

> it's very hard to implement because components are not simply downloaded and unpacked

hm, yes, I suspected as much.

> what is stopping you from adding the dependency from the component registry to all the examples 

Ah, a good question! What I neglected to explain in enough detail here (but explained more in the long thread of https://github.com/wolfSSL/wolfssl/issues/5909), is that in the case of the `esp32-homekit` example, there's some intricacy of where the wolfSSL settings are defined (i.e. the [CMakeLists.txt](https://github.com/AchimPieters/esp32-homekit/blob/3b5c3a4eb1217a2a5c6ac82710aae42e411ef89f/CMakeLists.txt#L11 ) for the `esp32-homekit`  vs the managed component [user_settings.h](https://github.com/AchimPieters/esp32-homekit-demo/blob/59a6252462e02ef5a4b4be5cc68f7083fd222d3f/examples/led/managed_components/wolfssl__wolfssl/include/user_settings.h#L48) in the example code).

I'm thinking in the end, the best place to add the managed component in this example is the [esp32-homekit](https://github.com/AchimPieters/esp32-homekit) submodule, as that's where the `crypto.c` [uses wolfssl](https://github.com/AchimPieters/esp32-homekit/blob/afba67e7df077dbc6eba59bd58cd5d13b4df0794/src/crypto.c#L5). Do you agree? Those [examples](https://github.com/AchimPieters/esp32-homekit-demo/tree/main/examples) don't even use wolfSSL directly.

Somewhat related to this topic is the desire / ability to not only _use_ a managed component, but also be able to _configure_ it. For example, today if the wolfSSL component is used as a managed component, I believe the `user_settings.h` is then locked in place with regards to the [.component_hash](https://github.com/AchimPieters/esp32-homekit-demo/blob/main/examples/led/managed_components/wolfssl__wolfssl/.component_hash) file. Once the `user_settings.h` is changed, the hash no longer matches, and there's a resulting `idf.py` warning / suggestion to move the code to a non-managed component directory. What are your thoughts on this topic? Do you see managed components as only a _getting started_ capability?

igrr, 2023-10-12:
Regarding user_settings.h: in IDF we generally follow the approach that users should not need to modify source or header files. Where compile-time configuration is necessary, it is achieved using Kconfig.

It is also possible to let the wolfssl component find user_settings.h in another location — for example, in the build directory. The user's project can then use CMake configure_file command to generate the user_settings.h file at build time.

However, I would generally suggest creating a Kconfig file for configuration options. Here's an example how this is done with MbedTLS:
- https://github.com/espressif/esp-idf/blob/master/components/mbedtls/port/include/mbedtls/esp_config.h — the equivalent of user_settings.h, part of the component, not intended to be modified.
- Kconfig options are defined here: https://github.com/espressif/esp-idf/blob/master/components/mbedtls/Kconfig

-

gojimmypi, 2023-11-06:
We were able to resolve the ESP32 HomeKit issues (it's really quite cool, and I created a little [blog](https://www.wolfssl.com/secure-your-apple-homekit-espressif-esp32-devices-with-wolfssl/) on the topic).

Given the [comment](https://github.com/espressif/idf-component-manager/issues/43#issuecomment-1759735922) from @kumekay regarding the complexity of implementing `install-component`, I'll close this issue. 

I plan to follow @igrr 's suggestion and use the kconfig for wolfSSL `user_settings.h` configuration. I'll look into if I can use the file from another directory outside of the component structure.

Thank you.

closed 2023-11-06 by gojimmypi