espressif/idf-component-manager #45: Unable to incorporate a component with an override_path via Git

leeebo, 2023-12-18:
### The Component Manager version

v1.4.1

### ESP-IDF Version

master (5.3)

### python Version

3.10

### Operating System

Ubuntu

### Description

```bash
CMake Error at /home/xuxin/esp/esp-idf-github-clean/tools/cmake/build.cmake:544 (message):
ERROR: The 'override_path' field in the manifest file
../rmaker common/idf component.yml' does not point to a directory. You
can safely remove this field from the manifest if this proiect is an
example copied from a component repository.
The dependency will be
downloaded from the Esp component registry. Documentation:
https://docs.espressif.com/projects/idf-component-manager/en/latest/reference/manifest file.html #foverride-path
```

### To Reproduce

```cpp
espressif/esp_schedule:
  path:components/esp_schedule
  git:https://github.com/espressif/esp-rainmaker.git
  version:"7d4c1e36f9110e04c30c21a708762ed91eea99ac"
```

### Expected behaviour

Component manager Handle/Delete the `override_path` in the yml

### I have checked existing issues and online Documentation

- [X] I confirm I have checked existing issues and online Documentation.

kumekay, 2023-12-18:
Hi @leeebo 

Thank you for opening the topic,
Your use case is not the one we considered for the feature. I.e., the expected flow is to make a clone of the entire Rainmaker repository manually and then use a directory as a local dependency or using `EXTRA_COMPONENT_DIRS`

We can, however, support this case, by ignoring `override_path` for git components, but in this case, it will download `rainmaker_common` from the registry. We can also print a warning when it happens. What do you think?

But unfortunately, we cannot make `override_path` just work because here component manager operates with git dependencies at the level of individual components, not a whole repo.

leeebo, 2023-12-18:
Hi @kumekay, we need the workaround because there is a use case: the fix already be merged, but not be released to `esp-registry`.

How about just ignore the `override_path`?  then print message tell which version it actually using. Prompt warning is not recommended, because CI Build processes Do Not allow warnings in the CMake process also.

kumekay, 2023-12-18:
@leeebo Ok, I'll check how we can quickly implement it

leeebo, 2023-12-19:
> what is the problem making a release?

This is up to the maintainer of the component, and we have no control over it.

leeebo, 2023-12-19:
@igrr While this leads to another topic, **What is the recommended component release workflow to follow?** 🤔 

If a repository only has one component, we can maintain different major or minor versions with different branches, but if a component is only part of a larger repository, maintaining versions becomes more complicated.