espressif/idf-component-manager #46: Component manager deletes required dependencies

Suxsem, 2023-12-23:
### The Component Manager version

1.4.1

### ESP-IDF Version

5.1.1

### python Version

3.11.4

### Operating System

Windows 10

### Description

I think there is a bug in how download_project_dependencies works. https://github.com/espressif/idf-component-manager/blob/2711ccc08bfd0db60c4dbd15bbdd72a5e0f5b3fa/idf_component_manager/dependencies.py#L258.

It should scan project requirements (idf_component.yml), build the dependencies tree and add download both project requirements and their dependencies.

Problem is, that only packages that are NOT in project requirements (... noi in ... and ... not in ..., see below) are actually considered for download.

https://github.com/espressif/idf-component-manager/blob/2711ccc08bfd0db60c4dbd15bbdd72a5e0f5b3fa/idf_component_manager/dependencies.py#L312-L316

Result is that I specify two dependencies in my idf_component.yml file but they are removed from the managed component folder because considered "unused".

### To Reproduce

```cpp
Create a idf_component.yml file:

dependencies:
  esp-zboss-lib: "^1.0.7"
  esp-zigbee-lib: "^1.0.7"
  idf: ">=5.0"
```

Run make menuconfig (or other cmake commands), this is the wrong output:

```
Deleting 2 unused components

 espressif__esp-zboss-lib

 espressif__esp-zigbee-lib
```

Processing 1 dependencies:

[1/1] idf (5.1.1)

Content of dependencies.lock file is this one:

```
dependencies:
  espressif__esp-zboss-lib:
    component_hash: null
    source:
      path: C:\Users\semer\Documents\PlatformIO\Projects\nfc-alarm-panel\managed_components\espressif__esp-zboss-lib
      type: local
    version: 1.0.7
  espressif__esp-zigbee-lib:
    component_hash: null
    source:
      path: C:\Users\semer\Documents\PlatformIO\Projects\nfc-alarm-panel\managed_components\espressif__esp-zigbee-lib
      type: local
    version: 1.0.7
  idf:
    component_hash: null
    source:
      type: idf
    version: 5.1.1
manifest_hash: 89bfe204db7dbbb4b6e85994425215a0d70c45f775606ba52a1aea9f595c6bac
target: esp32c6
version: 1.0.0
```
```

### Expected behaviour

Project requirements should be downloaded and kept.

Fix: requirement_dependencies list should be added to the project_requirements_dependencies one.

### I have checked existing issues and online Documentation

- [X] I confirm I have checked existing issues and online Documentation.

Suxsem, 2023-12-23:
I just saw that this commit (https://github.com/espressif/idf-component-manager/commit/91058b8a36d11f42e9f68dacd1f2095e145e5c18) could have fix the issue, but it's not released yet, let's me try it from master

Suxsem, 2023-12-23:
Nope...same issue

kumekay, 2024-01-03:
@Suxsem thanks for being patient, we were slow to respond during holidays.

I wonder why you dependencies.lock show them as local dependencies, I would expect something like:

```
dependencies:
  espressif/esp-zboss-lib:
    component_hash: 9dc1059b8a6695580ab0324d41f2f88e324c1486b196e20d087f3dcb78c1c654
    source:
      service_url: https://api.components.espressif.com/
      type: service
    version: 1.0.7
  espressif/esp-zigbee-lib:
    component_hash: 3171496afcbdf5c370a3f74f6bc6a7e3196227438b0405d8714796c9d6f68a36
    source:
      service_url: https://api.components.espressif.com/
      type: service
    version: 1.0.7
```

To help me to reproduce this, please share your project's and main component CMakeLists.txt files.
Do you have `set(COMPONENTS main)` in your CMakeLists.txt?

kumekay, 2024-02-15:
@Suxsem Did you manage to solve the issue? If not, could you please help us to reproduce it?

kumekay, 2024-02-28:
@Suxsem I believe this issue should be addressed in the component manager v1.5.1, so I'm closing this know. If you still experience the problem please don't hesitate reopening the issue.

closed 2024-02-28 by kumekay