espressif/idf-component-manager #47: compote: you are not allowed to perform this action (IDFGH-11810)

dannybackx, 2023-12-31:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I followed all the docs but I can't upload an initial version of my component.
No flags to trigger verbosity, no log files, no luck googling this...

Please help.

See svn+ssh://dannybackx@svn.code.sf.net/p/espdyndns/svncode

  Danny

victus: {726} compote component upload --service-profile=staging --name dydns
Selected profile "staging" from the idf_component_manager.yml config file
Saving archive to "/home/danny/src/esp-idf/dyndns/dist/dydns_0.0.1.tgz"
Uploading archive /home/danny/src/esp-idf/dyndns/dist/dydns_0.0.1.tgz
100%|██████████████████████████████████████████████████████████████████████████████| 7.20k/7.20k [00:02<00:00, 2.71kB/s]
ERROR: You are not allowed to perform this action.
victus: {727} compote version
1.4.1

dannybackx, 2023-12-31:
Yes indeed, --namespace <<my-namespace>> did the trick.
But as I said, no logging, error message not very helpful, ..

mickeyl, 2024-03-08:
I have a similar problem, however I'm specifying the (empty) namespace:

```sh
compote component upload --namespace "" --name esp-microsleep
Saving archive to "/home/mickey/Dokumente/late/esp-microsleep/dist/esp-microsleep_1.0.0.tgz"
Uploading archive /home/mickey/Dokumente/late/esp-microsleep/dist/esp-microsleep_1.0.0.tgz
ERROR: You are not allowed to perform this action.
```

kumekay, 2024-12-06:
Now errors say explicitly which permissions are missing.

closed 2024-12-06 by kumekay