espressif/idf-component-manager #48: Custom namespaces - when (or workaround)

higaski, 2024-01-29:
### Related area

custom namespaces

### Is your feature request related to a problem?

I'm owning an organization with my private GitHub account and I'd like to upload a component under that organizations namespace. Currently this does not seem to be possible as the `--namespace` option is not yet implemented? The [component registry](https://components.espressif.com) supports login via GitHub, however I can't switch from my private user to the organization.

Is there a workaround for this issue at the moment?

### Describe the solution you'd like

An implementation (or time frame of the implementation) of the `--namespace [YOUR_NAMESPACE]` option.

### I have checked existing list of Feature requests and I have provided enough information.

- [X] I confirm I have checked existing list of Feature requests and I have provided enough information.

kumekay, 2024-02-01:
Hello @higaski 
`--namespace` option is supported, but for now, we are creating new namespaces manually. Please contact us at components@espressif.com
(There is a note on the profile page about it)

higaski, 2024-02-01:
I've done so a week ago... When can I expect an answer?

kumekay, 2024-02-02:
@higaski I apologize, we usually answer in one day, we replied to you

closed 2024-02-02 by kumekay