espressif/idf-component-manager #51: Component manager can't use GIT version with leading 'v'

higaski, 2024-02-06:
### The Component Manager version

v1.4.2

### ESP-IDF Version

v5.2-beta1

### python Version

3.11.6

### Operating System

Linux

### Description

The `--version git` option does not work for GIT tags containing a leading `v` or so it seems.

I have a component with the following manifest file.

```yml
description: MDU protocol for ZPP and ZSU updates
url: http://zimo.at
documentation: https://github.com/ZIMO-Elektronik/MDU/blob/master/README.md
repository: https://github.com/ZIMO-Elektronik/MDU.git
issues: https://github.com/ZIMO-Elektronik/MDU/issues

dependencies:
  idf: ">=5.0.0"
```

The latest tag is v0.15.0

Uploading this component with 
```sh
compote component upload --namespace zimo-elektronik  --name MDU --version git --dry-run
```
does not work.

Using any tag without leading `v` does not give me any errors.
```sh
compote component upload --namespace zimo-elektronik  --name MDU --version 1.2.3 --dry-run
```

For some weird reason 

### To Reproduce

```cpp
.
```

### Expected behaviour

GIT version should be parsed correctly according to https://semver.org/

### I have checked existing issues and online Documentation

- [X] I confirm I have checked existing issues and online Documentation.

kumekay, 2024-02-06:
@higaski Thank you for opening the issue, 
I was trying to reproduce it, and while I didn't manage to do so, I found the related problem.
The CLI command only works with annotated tags, but entirely ignores lightweight ones. Is it possible that it is the issue you faced?

At the same time, the https://github.com/espressif/upload-components-ci-action is not affected and should work with any tags.

kumekay, 2024-02-06:
@higaski There is a difference how git traits tags created with `git tag v1.2.3` and `git tag v1.2.3 -m "My tag"`. The issue is the current version of the component only supports the latter ones. I'm just guessing, but likely it's the problem you were facing. I already made an internal pull request with the fix, it is likely to be published begging of the next week.

higaski, 2024-02-06:
I don't think that this is the issue. I only create annotated tags.
In fact I've only recently learned about lightweight tags because of some [obscure issue with GitHub actions](https://github.com/orgs/community/discussions/48693#discussioncomment-6132963)...

kumekay, 2024-02-28:
Hello @higaski. Could you please verify if you still have this problem with the component manager 1.5.1? 

While we only fixed the support for light-weight tags, we also improved logging, so even if the  problem is still there, we can get a better idea of what is going on.

kumekay, 2024-04-22:
@higaski I'm closing this issue now, if you face it once again, please feel free to reopen.

closed 2024-04-22 by kumekay