espressif/idf-component-manager #58: Random order of external component configs in sdkconfig

ConnorDuvalS, 2024-04-05:
### The Component Manager version

v1.5.2

### ESP-IDF Version

v5.1.2

### python Version

3.8.16

### Operating System

Mac OS Sonoma 14.4.1

### Description

This is detailed further in [my comment](https://github.com/espressif/esp-idf/issues/10419#issuecomment-2035833444) on https://github.com/espressif/esp-idf/issues/10419 but the component configs in the SDKConfig will be in a random order on `idf.py reconfigure` and on `idf.py build` which will result in the reconfiguration happening over and over. This may be a problem with the framework but, given the original fix from that framework issue features a change here, it is likely an issue with this Python module.

### To Reproduce

```cpp
1. Initialise new, empty IDF project with a `main/` component, a `CMakeLists.txt` file, and a .gitignore ignoring the `build/` and `managed_components/` directories.
2. `idf.py add-dependency espressif/esp-modbus=1.0.5`
3. `idf.py add-dependency espressif/mdns=1.0.7`
4. `idf.py reconfigure`
5. `git init && git add . && git commit -m "Initial commit"`
6. `idf.py reconfigure`
7. `git status`

The `sdkconfig` file changes at random so it may not happen the first time but running it a few times should produce:

Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git restore <file>..." to discard changes in working directory)
        modified:   sdkconfig
```
```

### Expected behaviour

`git status` should return "nothing to commit, working tree clean" always where it returns "modified: sdkconfig". The sdkconfig file should not change if the component manifest does not change.

### I have checked existing issues and online Documentation

- [X] I confirm I have checked existing issues and online Documentation.

dobairoland, 2024-04-17:
Linking https://github.com/espressif/esp-idf/issues/13490 as those comments might be also relevant to this.

kumekay, 2024-04-22:
@ConnorDuvalS It looks like it's a regression in a recent version, we will make a bugfix release soon

PerSmitt, 2024-05-23:
@kumekay will this bugfix be released for latest only or will it be backported to IDF v5.0.x?

hfudev, 2024-07-10:
This fix has been released with 1.5.3.

closed 2024-07-10 by hfudev