espressif/idf-component-manager #59: Remove a managed component (IDFGH-12644)

anujdeshpande, 2024-04-18:
### Is your feature request related to a problem?

Once I add a managed component, I can't find a method to remove it from my app.

### Describe the solution you'd like.

idf.py remove-dependency <name>

### Describe alternatives you've considered.

Manually deleting - but it comes back when I do a build.

### Additional context.

ESP-IDF v5.1.3

youngguyl, 2024-08-08:
Hi, @anujdeshpande 
How do you solve it? I have the same problem, I use `idf.py add-dependency "espressif/button^3.2.0"` in one project before. 
No matter I changed to any other projects, will download this components automatic while build. I want to know how to remove it. 
Thanks.

kumekay, 2024-08-09:
@youngguyl There should be a list of dependencies in `main/idf_component.yml` inside your project, just remove the line from the file.

youngguyl, 2024-08-09:
@kumekay thanks for your quick reply. 
I checked my main folder, I deleted the idf_component.yml and `dependencies.lock`, `sdkconfig`, `build` and `managed components`folder as well, if I try to build the project, the component will be downloaded automaticly. the problem with gt911 component. 

I tried to create a project by idf.py create-project command. still download the lcd and gt911 components. 

I deleted the folder `esp_idf/component/esp_lcd` and it works now.

baldhead69, 2024-10-07:
> Yeah I figured that part with grep - but it would be simpler to remove a dependency using the same way I added it. I don't think this is a bug - so marked it as a feature request.

Yes. I would like that feature too.

Brian-at-Baebies, 2025-05-13:
+1.  Also it would be handy if the "ESP Component Registry" part of the ESP-IDF Visual Studio Code extensions could present a list of installed dependencies with "Uninstall" buttons.  That might be a separate feature request, though..

XDanielPaul, 2025-06-13:
@baldhead69 

> If you wish to update the version of your LVGL dependency, you can try running idf.py update-dependencies

Yes, I agree! As mentioned earlier, we already have the `idf.py update-dependencies` command built-in, and at this time, we do not plan to support updating individual dependencies.

Regarding the command for removing a dependency, I am currently working on `idf.py remove-dependency DEPENDENCY`