espressif/idf-component-manager #61: `- if: "idf_version >=4.4"` does not work in version 1.5.2

tswen, 2024-04-28:
### The Component Manager version

1.5.2

### ESP-IDF Version

release/v5.1

### python Version

3.10.14

### Operating System

ubuntu

### Description

`- if: "idf_version >=4.4"` does not work in version 1.5.2

### To Reproduce

```cpp
dependencies:
  idf: ">=4.3"
  espressif/esp_modem_usb_dte:
    version: 1.1.0~1
    rules:
      - if: "idf_version >=4.4"
```

### Expected behaviour

```
.......HINT: Component "espressif/usb_host_cdc_acm" has suitable versions for other targets: "esp32s3", "esp32p4", "esp32s2". Is your current target "esp32" set correctly?
.......NOTICE: Skipping optional dependency: usb_device
CMake Error at /home/tiansenwen/idf/esp-idf4.3/tools/cmake/build.cmake:433 (message):
  WARNING: Component "espressif/usb_host_cdc_acm" not found

  WARNING: Component "idf" not found

  ERROR: Because espressif/esp_modem_usb_dte (1.1.0~1) depends on idf
  (>=4.4.0) which doesn't match any versions, espressif/esp_modem_usb_dte is
  forbidden.

  And because espressif/iot_bridge (0.11.4) depends on
  espressif/esp_modem_usb_dte (1.1.0~1), espressif/iot_bridge is forbidden.

  So, because no versions of espressif/iot_bridge match >=0.0.0,<0.11.4 ||
  >0.11.4

   and project depends on espressif/iot_bridge (*), version solving failed.

Call Stack (most recent call first):
  /home/tiansenwen/idf/esp-idf4.3/tools/cmake/project.cmake:454 (idf_build_process)
  CMakeLists.txt:6 (project)

-- Configuring incomplete, errors occurred!

```

### I have checked existing issues and online Documentation

- [X] I confirm I have checked existing issues and online Documentation.

kumekay, 2024-04-29:
@tswen Thank you for opening the ticket, 
we are aware of this problem and working on the solution.

We will update this issue once it is available

kumekay, 2024-05-20:
@tswen 
Current main can support this https://github.com/espressif/idf-component-manager?tab=readme-ov-file#installing-a-development-version-of-the-component-manager

Currently, you need to use the commit before the version bump, so ESP-IDF dependency checker won't complain: https://github.com/espressif/idf-component-manager/commit/b543e18dd76df29d51a9a481928519ff3a031951

tswen, 2024-05-27:
Hello, is the next version after 1.5.2 going to be 2.0.0? When will this version be released?

hfudev, 2024-08-13:
closing the issue since 2.0.0 has been released.

closed 2024-08-13 by hfudev