espressif/idf-component-manager #62: dependenies.lock is updated every time I reconfigure the project

Barabas5532, 2024-05-04:
### The Component Manager version

v1.5.2

### ESP-IDF Version

ESP-IDF v5.1.2

### python Version

Python 3.12.3

### Operating System

Arch Linux

### Description

After building the project, dependencies are automatically updated.

The diff of my lock file:

```
diff --git a/firmware/dependencies.lock b/firmware/dependencies.lock
index 98f2e133..d2c2eb43 100644
--- a/firmware/dependencies.lock
+++ b/firmware/dependencies.lock
@@ -1,15 +1,15 @@
 dependencies:
   espressif/mdns:
-    component_hash: 810ec139689ae93bf42520d05de4855fbb68f7140ef67797d91d8d61829589cb
+    component_hash: 9ff9c578fc3a5ea3c68e807233e47f2ce6b2492a92d7a9356ae5c29ab45f0fde
     source:
       service_url: https://api.components.espressif.com/
       type: service
-    version: 1.2.2
+    version: 1.3.0
   idf:
     component_hash: null
     source:
       type: idf
     version: 5.1.2
-manifest_hash: f07478b4217cadfb47c3d14602d60c69fe72bea2bcf6d412a6d50a2eb6af3c56
+manifest_hash: f2d6e0d52b55cc65c3c08e684f6a2c5691fb6492c0de06a8b57c804bc54956cb
 target: esp32
 version: 1.0.0
```

Similar to #29

### To Reproduce

```cpp
1. Checkout project
1. Build
1. `git restore dependencies.lock`
1. `idf.py reconfigure`

Both the initial build and the reconfigure update the dependencies.lock with this message:

Manifest files have changed, solving dependencies.
....Updating lock file at /path/to/repo/firmware/dependencies.lock
```
```

### Expected behaviour

Lock file is not updated unless I explicitly change a dependency.

### I have checked existing issues and online Documentation

- [X] I confirm I have checked existing issues and online Documentation.

Barabas5532, 2024-05-04:
I've managed to narrow it down to some breaking change in the manifest hash calculation between version 1.4.1 and 1.5.2. Each version calculates a different hash for the same file.

My development environment ended up downloading 1.5.2 after having to reinstall due to system python update. After downgrade to 1.4.1 I'm getting the same hash as what's saved to git and the lock file is no longer changing all the time.

I also noticed that this bug can end up breaking CI builds that are intended to be reproducible. If I had committed the updated lock file to the repo without updating the package version in CI to match the updated development environment version, then CI would think the manifest has changed, ignore the lock file and download the latest version of dependencies instead of the locked versions. Maybe instead of updating and continuing the build, the build should fail in the lock file is out of date and a separate command should be used to update the lock file?

Barabas5532, 2024-05-04:
I bisected versions through pip and found that 1.5.0-dev0 is the first broken version. 1.4.2 still works fine.

I did not bisect commits, but this commit updates hashes in the test and is only present on the 1.5.0-dev0 and beyond. Possibly this is the root cause: https://github.com/espressif/idf-component-manager/commit/a34a3cca40a45fc9087d4311106c526b0590dc5b

kumekay, 2024-05-05:
@Barabas5532 Thank you for reporting the issue, 

Could you please help us to reproduce this by sharing `idf_component.yml` files in your project? Changes made in 1.5 should not break the way hash sum of the manifest is calculated, so we need to find the cause.

Barabas5532, 2024-05-06:
Here is the file: https://github.com/ShrapnelDSP/ShrapnelMonorepo/blob/28a970fd1fa4ab8cab76f8549873f7477bb72f15/firmware/dependencies.lock#L1

hfudev, 2024-05-20:
Hi @Barabas5532

I'm trying to reproduce your issue with the same commit in your repo, and with v5.1.2 idf, plus 1.4.1 component manager.

Unfortuantely the error is still not reproducible. Here's the full log history

```
  ~/e/r/ShrapnelMonorepo  @28a970fd !1 ···········  11:45:23
❯ git reset --hard      
HEAD is now at 28a970fd Fix nanopb CI (#115)

  ~/e/reproduce-bugs/ShrapnelMonorepo  @28a970fd ·  11:45:25
❯ git status      
HEAD detached at 28a970fd
nothing to commit, working tree clean

  ~/e/reproduce-bugs/ShrapnelMonorepo  @28a970fd ·  11:45:27
❯ . ~/esp/esp-idf/export.sh
Detecting the Python interpreter
Checking "python3" ...
Python 3.12.3
"python3" has been detected
Checking Python compatibility
Checking other ESP-IDF version.
Adding ESP-IDF tools to PATH...
Checking if Python packages are up to date...
Constraint file: /home/fuhanxi/.espressif/espidf.constraints.v5.1.txt
Requirement files:
 - /home/fuhanxi/esp/esp-idf/tools/requirements/requirements.core.txt
Python being checked: /home/fuhanxi/.espressif/python_env/idf5.1_py3.12_env/bin/python
Python requirements are satisfied.
Updated PATH variable:
  /home/fuhanxi/esp/esp-idf/components/espcoredump:/home/fuhanxi/esp/esp-idf/components/partition_table:/home/fuhanxi/esp/esp-idf/components/app_update:/home/fuhanxi/.pyenv/shims:/home/fuhanxi/esp/esp-idf/components/espcoredump:/home/fuhanxi/esp/esp-idf/components/partition_table:/home/fuhanxi/esp/esp-idf/components/app_update:/home/fuhanxi/.espressif/python_env/idf5.1_py3.12_env/bin:/home/fuhanxi/.espressif/tools/xtensa-esp-elf-gdb/12.1_20221002/xtensa-esp-elf-gdb/bin:/home/fuhanxi/.espressif/tools/riscv32-esp-elf-gdb/12.1_20221002/riscv32-esp-elf-gdb/bin:/home/fuhanxi/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin:/home/fuhanxi/.espressif/tools/xtensa-esp32s2-elf/esp-12.2.0_20230208/xtensa-esp32s2-elf/bin:/home/fuhanxi/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin:/home/fuhanxi/.espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin:/home/fuhanxi/.espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin:/home/fuhanxi/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230921/openocd-esp32/bin:/home/fuhanxi/esp/esp-idf/tools:/home/fuhanxi/bin:/home/fuhanxi/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/bin:/home/fuhanxi/.cache/zsh4humans/v5/fzf/bin:/home/fuhanxi/go/bin:/home/fuhanxi/.cargo/bin:/home/fuhanxi/.yarn/bin:/home/fuhanxi/esp/scripts:/home/fuhanxi/esp/py_debug_backend:/home/fuhanxi/esp/qemu/bin:/home/fuhanxi/.local/share/gem/ruby/3.0.0/bin:/home/fuhanxi/.npm/bin:/home/fuhanxi/.deno/bin:/home/fuhanxi/.rvm/bin:/opt/flutter/bin:/home/fuhanxi/.local/share/JetBrains/Toolbox/scripts:/home/fuhanxi/.local/share/JetBrains/Toolbox/scripts:/home/fuhanxi/.rvm/bin

Detected installed tools that are not currently used by active ESP-IDF version.
For removing old versions of xtensa-esp-elf-gdb, riscv32-esp-elf-gdb, xtensa-esp-elf, riscv32-esp-elf, esp32ulp-elf, openocd-esp32, esp-rom-elfs use command 'python /home/fuhanxi/esp/esp-idf/tools/idf_tools.py uninstall'
To free up even more space, remove installation packages of those tools. Use option 'python3 /home/fuhanxi/esp/esp-idf/tools/idf_tools.py uninstall --remove-archives'.

Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build

  ~/e/r/ShrapnelMonorepo  @28a970fd ·········  3s  11:45:54
❯ pip -V             
pip 24.0 from /home/fuhanxi/.espressif/python_env/idf5.1_py3.12_env/lib/python3.12/site-packages/pip (python 3.12)

  ~/e/reproduce-bugs/ShrapnelMonorepo  @28a970fd ·  11:46:44
❯ pip install "idf-component-manager==1.4.1"      
Collecting idf-component-manager==1.4.1
  Using cached idf_component_manager-1.4.1-py2.py3-none-any.whl.metadata (11 kB)
Requirement already satisfied: packaging in /home/fuhanxi/.espressif/python_env/idf5.1_py3.12_env/lib/python3.12/site-packages (from idf-component-manager==1.4.1) (24.0)
Requirement already satisfied: requests<3 in /home/fuhanxi/.espressif/python_env/idf5.1_py3.12_env/lib/python3.12/site-packages (from idf-component-manager==1.4.1) (2.31.0)
Requirement already satisfied: urllib3<2 in /home/fuhanxi/.espressif/python_env/idf5.1_py3.12_env/lib/python3.12/site-packages (from idf-component-manager==1.4.1) (1.26.18)
Requirement already satisfied: requests-file in /home/fuhanxi/.espressif/python_env/idf5.1_py3.12_env/lib/python3.12/site-packages (from idf-component-manager==1.4.1) (1.5.1)
Requirement already satisfied: requests-toolbelt in /home/fuhanxi/.espressif/python_env/idf5.1_py3.12_env/lib/python3.12/site-packages (from idf-component-manager==1.4.1) (1.0.0)
Requirement already satisfied: schema in /home/fuhanxi/.espressif/python_env/idf5.1_py3.12_env/lib/python3.12/site-packages (from idf-component-manager==1.4.1) (0.7.7)
Requirement already satisfied: six in /home/fuhanxi/.espressif/python_env/idf5.1_py3.12_env/lib/python3.12/site-packages (from idf-component-manager==1.4.1) (1.16.0)
Requirement already satisfied: tqdm<5 in /home/fuhanxi/.espressif/python_env/idf5.1_py3.12_env/lib/python3.12/site-packages (from idf-component-manager==1.4.1) (4.66.4)
Requirement already satisfied: pyparsing in /home/fuhanxi/.espressif/python_env/idf5.1_py3.12_env/lib/python3.12/site-packages (from idf-component-manager==1.4.1) (3.0.9)
[... cut ...]
cmake failed with exit code 1, output of the command is in the /home/fuhanxi/esp/reproduce-bugs/ShrapnelMonorepo/firmware/build/log/idf_py_stderr_output_58711 and /home/fuhanxi/esp/reproduce-bugs/ShrapnelMonorepo/firmware/build/log/idf_py_stdout_output_58711
```

Inside the log, there's a few lines that are worthy to be noted.

```
Manifest files have changed, solving dependencies.
...Updating lock file at /home/fuhanxi/esp/reproduce-bugs/ShrapnelMonorepo/firmware/dependencies.lock
Processing 2 dependencies:
[1/2] espressif/mdns (1.3.1)
[2/2] idf (5.1.2)
```

Seems like the manifest files are changed somehow, which triggers the version solver again.

Barabas5532, 2024-05-20:
@hfudev I think you have reproduced it successfully. It should not print

```
Manifest files have changed, solving dependencies.
...Updating lock file at /home/fuhanxi/esp/reproduce-bugs/ShrapnelMonorepo/firmware/dependencies.lock
```

because you have not edited the manifest file. I don't think the manifest is actually changing. It seems like there has been a change in the way the file hash is calculated. If you try with component manager 1.4.2, then this line is not printed because it uses the same hash algorithm as the version used to generate the lock file commited to the project.

If you want to successfully build the project, you need to run `git submodule update --init --recursive` before configuring cmake, but I think you got far enough to notice the bug already.

hfudev, 2024-05-20:
@Barabas5532 I tried again with 1.4.2. Still unreproducible. Please check the following log.

```
  ~/e/r/ShrapnelMonorepo/firmware  @28a970fd ·········  22:17:22
❯ pip -V                             
pip 24.0 from /home/fuhanxi/.espressif/python_env/idf5.1_py3.12_env/lib/python3.12/site-packages/pip (python 3.12)

  ~/e/r/ShrapnelMonorepo/firmware  @28a970fd ·········  22:17:27
❯ pip list | grep idf-component
idf-component-manager 1.4.2

  ~/e/r/ShrapnelMonorepo/firmware  @28a970fd ·········  22:17:32
❯ git status
HEAD detached at 28a970fd
nothing to commit, working tree clean

  ~/e/r/ShrapnelMonorepo/firmware  @28a970fd ·········  22:17:37
❯ idf.py reconfigure                         
Executing action: reconfigure
Running cmake in directory /home/fuhanxi/esp/reproduce-bugs/ShrapnelMonorepo/firmware/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=/home/fuhanxi/.espressif/python_env/idf5.1_py3.12_env/bin/python -DESP_PLATFORM=1 -DCCACHE_ENABLE=1 /home/fuhanxi/esp/reproduce-bugs/ShrapnelMonorepo/firmware"...
-- IDF_TARGET is not set, guessed 'esp32' from sdkconfig '/home/fuhanxi/esp/reproduce-bugs/ShrapnelMonorepo/firmware/sdkconfig'
-- Found Git: /usr/bin/git (found version "2.45.1")
-- Component directory /home/fuhanxi/esp/esp-idf/components/esp_bootloader_format does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/fuhanxi/esp/esp-idf/components/esp_driver_ana_cmpr does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/fuhanxi/esp/esp-idf/components/esp_driver_cam does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/fuhanxi/esp/esp-idf/components/esp_driver_dac does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/fuhanxi/esp/esp-idf/components/esp_driver_gpio does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/fuhanxi/esp/esp-idf/components/esp_driver_gptimer does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/fuhanxi/esp/esp-idf/components/esp_driver_i2c does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/fuhanxi/esp/esp-idf/components/esp_driver_i2s does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/fuhanxi/esp/esp-idf/components/esp_driver_isp does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/fuhanxi/esp/esp-idf/components/esp_driver_jpeg does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/fuhanxi/esp/esp-idf/components/esp_driver_ledc does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/fuhanxi/esp/esp-idf/components/esp_driver_mcpwm does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/fuhanxi/esp/esp-idf/components/esp_driver_parlio does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/fuhanxi/esp/esp-idf/components/esp_driver_pcnt does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/fuhanxi/esp/esp-idf/components/esp_driver_rmt does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/fuhanxi/esp/esp-idf/components/esp_driver_sdio does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/fuhanxi/esp/esp-idf/components/esp_driver_sdm does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/fuhanxi/esp/esp-idf/components/esp_driver_sdmmc does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/fuhanxi/esp/esp-idf/components/esp_driver_sdspi does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/fuhanxi/esp/esp-idf/components/esp_driver_spi does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/fuhanxi/esp/esp-idf/components/esp_driver_tsens does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/fuhanxi/esp/esp-idf/components/esp_driver_uart does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/fuhanxi/esp/esp-idf/components/esp_driver_usb_serial_jtag does not contain a CMakeLists.txt file. No component will be added
-- ccache will be used for faster recompilation
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/fuhanxi/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /home/fuhanxi/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /home/fuhanxi/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
[... cut ...]
CMake Error at /home/fuhanxi/esp/reproduce-bugs/ShrapnelMonorepo/thirdparty/esp-idf-components/nanopb/CMakeLists.txt:8 (include):
  include could not find requested file:

    nanopb/extra/FindNanopb.cmake

-- Configuring incomplete, errors occurred!
cmake failed with exit code 1, output of the command is in the /home/fuhanxi/esp/reproduce-bugs/ShrapnelMonorepo/firmware/build/log/idf_py_stderr_output_6906 and /home/fuhanxi/esp/reproduce-bugs/ShrapnelMonorepo/firmware/build/log/idf_py_stdout_output_6906
```

Could you help share your log with the same commands that works at 1.4.2? Thanks!

Barabas5532, 2024-05-22:
@hfudev That's strange, I don't get the lock file update on that commit with the component manager 1.4.2.

<details>
<summary>Logs</summary>

```
[barabas@BarabasArchDesktop][* (HEAD detached at 28a970fd)][~/projects/hardware/guitar_dsp_platform/monorepo/core/firmware]
$ pip -V
pip 24.0 from /home/barabas/.espressif/python_env/idf5.1_py3.12_env/lib/python3.12/site-packages/pip (python 3.12)
[barabas@BarabasArchDesktop][* (HEAD detached at 28a970fd)][~/projects/hardware/guitar_dsp_platform/monorepo/core/firmware]
$ pip list | grep idf-component
idf-component-manager 1.4.2
[barabas@BarabasArchDesktop][* (HEAD detached at 28a970fd)][~/projects/hardware/guitar_dsp_platform/monorepo/core/firmware]
$ git status
HEAD detached at 28a970fd
nothing to commit, working tree clean
[barabas@BarabasArchDesktop][* (HEAD detached at 28a970fd)][~/projects/hardware/guitar_dsp_platform/monorepo/core/firmware]
$ idf.py reconfigure
Executing action: reconfigure
Running cmake in directory /home/barabas/projects/hardware/guitar_dsp_platform/monorepo/core/firmware/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=/home/barabas/.espressif/python_env/idf5.1_py3.12_env/bin/python -DESP_PLATFORM=1 -DCCACHE_ENABLE=0 /home/barabas/projects/hardware/guitar_dsp_platform/monorepo/core/firmware"...
-- Building ESP-IDF components for target esp32
Processing 2 dependencies:
[1/2] espressif/mdns (1.2.2)
[2/2] idf (5.1.2)
-- Project sdkconfig file /home/barabas/projects/hardware/guitar_dsp_platform/monorepo/core/firmware/sdkconfig
Loading defaults file /home/barabas/projects/hardware/guitar_dsp_platform/monorepo/core/firmware/sdkconfig.defaults...
Compiler supported targets: xtensa-esp32-elf

-- App "esp32-dsp" version: docker/v4-43-g28a970fd
-- Adding linker script /home/barabas/projects/hardware/guitar_dsp_platform/monorepo/core/firmware/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script /home/barabas/source/esp-idf/components/esp_system/ld/esp32/sections.ld.in
-- Adding linker script /home/barabas/source/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script /home/barabas/source/esp-idf/components/esp_rom/esp32/ld/esp32.rom.api.ld
-- Adding linker script /home/barabas/source/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script /home/barabas/source/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script /home/barabas/source/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script /home/barabas/source/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script /home/barabas/source/esp-idf/components/soc/esp32/ld/esp32.peripherals.ld
-- Components: app_trace app_update bootloader bootloader_support bt cmock console cxx driver efuse esp-dsp esp-tls esp_adc esp_app_format esp_coex esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_lcd esp_local_ctrl esp_mm esp_netif esp_netif_stack esp_partition esp_phy esp_pm esp_psram esp_ringbuf esp_rom esp_system esp_timer esp_wifi espcoredump espressif__mdns esptool_py fatfs freertos hal heap http_parser idf_test ieee802154 json log lwip main mbedtls mqtt newlib nvs_flash openthread partition_table perfmon protobuf-c protocomm pthread sdmmc soc spi_flash spiffs tcp_transport ulp unity usb vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: /home/barabas/source/esp-idf/components/app_trace /home/barabas/source/esp-idf/components/app_update /home/barabas/source/esp-idf/components/bootloader /home/barabas/source/esp-idf/components/bootloader_support /home/barabas/source/esp-idf/components/bt /home/barabas/source/esp-idf/components/cmock /home/barabas/source/esp-idf/components/console /home/barabas/source/esp-idf/components/cxx /home/barabas/source/esp-idf/components/driver /home/barabas/source/esp-idf/components/efuse /home/barabas/projects/hardware/guitar_dsp_platform/monorepo/core/thirdparty/esp-idf-components/esp-dsp /home/barabas/source/esp-idf/components/esp-tls /home/barabas/source/esp-idf/components/esp_adc /home/barabas/source/esp-idf/components/esp_app_format /home/barabas/source/esp-idf/components/esp_coex /home/barabas/source/esp-idf/components/esp_common /home/barabas/source/esp-idf/components/esp_eth /home/barabas/source/esp-idf/components/esp_event /home/barabas/source/esp-idf/components/esp_gdbstub /home/barabas/source/esp-idf/components/esp_hid /home/barabas/source/esp-idf/components/esp_http_client /home/barabas/source/esp-idf/components/esp_http_server /home/barabas/source/esp-idf/components/esp_https_ota /home/barabas/source/esp-idf/components/esp_https_server /home/barabas/source/esp-idf/components/esp_hw_support /home/barabas/source/esp-idf/components/esp_lcd /home/barabas/source/esp-idf/components/esp_local_ctrl /home/barabas/source/esp-idf/components/esp_mm /home/barabas/source/esp-idf/components/esp_netif /home/barabas/source/esp-idf/components/esp_netif_stack /home/barabas/source/esp-idf/components/esp_partition /home/barabas/source/esp-idf/components/esp_phy /home/barabas/source/esp-idf/components/esp_pm /home/barabas/source/esp-idf/components/esp_psram /home/barabas/source/esp-idf/components/esp_ringbuf /home/barabas/source/esp-idf/components/esp_rom /home/barabas/source/esp-idf/components/esp_system /home/barabas/source/esp-idf/components/esp_timer /home/barabas/source/esp-idf/components/esp_wifi /home/barabas/source/esp- [... cut ...]
- https://github.com/Barabas5532/esp-component-manager-bug-reproduction/commit/b6c64121f7f3f9db5d8a094ff2010d087806ed99 while having 1.4.2 installed. Lock file changes.
- https://github.com/Barabas5532/esp-component-manager-bug-reproduction/commit/b6c64121f7f3f9db5d8a094ff2010d087806ed99 while having 1.5.2 installed. Lock file doesn't change.

When the lock file changes, then the

```
Manifest files have changed, solving dependencies.
...Updating lock file at /path/to/dependencies.lock
```

messages also show up in the build output.

Barabas5532, 2024-05-22:
I have done a git bisect and arrived at the same commit I found earlier (https://github.com/espressif/idf-component-manager/commit/a34a3cca40a45fc9087d4311106c526b0590dc5b). It looks like a lot of hashes in tests were updated here, potentially even where the changes were not supposed to cause the hash to change.

[bisect-log.txt](https://github.com/espressif/idf-component-manager/files/15407951/bisect-log.txt)

hfudev, 2024-05-23:
@Barabas5532 Many thanks for your research. Yes, the commit is changing the way we calculate hash. [here](https://github.com/espressif/idf-component-manager/commit/a34a3cca40a45fc9087d4311106c526b0590dc5b#diff-4df38b71a0e9aaa8d85e3ea9d67d6f9c62bb40d076482fab0bdf4645b77f3217R252)

Unfortunately, I couldn't come up with a easy workaround. Seems like the only way is to generate the new lock file with 1.5.2, and submit one more commit.

Sorry for the inconvenience. We're working on 2.0 release to provide a more stable manifest file parser and lock file serialier.

hfudev, 2024-07-10:
Closing this issue, since unfortunately this problem has no fix nor good workaround.

closed 2024-07-10 by hfudev