espressif/idf-component-manager #63: Test fails with python3.12

stkw0, 2024-06-04:
### The Component Manager version

v1.5.2

### ESP-IDF Version

v5.2.1

### python Version

3.12

### Operating System

Gentoo

### Description

pytest fails when running with python3.12. Is it planned to add support for python3.12 in the short term?

The test log is the next:
```
 =========================== short test summary info ============================
 FAILED tests/manifest/test_validator.py::TestManifestValidator::test_validate_tags_invalid_length - AssertionError: assert 1 == 2
  +  where 1 = len(['Invalid manifest format'])
 FAILED tests/manifest/test_validator.py::TestManifestValidator::test_validate_tags_invalid_symbols - AssertionError: assert 1 == 2
  +  where 1 = len(['Invalid manifest format'])
 FAILED tests/manifest/test_validator.py::TestManifestValidator::test_validate_links_wrong_url - AssertionError: assert 1 == 2
  +  where 1 = len(['Invalid manifest format'])
  +    where ['Invalid manifest format'] = <idf_component_tools.manifest.validator.ManifestValidator object at 0x7f730f02a930>._errors
 FAILED tests/manifest/test_validator.py::TestManifestValidator::test_validate_links_wrong_git - AssertionError: assert 1 == 2
  +  where 1 = len(['Invalid manifest format'])
  +    where ['Invalid manifest format'] = <idf_component_tools.manifest.validator.ManifestValidator object at 0x7f730f029130>._errors
 FAILED tests/test_api_client.py::TestAPIClient::test_no_registry_url_error - idf_component_tools.registry.api_client_errors.NetworkConnectionError: HTTPConnectionPool(host='localhost', port=5000): Max retries exceeded with url: /api/components/example/cmp/versions (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7f730efd4860>: Failed to establish a new connection: [Errno 111] Connection refused'))
 FAILED tests/test_component_manager.py::test_create_example_component_not_exist - vcr.errors.CannotOverwriteExistingCassetteException: Can't overwrite existing cassette ('tests/fixtures/vcr_cassettes/test_create_example_component_not_exist.yaml') in your current record mode (<RecordMode.ONCE: 'once'>).
 No match for the request (<Request (GET) http://localhost:9000/test-public/components/espressif/test.json>) was found.
 Found 1 similar requests with 4 different matcher(s) :

 1 - (<Request (GET) https://components-file.espressif.com/components/espressif/test.json>).
 Matchers succeeded : ['method', 'query']
 Matchers failed :
 scheme - assertion failure :
 http != https
 host - assertion failure :
 localhost != components-file.espressif.com
 port - assertion failure :
 9000 != 443
 path - assertion failure :
 /test-public/components/espressif/test.json != /components/espressif/test.json
 FAILED tests/test_config.py::test_component_registry_url_storage_env - AssertionError: assert (None, ['https://storage.com/']) == ('http://localhost:5000/api/', ['https://storage.com/'])

   At index 0 diff: None != 'http://localhost:5000/api/'

   Full diff:
     (
   -     'http://localhost:5000/api/',
   +     None,
         [
             'https://storage.com/',
         ],
     )
 FAILED tests/test_config.py::test_component_registry_url_multiple_storage_env - AssertionError: assert (None, ['https://storage.com/', 'https://test.com/']) == ('http://localhost:5000/api/', ['https://storage.com/', 'https://test.com/'])

   At index 0 diff: None != 'http://localhost:5000/api/'

   Full diff:
     (
   -     'http://localhost:5000/api/',
   +     None,
         [
[... 135 lines cut ...]
   + ec2dcddbc9340dd9f363da95df2ea99a6c6408f701ce62dfbf66743e3489031d
 FAILED tests/test_service_details.py::test_service_details_success - AssertionError: assert 'http://localhost:5000/api/' == 'https://example.com/api/'

   - https://example.com/api/
   + http://localhost:5000/api/
 FAILED tests/test_service_details.py::test_get_component_registry_url_with_profile - AssertionError: assert 'http://localhost:5000/api/' == 'https://example.com/api/'

   - https://example.com/api/
   + http://localhost:5000/api/
 ========== 23 failed, 458 passed, 6 deselected, 33 warnings in 13.80s ==========
```

### To Reproduce

```cpp
Run pytest with python3.12
```

### Expected behaviour

Test pass 

### I have checked existing issues and online Documentation

- [X] I confirm I have checked existing issues and online Documentation.

hfudev, 2024-06-05:
Hi @stkw0, thanks for bringing us this issue. We're tracking it internally as well. IIRC it's related to the test execution order, which is changed in pytest 8 version.

As a quick workaround, you may run `pip install pytest<8`. Could you give it a try?

stkw0, 2024-06-05:
I found this while packaging the program for a Linux distro. And indeed, it seems that downgrading pytest workarounds the issue.

hfudev, 2024-07-10:
done in https://github.com/espressif/idf-component-manager/commit/ffb0a8e65377cdfc6d5e20adb051a6202221bdc5

closed 2024-07-10 by hfudev