espressif/idf-component-manager #64: ERROR: HTTP request error when uploading component on windows

nicklasb, 2024-06-06:
### The Component Manager version

1.5.2

### ESP-IDF Version

5.*

### python Version

3.11.7

### Operating System

Windows 11

### Browser (for https://components.espressif.com  Issues)

Chrome

### Description

When I try to upload, with a valid token, I get a generic HTTP request error.
I have tried with all pythons installed, so it doesn't seem to be that, I have the ESP-IDF toolchain installed.

```
C:\Users\**\Projects\Robusto\components\robusto> compote component upload --namespace robusto --name Robusto
WARNING: Unknown string field "maintainers" in the manifest file
WARNING: Dropping key "maintainers" from manifest.
Saving archive to "C:\Users\**\Projects\Robusto\components\robusto\dist\Robusto_1.2.6.tgz"
ERROR: HTTP request error
URL: components.espressif.com/api/components/robusto/robusto
```

It would be great if it was possible to increase the verbosity of compote in some way. 

Note: I succeeded in uploading from my MacOS laptop, and I cannot se any major differences versus the Windows 11 setup from my end, so to say,

### To Reproduce

```cpp
Be me and try to publish the Robusto framework from a Windows 11 box.
```

### Expected behaviour

No errors.

### I have checked existing issues and online Documentation

- [X] I confirm I have checked existing issues and online Documentation.

kumekay, 2024-06-07:
Hello Nicklas, 
thank you for opening the issue,

unfortunately, I wasn't able to become you and reproduce the issue just yet,
but we'll improve network error reporting in the coming release, so it should be easier to figure out what is going on.

We had one similar report in the past, the problem was due to firewall restrictions in the corporate network. However, I believe, it's not your case.

nicklasb, 2024-06-23:
Ok, so I managed to find out what was the issue; I had no https:// before the components.espressif.com,
in the environment IDF_COMPONENT_REGISTRY_URL. 

This took some time to find our as I had to actually restart the vscode application, it didn't help to just restarting the command line instances.

Obvious in hindsight. However better error messaging is very welcom. Basically one cannot over-message on errors in a application like this.

closed 2024-06-23 by nicklasb