espressif/idf-component-manager #65: Dependency error during 'idf.py set-target' (IDFGH-13026)

mjdswan, 2024-06-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF v5.2.2 + ESP32C6

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

I expected to be able to set the project target to esp32c6 and then edit configuration via menuconfig or IDE config editor

### What is the actual behavior?

I performed a fresh install of IDE 3.0.0 + IDF 5.2.2 via the offline installer on a new Windows 11 laptop, then tried the following inside the standard 'led_light' example project folder with PS:

install.ps1
export.ps1
idf.py set-target esp32c6

The result is a dependency error that is probably simple, but I don't understand it. Please see my pasted log below.

### Steps to reproduce.

In project folder:

1. install.ps1
2. export.ps1
3. idf.py set-target esp32c6

### Build or installation Logs.

```plain
PS C:\Espressif\workspace\esp-rainmaker\examples\led_light> idf.py set-target esp32c6
Adding "set-target"'s dependency "fullclean" to list of commands with default set of options.
Executing action: fullclean
Build directory 'C:\Espressif\workspace\esp-rainmaker\examples\led_light\build' not found. Nothing to clean.
Executing action: set-target
Set Target to: esp32c6, new sdkconfig will be created.
Running cmake in directory C:\Espressif\workspace\esp-rainmaker\examples\led_light\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=C:\Espressif\python_env\idf5.2_py3.11_env\Scripts\python.exe -DESP_PLATFORM=1 -DIDF_TARGET=esp32c6 -DCCACHE_ENABLE=1 C:\Espressif\workspace\esp-rainmaker\examples\led_light"...
-- Found Git: C:/Espressif/tools/idf-git/2.44.0/cmd/git.exe (found version "2.44.0.windows.1")
-- Component directory C:/Espressif/workspace/esp-rainmaker/components/esp-insights does not contain a CMakeLists.txt file. No component will be added
-- ccache will be used for faster recompilation
-- The C compiler identification is GNU 13.2.0
-- The CXX compiler identification is GNU 13.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Espressif/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/bin/riscv32-esp-elf-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Espressif/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/bin/riscv32-esp-elf-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Espressif/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/bin/riscv32-esp-elf-g++.exe - skipped
[... 30 lines cut ...]
.Using component placed at C:\Espressif\workspace\esp-rainmaker\components\jsmn for dependency espressif/jsmn(~1.1), specified in C:/Espressif/workspace/esp-rainmaker/components/json_parser\idf_component.yml
.Using component placed at C:\Espressif\workspace\esp-rainmaker\components\esp-insights\components\esp_insights for dependency espressif/esp_insights(~1.0.1), specified in C:/Espressif/workspace/esp-rainmaker/examples/common/app_insights\idf_component.yml
Using component placed at C:\Espressif\workspace\esp-rainmaker\components\esp_rainmaker for dependency espressif/esp_rainmaker(~=1.0), specified in C:/Espressif/workspace/esp-rainmaker/examples/common/app_insights\idf_component.yml
Using component placed at C:\Espressif\workspace\esp-rainmaker\components\esp_schedule for dependency espressif/esp_schedule(~1.2.0)(introduced by component espressif/esp_rainmaker), specified in C:/Espressif/workspace/esp-rainmaker/examples/common/app_insights\idf_component.yml
Using component placed at C:\Espressif\workspace\esp-rainmaker\components\json_generator for dependency espressif/json_generator(~1.1.1)(introduced by component espressif/esp_rainmaker), specified in C:/Espressif/workspace/esp-rainmaker/examples/common/app_insights\idf_component.yml
Using component placed at C:\Espressif\workspace\esp-rainmaker\components\json_parser for dependency espressif/json_parser(~1.0.3)(introduced by component espressif/esp_rainmaker), specified in C:/Espressif/workspace/esp-rainmaker/examples/common/app_insights\idf_component.yml
Using component placed at C:\Espressif\workspace\esp-rainmaker\components\esp-insights\components\rmaker_common for dependency espressif/rmaker_common(~1.4.2)(introduced by component espressif/esp_rainmaker), specified in C:/Espressif/workspace/esp-rainmaker/examples/common/app_insights\idf_component.yml
CMake Error at C:/Espressif/frameworks/esp-idf-v5.2.2/tools/cmake/build.cmake:544 (message):
  ERROR: Because no versions of espressif/esp_insights match >1.0.1,<1.1.0

   and espressif/esp_insights (1.0.1) depends on espressif/cbor (~0.6), espressif/esp_insights (>=1.0.1,<1.1.0) requires espressif/cbor (~0.6).

  So, because no versions of espressif/cbor match ~0.6

   and project depends on espressif/esp_insights (~1.0.1), version solving failed.

Call Stack (most recent call first):
  C:/Espressif/frameworks/esp-idf-v5.2.2/tools/cmake/project.cmake:605 (idf_build_process)
  CMakeLists.txt:16 (project)

.-- Configuring incomplete, errors occurred!
See also "C:/Espressif/workspace/esp-rainmaker/examples/led_light/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1, output of the command is in the C:\Espressif\workspace\esp-rainmaker\examples\led_light\build\log\idf_py_stderr_output_13704 and C:\Espressif\workspace\esp-rainmaker\examples\led_light\build\log\idf_py_stdout_output_13704
PS C:\Espressif\workspace\esp-rainmaker\examples\led_light>
```

dobairoland, 2024-06-19:
@mjdswan Yes, you are right. The workaround I've posted is just for the master branch and not for v5.2.2 you are using. I'm sorry but I won't be much of a help regarding this. 

We've moved the issue into the component manager repository. Please wait until the team gets back to you.

mjdswan, 2024-06-19:
@dobairoland thank you for the update.

kumekay, 2024-06-19:
@mjdswan 

It looks like this issue didn't present in the existing 1.5 releases of the component manager. 
Could you please try one of this 2 workarounds. 

- Use an older version of the component manager: In your ESP-IDF shell run `pip install idf-component-manager==1.4.2`
- Another approach is to force ESP-IDF 5.2 to accept component manager v2. For this you need to remove the line starting with `idf-component-manager` from `[HOME]\.espressif\espidf.constraints.v5.2.txt` to allow 2.0 of the idf-component-manager and install the development version from git:

```
# remove old version of the component manager
python -m pip uninstall -y idf-component-manager
# install the development version (from the main branch)
python -m pip install git+https://github.com/espressif/idf-component-manager.git@main
```

mjdswan, 2024-06-20:
@kumekay Thanks, your second approach worked well.  The first approach produced errors regarding extra components it was unable to find, even though they were definitely there in the tree.  After installing idf-component-manager 2.0, those components were found and I'm able to proceed and set the target successfully.

closed 2024-06-20 by mjdswan

kumekay, 2024-06-21:
@mjdswan  espidf.constraints.v5.2.txt may be overwritten by install.ps1, but it won't be a problem once we release component manager 2.0