espressif/idf-component-manager #69: ERROR: Cannot establish a connection to the component registry (IDFGH-13366)

Oleg-Perevyshin, 2024-07-31:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF v 5.3.0

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

The project from the examples should come together

### What is the actual behavior?

Anything related to the component registry is not collected

### Steps to reproduce.

1. New Project Wizard
2. I choose a project from examples (blink)
3. I'm trying to put together a project

### Build or installation Logs.

```plain
*  Выполнение задачи: c:\Users\perevyshin_oe\.espressif\tools\cmake\3.24.0\bin\cmake.exe -G=Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -B=e:\_ESP_\BaseESP\blink\build -S=e:\_ESP_\BaseESP\blink 

-- Found Git: C:/Users/perevyshin_oe/.espressif/tools/idf-git/2.39.2/cmd/git.exe (found version "2.39.2.windows.1") 
-- The C compiler identification is GNU 13.2.0
-- The CXX compiler identification is GNU 13.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Users/perevyshin_oe/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/xtensa-esp32-elf-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Users/perevyshin_oe/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/xtensa-esp32-elf-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Users/perevyshin_oe/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/xtensa-esp32-elf-g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- git rev-parse returned 'fatal: not a git repository (or any of the parent directories): .git'
-- Could not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32
Dependencies lock doesn't exist, solving dependencies.
CMake Error at C:/Users/perevyshin_oe/esp/v5.3/esp-idf/tools/cmake/build.cmake:552 (message):
  ERROR: Cannot establish a connection to the component registry.  Are you
  connected to the internet?

  URL:
  https://components-file.espressif.com/components/espressif/led_strip.json

Call Stack (most recent call first):
  C:/Users/perevyshin_oe/esp/v5.3/esp-idf/tools/cmake/project.cmake:710 (idf_build_process)
  CMakeLists.txt:6 (project)

-- Configuring incomplete, errors occurred!
See also "E:/_ESP_/BaseESP/blink/build/CMakeFiles/CMakeOutput.log".

 *  Процесс терминала "c:\Users\perevyshin_oe\.espressif\tools\cmake\3.24.0\bin\cmake.exe '-G=Ninja', '-DPYTHON_DEPS_CHECKED=1', '-DESP_PLATFORM=1', '-B=e:\_ESP_\BaseESP\blink\build', '-S=e:\_ESP_\BaseESP\blink'" был завершен с кодом выхода 1
```

### More Information.

The project does not build if there is at least one component from the component registry.
The component can be accessed via a browser:
https://components-file.espressif.com/components/espressif/led_strip.json

kumekay, 2024-08-01:
Hello @Oleg-Perevyshin  

Could you please try temporary disabling TLS validation by setting the environment variable 
`IDF_COMPONENT_VERIFY_SSL` to `0` and trying to run the command again:

Please open `ESP-IDF: Open ESP-IDF Terminal` in VSCode and then run

```
$env:IDF_COMPONENT_VERIFY_SSL = "0"
cd e:\_ESP_\BaseESP\blink
idf.py build
```

Similar issue (not necessary connected to your case): https://github.com/espressif/idf-component-manager/issues/27

Oleg-Perevyshin, 2024-08-12:
It doesn't work
The file ninja.exe is present in the path C:\Users\perevyshin_oe\.espressif\tools\ninja\1.11.1\
Unclear symbols mean - The specified file cannot be found
```
 *  Выполнение задачи: C:\Users\perevyshin_oe\.espressif\tools\ninja\1.11.1\ninja.exe  

ninja: error: loading 'build.ninja': ╨Э╨╡ ╤Г╨┤╨░╨╡╤В╤Б╤П ╨╜╨░╨╣╤В╨╕ ╤Г╨║╨░╨╖╨░╨╜╨╜╤Л╨╣ ╤Д╨░╨╣╨╗.

 *  Процесс терминала "C:\Users\perevyshin_oe\.espressif\tools\ninja\1.11.1\ninja.exe" был завершен с кодом выхода 1. 
```

Oleg-Perevyshin, 2024-08-19:
Some miracles.
Reinstalled the extension and all packages from scratch. (Windows 11)
<details>
  <summary>CMakeError.log</summary>

```cmake
Compiling the C compiler identification source file "CMakeCCompilerId.c" failed.
Compiler: xtensa-esp32-elf-gcc 
Build flags: -mlongcalls;-Wno-frame-address;;-fno-builtin-memcpy;-fno-builtin-memset;-fno-builtin-bzero;-fno-builtin-stpcpy;-fno-builtin-strncpy
Id flags:  

The output was:
Не удается найти указанный файл

Compiling the C compiler identification source file "CMakeCCompilerId.c" failed.
Compiler: xtensa-esp32-elf-gcc 
Build flags: -mlongcalls;-Wno-frame-address;;-fno-builtin-memcpy;-fno-builtin-memset;-fno-builtin-bzero;-fno-builtin-stpcpy;-fno-builtin-strncpy
Id flags: -c 

The output was:
Не удается найти указанный файл

Compiling the C compiler identification source file "CMakeCCompilerId.c" failed.
Compiler: xtensa-esp32-elf-gcc 
Build flags: -mlongcalls;-Wno-frame-address;;-fno-builtin-memcpy;-fno-builtin-memset;-fno-builtin-bzero;-fno-builtin-stpcpy;-fno-builtin-strncpy
Id flags: -Aa 

The output was:
Не удается найти указанный файл

Compiling the C compiler identification source file "CMakeCCompilerId.c" failed.
Compiler: xtensa-esp32-elf-gcc 
Build flags: -mlongcalls;-Wno-frame-address;;-fno-builtin-memcpy;-fno-builtin-memset;-fno-builtin-bzero;-fno-builtin-stpcpy;-fno-builtin-strncpy
Id flags: -D__CLASSIC_C__ 

The output was:
Не удается найти указанный файл

Compiling the C compiler identification source file "CMakeCCompilerId.c" failed.
Compiler: xtensa-esp32-elf-gcc 
Build flags: -mlongcalls;-Wno-frame-address;;-fno-builtin-memcpy;-fno-builtin-memset;-fno-builtin-bzero;-fno-builtin-stpcpy;-fno-builtin-strncpy
Id flags: --target=arm-arm-none-eabi;-mcpu=cortex-m3 

The output was:
Не удается найти указанный файл

Compiling the C compiler identification source file "CMakeCCompilerId.c" failed.
Compiler: xtensa-esp32-elf-gcc 
Build flags: -mlongcalls;-Wno-frame-address;;-fno-builtin-memcpy;-fno-builtin-memset;-fno-builtin-bzero;-fno-builtin-stpcpy;-fno-builtin-strncpy
Id flags: -c;-I__does_not_exist__ 

The output was:
Не удается найти указанный файл

Compiling the C compiler identification source file "CMakeCCompilerId.c" failed.
Compiler: xtensa-esp32-elf-gcc 
Build flags: 
Id flags:  

The output was:
Не удается найти указанный файл

Compiling the C compiler identification source file "CMakeCCompilerId.c" failed.
Compiler: xtensa-esp32-elf-gcc 
Build flags: 
Id flags: -c 

The output was:
Не удается найти указанный файл

[... 128 lines cut ...]
Не удается найти указанный файл

Checking whether the CXX compiler is IAR using "" did not match "IAR .+ Compiler":
Checking whether the CXX compiler is IAR using "" did not match "IAR .+ Compiler":
Checking whether the ASM compiler is GNU using "--version" did not match "(GNU assembler)|(GCC)|(Free Software Foundation)":
Checking whether the ASM compiler is Clang using "--version" did not match "(clang version)":
Checking whether the ASM compiler is AppleClang using "--version" did not match "(Apple LLVM version)":
Checking whether the ASM compiler is ARMClang using "--version" did not match "armclang":
Checking whether the ASM compiler is HP using "-V" did not match "HP C":
Checking whether the ASM compiler is Intel using "--version" did not match "(ICC)":
Checking whether the ASM compiler is IntelLLVM using "--version" did not match "(Intel[^
]+oneAPI)":
Checking whether the ASM compiler is SunPro using "-V" did not match "Sun C":
Checking whether the ASM compiler is XL using "-qversion" did not match "XL C":
Checking whether the ASM compiler is MSVC using "-?" did not match "Microsoft":
Checking whether the ASM compiler is TI using "-h" did not match "Texas Instruments":
Checking whether the ASM compiler is IAR using "" did not match "IAR Assembler":
Checking whether the ASM compiler is ARMCC using "" did not match "(ARM Compiler)|(ARM Assembler)|(Arm Compiler)":
Checking whether the ASM compiler is NASM using "-v" did not match "(NASM version)":
Checking whether the ASM compiler is YASM using "--version" did not match "(yasm)":
Checking whether the ASM compiler is ADSP using "-version" did not match "Analog Devices":
Checking whether the ASM compiler is QCC using "-V" did not match "gcc_nto":
```

</details>

kumekay, 2024-08-19:
Hi @Oleg-Perevyshin 

It looks like this issue is not related to the component manager, but something is wrong with your installation. @brianignacio5 Could you please take a look?

brianignacio5, 2024-08-19:
I would suggest to delete the build directory entirely and try again. 

To check if the esp-idf vscode extension is ok please share the `ESP-IDF: Doctor Command` output as described in https://github.com/espressif/vscode-esp-idf-extension/blob/master/README.md#Troubleshooting 

You can add an item  `IDF_COMPONENT_VERIFY_SSL` and set value to `0` in the vscode settings UI `Idf: Custom Extra Vars` so that the build task can pick this value.

Oleg-Perevyshin, 2024-08-19:
I completely reinstalled the extension and all packages. I deleted everything beforehand.
I tried the ESP-IDF command: Doctor Command

<details>
  <summary>report.txt</summary>

```txt
---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.22621 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files\Microsoft VS Code\bin;C:<HOMEPATH>\esp\v5.3\esp-idf\tools;C:\Program Files\nodejs\;C:\Program Files\Git\cmd;C:<HOMEPATH>\.espressif\tools\ninja\1.11.1;C:<HOMEPATH>\AppData\Local\GitHubDesktop\bin;C:<HOMEPATH>\AppData\Roaming\npm;C:<HOMEPATH>\.espressif\tools\cmake\3.24.0\bin; 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.92.2 
Visual Studio Code language ru 
Visual Studio Code shell C:\Windows\System32\WindowsPowerShell\v1.0\powershell.exe 
ESP-IDF Extension version 1.8.0 
Workspace folder e:\ESP-Basic 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) C:<HOMEPATH>\esp\v5.3\esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
ESP-HomeKit-SDK Path (idf.espHomeKitSdkPath) ${env:HOMEKIT_PATH}
Custom extra paths (idf.customExtraPaths) C:<HOMEPATH>\.espressif\tools\xtensa-esp-elf-gdb\14.2_20240403\xtensa-esp-elf-gdb\bin;C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf-gdb\14.2_20240403\riscv32-esp-elf-gdb\bin;C:<HOMEPATH>\.espressif\tools\xtensa-esp-elf\esp-13.2.0_20240530\xtensa-esp-elf\bin;C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf\esp-13.2.0_20240530\riscv32-esp-elf\bin;C:<HOMEPATH>\.espressif\tools\esp32ulp-elf\2.38_20240113\esp32ulp-elf\bin;C:<HOMEPATH>\.espressif\tools\cmake\3.24.0\bin;C:<HOMEPATH>\.espressif\tools\openocd-esp32\v0.12.0-esp32-20240318\openocd-esp32\bin;C:<HOMEPATH>\.espressif\tools\ninja\1.11.1;C:<HOMEPATH>\.espressif\tools\idf-exe\1.0.3;C:<HOMEPATH>\.espressif\tools\ccache\4.8\ccache-4.8-windows-x86_64;C:<HOMEPATH>\.espressif\tools\dfu-util\0.11\dfu-util-0.11-win64;C:<HOMEPATH>\.espressif\tools\esp-rom-elfs\20240305
Custom extra vars (idf.customExtraVars)
    OPENOCD_SCRIPTS: C:<HOMEPATH>\.espressif\tools\openocd-esp32\v0.12.0-esp32-20240318/openocd-esp32/share/openocd/scripts
    IDF_CCACHE_ENABLE: 1
    ESP_ROM_ELF_DIR: C:<HOMEPATH>\.espressif\tools\esp-rom-elfs\20240305/
Virtual env Python Path (idf.pythonBinPath) C:<HOMEPATH>\.espressif\python_env\idf5.3_py3.11_env\Scripts\python.exe
Serial port (idf.port) COM8
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitj_v1.cfg,target/esp32.cfg
ESP-IDF Tools Path (idf.toolsPath) C:<HOMEPATH>\.espressif
Git Path (idf.gitPath) C:<HOMEPATH>\.espressif\tools\idf-git\2.39.2\cmd\git.exe
Notification Mode (idf.notificationMode) All
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-HomeKit Path (idf.espHomeKitSdkPath) false
Access to ESP-IDF Custom extra paths
Access to C:<HOMEPATH>\.espressif\tools\xtensa-esp-elf-gdb\14.2_20240403\xtensa-esp-elf-gdb\bin: true
Access to C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf-gdb\14.2_20240403\riscv32-esp-elf-gdb\bin: true
Access to C:<HOMEPATH>\.espressif\tools\xtensa-esp-elf\esp-13.2.0_20240530\xtensa-esp-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf\esp-13.2.0_20240530\riscv32-esp-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\esp32ulp-elf\2.38_20240113\esp32ulp-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\cmake\3.24.0\bin: true
Access to C:<HOMEPATH>\.espressif\tools\openocd-esp32\v0.12.0-esp32-20240318\openocd-esp32\bin: true
Access to C:<HOMEPATH>\.espressif\tools\ninja\1.11.1: true
Access to C:<HOMEPATH>\.espressif\tools\idf-exe\1.0.3: true
Access to C:<HOMEPATH>\.espressif\tools\ccache\4.8\ccache-4.8-windows-x86_64: true
Access to C:<HOMEPATH>\.espressif\tools\dfu-util\0.11\dfu-util-0.11-win64: true
Access to C:<HOMEPATH>\.espressif\tools\esp-rom-elfs\20240305: true
Access to Virtual env Python Path (idf.pythonBinPath) true
Access to CMake in environment PATH undefined
Access to Ninja in environment PATH undefined
Access to ESP-IDF Tools Path (idf.toolsPath) true
-------------------------------------------------------- Configurations has spaces -------------------------------------------------------------
Spaces in system environment Path true
Spaces in ESP-ADF Path (idf.espAdfPath) false
Spaces in ESP-IDF Path (idf.espIdfPath) false
Spaces in ESP-MDF Path (idf.espMdfPath) false
Spaces in ESP-Matter Path (idf.espMatterPath) false
Spaces in ESP-HomeKit-SDK Path (idf.espHomeKitSdkPath) false
Spaces in ESP-IDF Custom extra paths
Spaces in C:<HOMEPATH>\.espressif\tools\xtensa-esp-elf-gdb\14.2_20240403\xtensa-esp-elf-gdb\bin: false
Spaces in C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf-gdb\14.2_20240403\riscv32-esp-elf-gdb\bin: false
[... 98 lines cut ...]
[... cut ...]
{"message":"ENOENT: no such file or directory, open 'e:\\ESP-Basic\\.vscode\\launch.json'","stack":"Error: ENOENT: no such file or directory, open 'e:\\ESP-Basic\\.vscode\\launch.json'","level":"error","timestamp":"2024-08-19T08:57:38.324Z"}
{"user":true,"level":"warn","message":"Отчет о конфигурации расширения скопирован в буфер обмена с ошибками.","timestamp":"2024-08-19T08:57:38.324Z"}

--------------------------------------------------------------------------------------------------------------------------------------------

```

</details>

brianignacio5, 2024-08-19:
Doesn't seem to be any issue with your setup.

What is the build output? Have you tried to rebuild after deleting the build directory? Ninja seems to be found and have access as shown in doctor command. Maybe your current directory or terminal have some issue? Did you modify your vscode terminal profile in any way ? Info about vscode profiles in here: https://code.visualstudio.com/docs/terminal/profiles

Oleg-Perevyshin, 2024-08-19:
Yes, I deleted the build folder.
Now in the terminal it writes this:
```
 *  Выполнение задачи: C:\Users\perevyshin_oe\.espressif\tools\ninja\1.11.1\ninja.exe  

ninja: error: loading 'build.ninja': ╨Э╨╡ ╤Г╨┤╨░╨╡╤В╤Б╤П ╨╜╨░╨╣╤В╨╕ ╤Г╨║╨░╨╖╨░╨╜╨╜╤Л╨╣ ╤Д╨░╨╣╨╗.

 *  Процесс терминала "C:\Users\perevyshin_oe\.espressif\tools\ninja\1.11.1\ninja.exe" был завершен с кодом выхода 1. 

 *  Выполнение задачи: C:\Users\perevyshin_oe\.espressif\tools\ninja\1.11.1\ninja.exe  

ninja: error: loading 'build.ninja': ╨Э╨╡ ╤Г╨┤╨░╨╡╤В╤Б╤П ╨╜╨░╨╣╤В╨╕ ╤Г╨║╨░╨╖╨░╨╜╨╜╤Л╨╣ ╤Д╨░╨╣╨╗.

 *  Процесс терминала "C:\Users\perevyshin_oe\.espressif\tools\ninja\1.11.1\ninja.exe" был завершен с кодом выхода 1. 
 ```

I didn't do anything in the terminal settings.
I created a default project. Then I connected littlefs from the component manager.
I tried to build the project on one machine (mac os), the project built successfully. I saved it on Github.

Now I opened the project on Windows, but it won't build.

Oleg-Perevyshin, 2024-08-19:
After deleting the build folder, I see this in the terminal:
```
 *  Выполнение задачи: C:\Users\perevyshin_oe\.espressif\tools\cmake\3.24.0\bin\cmake.exe -G=Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -B=e:\ESP-Basic\build -S=e:\ESP-Basic 

-- Found Git: C:/Users/perevyshin_oe/.espressif/tools/idf-git/2.39.2/cmd/git.exe (found version "2.39.2.windows.1") 
-- The C compiler identification is GNU 13.2.0
-- The CXX compiler identification is GNU 13.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Users/perevyshin_oe/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/xtensa-esp32-elf-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Users/perevyshin_oe/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/xtensa-esp32-elf-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Users/perevyshin_oe/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/xtensa-esp32-elf-g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
HINT: Cannot establish a connection to the component registry. Skipping checks of dependency changes.
Processing 2 dependencies:
[1/2] idf (5.3.0)
[2/2] joltwallet/littlefs (1.14.8)
CMake Error at C:/Users/perevyshin_oe/esp/v5.3/esp-idf/tools/cmake/build.cmake:552 (message):
  ERROR: Some components (joltwallet/littlefs) in the "managed_components"
  directory were modified on the disk since the last run of the CMake.
  Content of this directory is managed automatically.

  If you want to keep the changes, you can move the directory with the
  component to the "components"directory of your project.

  I.E.  for "joltwallet__littlefs" run:

  mv E:\ESP-Basic\managed_components\joltwallet__littlefs
  E:\ESP-Basic\components\joltwallet__littlefs

  Or, if you want to discard the changes remove the ".component_hash" file
  from the component's directory.

  I.E.  for "joltwallet__littlefs" run:

  rm E:\ESP-Basic\managed_components\joltwallet__littlefs\.component_hash

Call Stack (most recent call first):
  C:/Users/perevyshin_oe/esp/v5.3/esp-idf/tools/cmake/project.cmake:710 (idf_build_process)
  CMakeLists.txt:8 (project)

-- Configuring incomplete, errors occurred!
See also "E:/ESP-Basic/build/CMakeFiles/CMakeOutput.log".

 *  Процесс терминала "C:\Users\perevyshin_oe\.espressif\tools\cmake\3.24.0\bin\cmake.exe '-G=Ninja', '-DPYTHON_DEPS_CHECKED=1', '-DESP_PLATFORM=1', '-B=e:\ESP-Basic\build', '-S=e:\ESP-Basic'" был завершен с кодом выхода 1. 
```

Oleg-Perevyshin, 2024-08-19:
I entered the command
rm E:\ESP-Basic\managed_components\joltwallet__littlefs\.component_hash

Deleted the build folder
In the terminal I see this:
```
 *  Выполнение задачи: C:\Users\perevyshin_oe\.espressif\tools\cmake\3.24.0\bin\cmake.exe -G=Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -B=e:\ESP-Basic\build -S=e:\ESP-Basic 

-- Found Git: C:/Users/perevyshin_oe/.espressif/tools/idf-git/2.39.2/cmd/git.exe (found version "2.39.2.windows.1") 
-- The C compiler identification is GNU 13.2.0
-- The CXX compiler identification is GNU 13.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Users/perevyshin_oe/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/xtensa-esp32-elf-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Users/perevyshin_oe/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/xtensa-esp32-elf-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Users/perevyshin_oe/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/xtensa-esp32-elf-g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
HINT: Cannot establish a connection to the component registry. Skipping checks of dependency changes.
Processing 2 dependencies:
[1/2] idf (5.3.0)
[2/2] joltwallet/littlefs (1.14.8)
CMake Error at C:/Users/perevyshin_oe/esp/v5.3/esp-idf/tools/cmake/build.cmake:552 (message):
  ERROR: Cannot establish a connection to the component registry.  Are you
  connected to the internet?

  URL:
  https://components-file.espressif.com/components/joltwallet/littlefs.json

Call Stack (most recent call first):
  C:/Users/perevyshin_oe/esp/v5.3/esp-idf/tools/cmake/project.cmake:710 (idf_build_process)
  CMakeLists.txt:8 (project)

-- Configuring incomplete, errors occurred!
See also "E:/ESP-Basic/build/CMakeFiles/CMakeOutput.log".

 *  Процесс терминала "C:\Users\perevyshin_oe\.espressif\tools\cmake\3.24.0\bin\cmake.exe '-G=Ninja', '-DPYTHON_DEPS_CHECKED=1', '-DESP_PLATFORM=1', '-B=e:\ESP-Basic\build', '-S=e:\ESP-Basic'" был завершен с кодом выхода 1. 
```
There is nothing wrong with the internet connection.

Oleg-Perevyshin, 2024-08-21:
I did all this, everything is fine.
But still nothing has changed.

Apparently I have a problem with the Internet, something is blocked somewhere...

closed 2024-08-21 by Oleg-Perevyshin

kumekay, 2024-08-21:
@Oleg-Perevyshin Could you please try to run this command and provide us an error, if any?

In ESP-IDF terminal.
```
python -c "import requests; response = requests.get('https://components-file.espressif.com/components/joltwallet/littlefs.json'); print(response.text)"
```
In some configurations of Windows python may use a system set of TLS certificates, instead of supplied one.

If the previous command fails, try to access something else through python, like:

```
python -c "import requests; response = requests.get('https://jsonplaceholder.typicode.com/posts/1'); print(response.text)"    
```

fariscooool, 2025-02-13:
Hello, I have also encountered a similar problem when downloading components or compiling with idf.Py
 ERROR: Cannot establish a connection to the component registry. Are you connected to the internet?
 But my network is all normal.
I tried many methods but couldn't solve it, but I can run it normally on another laptop. What is the reason for this situation

kumekay, 2025-02-14:
Hello @fariscooool because this issue is closed a while ago, it's likely that your problem is not related. Could you please open a new one, describe your environment and application and paste the logs?

fariscooool, 2025-02-14:
Thank you for your  @kumekay . I have submitted a new issue; please take a look at it: https://github.com/espressif/idf-component-manager/issues/83