espressif/idf-component-manager #70: libmodbus

diplfranzhoepfinger, 2024-08-18:
### Related area

libmodbus

### Is your feature request related to a problem?

libmodbus

### Describe the solution you'd like

libmodbus

### Describe alternatives you've considered

libmodbus

### Additional context

libmodbus

### I have checked existing list of Feature requests and I have provided enough information.

- [X] I confirm I have checked existing list of Feature requests and I have provided enough information.

igrr, 2024-08-19:
Hi @diplfranzhoepfinger, this repository hosts the python client of IDF component manager. The issue tracker is for the client only.

You are welcome to open a topic on esp32.com if you would like to get other developers' feedback on your idea.

closed 2024-08-19 by igrr

diplfranzhoepfinger, 2024-08-19:
done already: https://esp32.com/viewtopic.php?f=13&t=3302