espressif/idf-component-manager #71: Error in uploading a new version for existing component.

ronny-antoon, 2024-09-02:
### The Component Manager version

v1.5.3

### ESP-IDF Version

v5.3.0

### python Version

3.11.2

### Operating System

Windows 10

### Browser (for https://components.espressif.com  Issues)

Chrome

### Description

I have some components uploaded and maintained by me. all was working as expected.
It past a good time that last time i uploaded a new version. 
today i fixed some code in my component and i want to upload a new version and i got this error:-

```
PS C:\Users\*******\Documents\chatgptcoder\DeviceModule\components\DeviceModule> compote component upload --namespace ronny-antoon --name DeviceModule
Selected profile "default" from the idf_component_manager.yml config file
Saving archive to "C:\Users\*******\Documents\chatgptcoder\DeviceModule\components\DeviceModule\dist\DeviceModule_5.1.0.tgz"
Uploading archive C:\Users\*******\Documents\chatgptcoder\DeviceModule\components\DeviceModule\dist\DeviceModule_5.1.0.tgz
Wait for processing, it is safe to press CTRL+C and exit
You can check the state of processing by running CLI command "compote component upload-status --job=02786495-c2a5-4bbb-a92d-0e980c97a132 "
WARNING: A homepage URL has not been provided in the manifest file. Please update your `idf_component.yml` file and include the URL to the component's homepage. Documentation: https://docs.espressif.com/pWARNING: A component description has not been provided in the manifest file. Please update your `idf_component.yml` file and include a brief description of the component.This will help other developers understand the purpose and functionality of your component. Documentation: https://docs.espressif.com/projects/idf-component-manager/en/latest/guides/packaging_components.html#create-idf-component-yml      
ERROR: Uploaded component wasn't processed successfully.
https://github.com/espressif/idf-component-manager/issues
```

### To Reproduce

```cpp
Just try to upload a new version.
```

### Expected behaviour

should upload a new version.

### I have checked existing issues and online Documentation

- [X] I confirm I have checked existing issues and online Documentation.

kumekay, 2024-09-04:
@ronny-antoon Thank you for reporting the issue. We have found the regression and fixed it, please try to re-upload the component.

As a side note, consider adding the home page of the project and non-empty readme file, to help others to understand your component!

closed 2024-09-04 by kumekay

ronny-antoon, 2024-09-04:
> @ronny-antoon Thank you for reporting the issue. We have found the regression and fixed it, please try to re-upload the component.
> 
> As a side note, consider adding the home page of the project and non-empty readme file, to help others to understand your component!

I really Appreciate your work. thank you. its working now