espressif/idf-component-manager #73: Error Processing Dependencies during IDF Build

Otto45, 2024-09-12:
### The Component Manager version

v2.0.3

### ESP-IDF Version

v5.2.2

### python Version

3.11.2

### Operating System

Windows 11

### Browser (for https://components.espressif.com  Issues)

Firefox

### Description

I installed IDF v5.2.2 using the VS Code extension, and the component manager now fails when I try to build my project. I previously was using IDF v5.1.4 with no issues.

### To Reproduce

```cpp
1. Create a project in VS Code using IDF v5.1.4
2. Install some dependencies (i.e. `idf.py add-dependency "bblanchon/arduinojson^7.1.0"`)
3. Build the project.
4. Install IDF v5.2.2
5. Build the project.

Here is the relevant part of the build log:

-- Building ESP-IDF components for target esp32s3
Processing 5 dependencies:
[1/5] bblanchon/arduinojson (7.1.0)
CMake Error at C:/Users/austi/esp/v5.2.2/esp-idf/tools/cmake/build.cmake:544 (message):
  Traceback (most recent call last):

    File "<frozen runpy>", line 198, in _run_module_as_main
    File "<frozen runpy>", line 88, in _run_code
    File "C:\Users\austi\.espressif\python_env\idf5.2_py3.11_env\Lib\site-packages\idf_component_manager\prepare_components\__main__.py", line 6, in <module>
      main()
    File "C:\Users\austi\.espressif\python_env\idf5.2_py3.11_env\Lib\site-packages\idf_component_manager\prepare_components\prepare.py", line 130, in main
      args.func(args)
    File "C:\Users\austi\.espressif\python_env\idf5.2_py3.11_env\Lib\site-packages\idf_component_manager\prepare_components\prepare.py", line 29, in prepare_dep_dirs
      ).prepare_dep_dirs(
        ^^^^^^^^^^^^^^^^^
    File "C:\Users\austi\.espressif\python_env\idf5.2_py3.11_env\Lib\site-packages\idf_component_manager\core.py", line 98, in wrapper
      return func(self, *args, **kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^
    File "C:\Users\austi\.espressif\python_env\idf5.2_py3.11_env\Lib\site-packages\idf_component_manager\core.py", line 757, in prepare_dep_dirs
      downloaded_components = download_project_dependencies(
                              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    File "C:\Users\austi\.espressif\python_env\idf5.2_py3.11_env\Lib\site-packages\idf_component_manager\dependencies.py", line 392, in download_project_dependencies
      download_path = fetcher.download()
                      ^^^^^^^^^^^^^^^^^^
    File "C:\Users\austi\.espressif\python_env\idf5.2_py3.11_env\Lib\site-packages\idf_component_tools\sources\fetcher.py", line 70, in download
      return self.source.download(self.component, self.managed_path)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    File "C:\Users\austi\.espressif\python_env\idf5.2_py3.11_env\Lib\site-packages\idf_component_tools\sources\web_service.py", line 234, in download
      if validate_managed_component_by_hashdir(
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    File "C:\Users\austi\.espressif\python_env\idf5.2_py3.11_env\Lib\site-packages\idf_component_tools\hash_tools\validate_managed_component.py", line 21, in validate_managed_component_by_hashdir
      manifest = manifest_manager.load()
                 ^^^^^^^^^^^^^^^^^^^^^^^
    File "C:\Users\austi\.espressif\python_env\idf5.2_py3.11_env\Lib\site-packages\idf_component_tools\manager.py", line 154, in load
      if self.is_valid:
         ^^^^^^^^^^^^^
    File "C:\Users\austi\.espressif\python_env\idf5.2_py3.11_env\Lib\site-packages\idf_component_tools\manager.py", line 139, in is_valid
      self.validate()
    File "C:\Users\austi\.espressif\python_env\idf5.2_py3.11_env\Lib\site-packages\idf_component_tools\manager.py", line 76, in validate
      manifest_dict = yaml.safe_load(self.path.read_text()) or {}
                                     ^^^^^^^^^^^^^^^^^^^^^
    File "pathlib.py", line 1059, in read_text
    File "encodings\cp1252.py", line 23, in decode

  UnicodeDecodeError: 'charmap' codec can't decode byte 0x90 in position 69:
  character maps to <undefined>

Call Stack (most recent call first):
  C:/Users/austi/esp/v5.2.2/esp-idf/tools/cmake/project.cmake:605 (idf_build_process)
  CMakeLists.txt:26 (project)
```

### Expected behaviour

The component manager should run successfully, and the build should complete successfully.

### Additional info

I tried doing a full clean, deleting my dependencies.lock file, and reinstalling my dependencies using the `idf.py add-dependency` command all at once to start from scratch, but I am still getting this issue.

### I have checked existing issues and online Documentation

- [X] I confirm I have checked existing issues and online Documentation.

kumekay, 2024-09-13:
Hi @Otto45, thank you for the detailed report, It looks like a regression, we'll make a fix soon

hfudev, 2024-10-02:
Hi @Otto45 v2.0.4 has been released, which solves this problem. Please have a try

`pip install -U idf-component-manager`

closed 2024-10-02 by hfudev

Otto45, 2024-10-02:
@hfudev it works, thank you!