espressif/idf-component-manager #74: Add component script causes problems (IEC-165)

BenthamSoftwareServices, 2024-09-14:
### Answers checklist.

- [X] I have read the component documentation [ESP-IDF Components](https://components.espressif.com) and the issue is not addressed there.
- [X] I am using target and esp-idf version as defined in component's idf_component.yml
- [X] I have searched the [issue tracker](https://github.com/espressif/idf-extra-components/issues?q=is%3Aissue) for a similar issue and not found any related issue.

### Which component are you using? If you choose Other, provide details in More Information.

led_strip

### ESP-IDF version.

5.2.2

### Development Kit.

Any ESP32-S3

### Used Component version.

2.5.5

### More Information.

I have tried adding many components using the supplied scripts successfully with the VSCode ESP-IDF extension, but using idf.py add-dependency "espressif/led_strip^2.5.5" cause numerous build errors and general problems e.g. running idf.py reconfigure and/or idf.py update-dependencies produces more error messages.  I can download the archive an add it to a components folder in the project folder and that works OK, so I assume the component is OK it's just the addition script that causes me problems.

kumekay, 2024-09-16:
Hi @BenthamSoftwareServices thank you for the issue, could you please add a log with errors you receive running `idf.py reconfigure`?

BenthamSoftwareServices, 2024-09-17:
OK I created a new sample project, which builds OK.  I ran the script for lvgl which created the yml file and after building successfully it created a managed_components folder with a sub folder lvgl_lvgl with all the contents, I added an #include and rebuilt OK.  Next I ran the script for esp-lcd-touch which added its entry to the yml file and after a successful build created the folder espressif__esp_lcd_touch in the managed components folder, I added an #include and rebuilt OK - all good so far.

Then I ran the script for led_strip which added its entry in the yml file but when I tried to build the output was:

Executing task: c:\Espressif\tools\tools\ninja\1.11.1\ninja.exe  

[0/1] Re-running CMake...-- git rev-parse returned 'fatal: not a git repository (or any of the parent directories): .git'
-- Could not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32s3
Processing 4 dependencies:
[1/4] espressif/esp_lcd_touch (1.1.2)
[2/4] espressif/led_strip (2.5.5)
CMake Error at C:/Espressif/frameworks/esp-idf-v5.2.2/tools/cmake/build.cmake:544 (message):
  Traceback (most recent call last):

    File "<frozen runpy>", line 198, in _run_module_as_main
    File "<frozen runpy>", line 88, in _run_code
    File "c:\Espressif\tools\python_env\idf5.2_py3.11_env\Lib\site-packages\idf_component_manager\prepare_components\__main__.py", line 6, in <module>
      main()
    File "c:\Espressif\tools\python_env\idf5.2_py3.11_env\Lib\site-packages\idf_component_manager\prepare_components\prepare.py", line 130, in main
      args.func(args)
    File "c:\Espressif\tools\python_env\idf5.2_py3.11_env\Lib\site-packages\idf_component_manager\prepare_components\prepare.py", line 29, in prepare_dep_dirs
      ).prepare_dep_dirs(
        ^^^^^^^^^^^^^^^^^
    File "c:\Espressif\tools\python_env\idf5.2_py3.11_env\Lib\site-packages\idf_component_manager\core.py", line 98, in wrapper
      return func(self, *args, **kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^
    File "c:\Espressif\tools\python_env\idf5.2_py3.11_env\Lib\site-packages\idf_component_manager\core.py", line 757, in prepare_dep_dirs
      downloaded_components = download_project_dependencies(
                              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    File "c:\Espressif\tools\python_env\idf5.2_py3.11_env\Lib\site-packages\idf_component_manager\dependencies.py", line 392, in download_project_dependencies
      download_path = fetcher.download()
                      ^^^^^^^^^^^^^^^^^^
    File "c:\Espressif\tools\python_env\idf5.2_py3.11_env\Lib\site-packages\idf_component_tools\sources\fetcher.py", line 70, in download
      return self.source.download(self.component, self.managed_path)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    File "c:\Espressif\tools\python_env\idf5.2_py3.11_env\Lib\site-packages\idf_component_tools\sources\web_service.py", line 255, in download
      unpack_archive(file_path, self.component_cache_path(component))
    File "c:\Espressif\tools\python_env\idf5.2_py3.11_env\Lib\site-packages\idf_component_tools\archive_tools.py", line 94, in unpack_archive
      prepare_empty_directory(destination_directory)
    File "c:\Espressif\tools\python_env\idf5.2_py3.11_env\Lib\site-packages\idf_component_tools\file_tools.py", line 111, in prepare_empty_directory
      rmtree(directory)
    File "shutil.py", line 759, in rmtree
    File "shutil.py", line 617, in _rmtree_unsafe
    File "shutil.py", line 603, in _rmtree_unsafe
    File "shutil.py", line 600, in _rmtree_unsafe

  PermissionError: [WinError 5] Access is denied:
  'C:\\Users\\Steven\\AppData\\Local\\Espressif\\ComponentManager\\Cache\\service_d92d8f1e\\espressif__led_strip_2.5.5_28c6509a\\examples'

Call Stack (most recent call first):
  C:/Espressif/frameworks/esp-idf-v5.2.2/tools/cmake/project.cmake:605 (idf_build_process)
  CMakeLists.txt:8 (project)

-- Configuring incomplete, errors occurred!
See also "E:/Users/Steven/Documents/GitHub/ESP-IDF/test4/build/CMakeFiles/CMakeOutput.log".
ninja: error: rebuilding 'build.ninja': subcommand failed

FAILED: build.ninja
C:\Espressif\tools\tools\cmake\3.24.0\bin\cmake.exe --regenerate-during-build -SE:\Users\Steven\Documents\GitHub\ESP-IDF\test4 -BE:\Users\Steven\Documents\GitHub\ESP-IDF\test4\build

 *  The terminal process "c:\Espressif\tools\tools\ninja\1.11.1\ninja.exe" terminated with exit code: 1.

Next I removed espressif/led_strip: "^2.5.5" from the yml file and I could then build without problems.  So the problem seems to be confined to the led_strip component?

BenthamSoftwareServices, 2024-09-18:
Done a few more tests.  If I use the Show Examples button and then select an example and create a new project based on it e.g. SPI_LCD_TOUCH I noticed that after setting the target the Managed Component folder is added and populated.  If I do the same with the Blink example, which uses Led_strip component the include in main has red error stating the file can not be found and if I try to set COM port and target it fails.