espressif/idf-component-manager #75: dependencies added for all components, also for unused components

morgana2313, 2024-09-19:
### The Component Manager version

v2.0.3

### ESP-IDF Version

v5.3

### python Version

3.12.6

### Operating System

macOS

### Description

idf-component-manager appears to add component dependencies for **all** components found in `COMPONENT_DIRS ` and `EXTRA_COMPONENT_DIRS `, not just for the components that are actually used.

This causes problems when a component with a dependency that cannot build is in one of these directories. (example:`TARGET=linux` and `espressif/onewire_bus`)

### To Reproduce

```cpp
* add a dependency to an unused component
* this dependency is added to `managed-components` and thus compiled.
```

### Expected behaviour

* unused component's dependencies should not be added to `managed-components`.

### I have checked existing issues and online Documentation

- [X] I confirm I have checked existing issues and online Documentation.

igrr, 2024-09-19:
@morgana2313 That's right, the build system downloads dependencies earlier than we know which components are actually used or not.

You probably can solve this problem by not adding onewire-bus dependency if the target is `linux`.

Please check this section for conditions in idf_component.yml: https://docs.espressif.com/projects/idf-component-manager/en/latest/reference/manifest_file.html#conditional-dependencies

Here's an example: https://github.com/espressif/idf-extra-components/blob/f16d432ef6d00fb9d48adde64f22aefc55bf16bc/esp_jpeg/examples/get_started/main/idf_component.yml#L8-L9

morgana2313, 2024-09-23:
Thanks that works for me!

closed 2024-09-23 by morgana2313