espressif/idf-component-manager #76: how can I change the default lock file version?

mksafavi, 2024-09-25:
### The Component Manager version

2.0.2

### ESP-IDF Version

v5.3.0

### python Version

3.12.4

### Operating System

Linux

### Description

After updating IDF to version 5.3, the `dependencies.lock` schema version changed to `2.0.0`. 

It modified the `dependencies.lock` like changing service_url to registry_url and so on.... which is fine, but on the official docker image, it still uses the first version schema and reverts the file back. I noticed this when my build pipeline added `-dirty` to the versions.

I couldn't find the documentation on how to set the lock version.
Manually changing 2.0.0 to 1.0.0 gives this warning:
```
NOTICE: Current idf-component-manager default lock file version is 2.0.0, but found 1.0.0 in 
<path/to/project>/dependencies.lock. Recreating lock file with the current version.
```
So I assume there is a way to set this?

### To Reproduce

```cpp
-
```

### Expected behaviour

setting the default lock version.

### I have checked existing issues and online Documentation

- [X] I confirm I have checked existing issues and online Documentation.

hfudev, 2024-09-25:
Hi @mksafavi I can help provide a workaround:

if you want to set the lock version to 1.0. You need to downgrade idf-component-manager to 1.x.

```bash
pip install "idf-component-manager<2"
```

could you help share the full file path which is being modified in the official docker image? We could see what we can do.

mksafavi, 2024-09-25:
> if you want to set the lock version to 1.0. You need to downgrade idf-component-manager to 1.x.
> 
> ```shell
> pip install "idf-component-manager<2"
> ```

Oh! It makes sense now. I thought the new `idf-component-manager` supported both of the schemas.
On my machine with idf v5.3 the version 2.0.2 of component manager is installed. The v5.3 container has 1.5.3 while v5.3.1 has 2.0.3.

It seems like my distro is shipping different versions. So the issue isn't from here ✌️

I looked in the esp-idf repository and the requirement.txt files don't contain the version limits. The docs mentioned that they are stored in espidf.constraints.*.txt.
In the docker container I can find this file: `./opt/esp/espidf.constraints.v5.3.txt`

Is this file available in the Github repository?  
We might be able to use it to match the container versions.

Thank you!

hfudev, 2024-09-25:
@mksafavi you may download the constraint files here: https://dl.espressif.com/dl/esp-idf/espidf.constraints.v5.3.txt

```
idf-component-manager>=1.2,<3
```

mksafavi, 2024-09-25:
> @mksafavi you may download the constraint files here: https://dl.espressif.com/dl/esp-idf/espidf.constraints.v5.3.txt
> 
> ```
> idf-component-manager>=1.2,<3
> ```

Thanks 👍
I think we can close this issue.

closed 2024-09-25 by hfudev