espressif/idf-component-manager #78: Examples pull wrong component version

gojimmypi, 2024-12-10:
### The Component Manager version

 v1.4.1

### ESP-IDF Version

v5.2

### python Version

3.10.12 

### Operating System

Windows 11 WSL

### Browser (for https://components.espressif.com  Issues)

chrome

### Description

Specifying an explicit release version of wolfSSL library instead pulls a preview version.

Note the current wolfssl version (as of this issue date) is `5.7.2`, but there are several preview versions of `5.7.4` available:

[image]

Note that when running `pip install -U idf-component-manager` broke my ability to install the examples:

```
$ pip install -U idf-component-manager
Requirement already satisfied: idf-component-manager in /home/gojimmypi/.espressif/python_env/idf5.2_py3.10_env/lib/python3.10/site-packages (1.4.1)
Collecting idf-component-manager
  Downloading idf_component_manager-2.1.1-py3-none-any.whl.metadata (11 kB)
Requirement already satisfied: click in /home/gojimmypi/.espressif/python_env/idf5.2_py3.10_env/lib/python3.10/site-packages (from idf-component-manager) (8.1.7)
Requirement already satisfied: colorama in /home/gojimmypi/.espressif/python_env/idf5.2_py3.10_env/lib/python3.10/site-packages (from idf-component-manager) (0.4.6)
Requirement already satisfied: pyparsing in /home/gojimmypi/.espressif/python_env/idf5.2_py3.10_env/lib/python3.10/site-packages (from idf-component-manager) (3.1.1)
Requirement already satisfied: pyyaml in /home/gojimmypi/.espressif/python_env/idf5.2_py3.10_env/lib/python3.10/site-packages (from idf-component-manager) (6.0)
Collecting ruamel.yaml (from idf-component-manager)
  Using cached ruamel.yaml-0.18.6-py3-none-any.whl.metadata (23 kB)
Requirement already satisfied: requests in /home/gojimmypi/.espressif/python_env/idf5.2_py3.10_env/lib/python3.10/site-packages (from idf-component-manager) (2.31.0)
Requirement already satisfied: requests-file in /home/gojimmypi/.espressif/python_env/idf5.2_py3.10_env/lib/python3.10/site-packages (from idf-component-manager) (1.5.1)
Requirement already satisfied: requests-toolbelt in /home/gojimmypi/.espressif/python_env/idf5.2_py3.10_env/lib/python3.10/site-packages (from idf-component-manager) (1.0.0)
Requirement already satisfied: tqdm in /home/gojimmypi/.espressif/python_env/idf5.2_py3.10_env/lib/python3.10/site-packages (from idf-component-manager) (4.65.0)
Collecting jsonref (from idf-component-manager)
  Downloading jsonref-1.1.0-py3-none-any.whl.metadata (2.7 kB)
Collecting pydantic (from idf-component-manager)
  Downloading pydantic-2.10.3-py3-none-any.whl.metadata (172 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 172.0/172.0 kB 1.5 MB/s eta 0:00:00
Collecting pydantic-core (from idf-component-manager)
  Downloading pydantic_core-2.27.1-cp310-cp310-manylinux_2_17_x86_64.manylinux2014_x86_64.whl.metadata (6.6 kB)
Collecting pydantic-settings (from idf-component-manager)
  Downloading pydantic_settings-2.6.1-py3-none-any.whl.metadata (3.5 kB)
Collecting typing-extensions (from idf-component-manager)
  Using cached typing_extensions-4.12.2-py3-none-any.whl.metadata (3.0 kB)
Collecting annotated-types>=0.6.0 (from pydantic->idf-component-manager)
  Using cached annotated_types-0.7.0-py3-none-any.whl.metadata (15 kB)
Collecting python-dotenv>=0.21.0 (from pydantic-settings->idf-component-manager)
  Downloading python_dotenv-1.0.1-py3-none-any.whl.metadata (23 kB)
Requirement already satisfied: charset-normalizer<4,>=2 in /home/gojimmypi/.espressif/python_env/idf5.2_py3.10_env/lib/python3.10/site-packages (from requests->idf-component-manager) (3.2.0)
Requirement already satisfied: idna<4,>=2.5 in /home/gojimmypi/.espressif/python_env/idf5.2_py3.10_env/lib/python3.10/site-packages (from requests->idf-component-manager) (3.4)
Requirement already satisfied: urllib3<3,>=1.21.1 in /home/gojimmypi/.espressif/python_env/idf5.2_py3.10_env/lib/python3.10/site-packages (from requests->idf-component-manager) (1.26.16)
Requirement already satisfied: certifi>=2017.4.17 in /home/gojimmypi/.espressif/python_env/idf5.2_py3.10_env/lib/python3.10/site-packages (from requests->idf-component-manager) (2023.5.7)
Requirement already satisfied: six in /home/gojimmypi/.espressif/python_env/idf5.2_py3.10_env/lib/python3.10/site-packages (from requests-file->idf-component-manager) (1.16.0)
Collecting ruamel.yaml.clib>=0.2.7 (from ruamel.yaml->idf-component-manager)
  Downloading ruamel.yaml.clib-0.2.12-cp310-cp310-manylinux_2_17_x86_64.manylinux2014_x86_64.whl.metadata (2.7 kB)
Downloading idf_component_manager-2.1.1-py3-none-any.whl (159 kB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 159.3/159.3 kB 2.6 MB/s eta 0:00:00
Downloading jsonref-1.1.0-py3-none-any.whl (9.4 kB)
[... 132 lines cut ...]
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- git rev-parse returned 'fatal: not a git repository (or any parent up to mount point /mnt)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).'
-- Could not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32
Dependencies lock doesn't exist, solving dependencies.
..Updating lock file at /mnt/c/test/component/wolfssl_client/dependencies.lock
Processing 2 dependencies:
[1/2] idf (5.2.0)
[2/2] wolfssl/wolfssl (5.7.4-preview1e)
-- Begin wolfssl
```

### Expected behaviour

only the explicit version of the managed component library should have been used.

### Additional info

See also https://github.com/wolfSSL/wolfssl/pull/8251

### I have checked existing issues and online Documentation

- [X] I confirm I have checked existing issues and online Documentation.

gojimmypi, 2024-12-10:
See also https://github.com/wolfSSL/wolfssl/pull/8251

igrr, 2024-12-10:
@gojimmypi could you please add, what does the main/idf_component.yml file of that example look like? Interested in how you are specifying the dependency on wolfSSL component there.

gojimmypi, 2024-12-10:
Hi @igrr - thanks for taking a look at this. 

TLDR: my release `5.7.2` examples have an undesired version `^5.6.6-Stable-Preview1` text _but_ I would have still expected the most recent _release_ version after that to be used. Instead, I see the most recent _preview_ is used, `version: 5.7.4-preview1e`.

I would think (hope) under all circumstances, that when pulling anything other than an explicit preview version (e.g. `5.7.4-preview1e`), that any specified release  version (e.g. ` 5.7.2`) should never pull in a newer _preview_ by default.

I think this is what I understand from the [Compatible Major Release Clause docs](https://docs.espressif.com/projects/idf-component-manager/en/latest/reference/versioning.html#compatible-release-clause).

With the updated version of the component manager, trying to fetch explicit wolfSSL version  v5.7.2 in the `wolfssl_client` example, I see the text `version: ^5.6.6-Stable-Preview1` in the fetched `main/idf_component.yml` file:

```
idf.py create-project-from-example "wolfssl/wolfssl=5.7.2:wolfssl_client"
$ cat wolfssl_client/main/idf_component.yml
dependencies:
  idf:
    version: '>=4.1.0'
  wolfssl/wolfssl:
    version: ^5.6.6-Stable-Preview1
```

But instead, the resulting version in the example project `dependencies.lock` file is `version: 5.7.4-preview1e` of wolfssl:

```
dependencies:
  idf:
    source:
      type: idf
    version: 5.2.0
  wolfssl/wolfssl:
    component_hash: a6cd6a27ab3cc6874c3cff580c286e6a4f54c09216ca7f7d253c5cc7897baf8c
    dependencies: []
    source:
      registry_url: https://components.espressif.com/
      type: service
    version: 5.7.4-preview1e
direct_dependencies:
- idf
- wolfssl/wolfssl
manifest_hash: 52adc2805c506aae51504b6d9f304aa18e4eae68376b3ec00adcb2936aca9dc9
target: esp32
version: 2.0.0
a
```

The contents in my `dist\wolfssl_5.7.4-preview1e.tgz\idf_component.yml` file that was created during publish is confirmd correct: `version: 5.7.4-preview1e`:

```
description: wolfSSL Embedded SSL/TLS Library
discussion: https://www.wolfssl.com/forums
documentation: https://www.wolfssl.com/documentation/manuals/wolfssl/index.html
files:
  exclude:
  - '**/*'
  - INSTALL.md
  - '*.py'
  - '**/*.list'
  - '**/.vs/**/*'
  - '**/build/**/*'
  - '**/dist/**/*'
  - '**/dist.bak/**/*'
  - '**/out/**/*'
  - '*.sh'
  - '*.bak'
  - sdkconfig
  - README_DIST.md
  - README_REGISTRY_PREPEND.md
  - README_STAGING_PREPEND.md
  - README_STAGING.md
  - ./src/sp_arm64.c
  - ./src/sp_armthumb.c
  - ./src/sp_x86_64.c
  - ./src/sp_arm32.c
  - ./src/sp_c32.c
  - ./src/sp_cortexm.c
  include:
  - ./README.md
  - ./LICENSE.txt
  - ./idf_component.yml
  - ./CMakeLists.txt
  - ./Kconfig
  - ./install_log.txt
  - ./src/**/*
  - ./include/**/*
  - ./wolfssl/**/*
  - ./wolfcrypt/**/*
  - ./examples/**/*
issues: https://github.com/wolfSSL/wolfssl/issues
repository: https://github.com/wolfSSL/wolfssl.git
url: https://www.wolfssl.com
version: 5.7.4-preview1e
```

The respective `dist\wolfssl_5.7.2.tgz\idf_component.yml` is confirmed to have a similar value: `version: 5.7.2` in the last line.

## Examples

The contents of `dist\wolfssl_5.7.4-preview1e.tgz\examples\wolfssl_client\main\idf_component.yml` is this: 

```
## IDF Component Manager Manifest File For Examples
dependencies:
  wolfssl/wolfssl: "^5.7.4-preview1e"
  ## Required IDF version
  idf:
    version: ">=4.1.0"
  # # Put list of dependencies here
  # # For components maintained by Espressif:
  # component: "~1.0.0"
  # # For 3rd party components:
  # username/component: ">=1.0.0,<2.0.0"
  # username2/component2:
  #   version: "~1.0.0"
  #   # For transient dependencies `public` flag can be set.
  #   # `public` flag doesn't have an effect dependencies of the `main` component.
  #   # All dependencies of `main` are public by default.
  #   public: true
```

and in `dist\wolfssl_5.7.2.tgz\examples\wolfssl_client\main\idf_component.yml` is this, which appears in this case was my bad in that the publishing script has the *wrong* version in the examples `5.6.6-Stable-Preview1`:

```
## IDF Component Manager Manifest File For Examples
dependencies:
  wolfssl/wolfssl: "^5.6.6-Stable-Preview1"
  ## Required IDF version
  idf:
    version: ">=4.1.0"
  # # Put list of dependencies here
  # # For components maintained by Espressif:
  # component: "~1.0.0"
  # # For 3rd party components:
  # username/component: ">=1.0.0,<2.0.0"
  # username2/component2:
  #   version: "~1.0.0"
  #   # For transient dependencies `public` flag can be set.
  #   # `public` flag doesn't have an effect dependencies of the `main` component.
  #   # All dependencies of `main` are public by default.
  #   public: true
```

And so yes, I see that I'm the one that specified the "_this version or newer_" in the component examples, but in the case where I wanted to test an explicit version, I would have expected that specific version.

Perhaps it would be best for me to remove the older preview versions altogether?

Do you think it is a best practice that instead of using the `^` version specifier, that instead I only use the `~` for the examples and let the end user choose to include newer versions automatically or not?

igrr, 2024-12-10:
I think it is a reasonable expectation that when an example of a component of a particular version is requested, the example should use the component of that specific version, even though a newer version is available. While it's possible to achieve this behavior by always pinning the dependency version in the examples and updating it in sync with the component version, it makes the release process more complex.
We'll discuss this internally and get back to you!

gojimmypi, 2024-12-10:
> I think it is a reasonable expectation that when an example of a component of a particular version is requested, the example should use the component of that specific version, even though a newer version is available.

Yes, I agree.

Suggestion: is it possible for the receiving component publisher to scan the incoming source code and enforce this? For example, as I noted above, I incorrectly included a bad component requirement in the respective example `main/idf_component.yml` file for the [5.7.2 release](https://components.espressif.com/components/wolfssl/wolfssl/versions/5.7.2?language=en). It would have been preferable to have the publish fail. 

While considering inbound processing side, I also suggest enforcing all version text to be lower case. For example, my [5.7.4-Preview1a](https://components.espressif.com/components/wolfssl/wolfssl/versions/5.7.4-preview1a?language=en) seems to have ended up with a mix of some instances of `Preview1a` converted to lower case `preview1a`. My publishing script now does this on my side.

> it makes the release process more complex

Do you have an example? I would think that simply changing the enclosed example `main/idf_component.yml` files would suffice?

Each enclosed example app `idf_component.yml` would be the exact same version as the component, with just a different version character: `^` vs `=`, eh?

```
dependencies:
  wolfssl/wolfssl: "^5.7.4"
```

vs

```
dependencies:
  wolfssl/wolfssl: "=5.7.4"
```

Thanks again for taking a look at this. Cheers.

hfudev, 2024-12-11:
Hi @gojimmypi Here I'll explain why and provide a possible workaround.

The version solver is picking up preview versions, since preview version is declared in the version spec

```yaml
dependencies:
  idf:
    version: '>=4.1.0'
  wolfssl/wolfssl:
    version: ^5.6.6-Stable-Preview1  # <-- here
```

if you want to pick only released versions, you may update the version spec to 

```yaml
dependencies:
  idf:
    version: '>=4.1.0'
  wolfssl/wolfssl:
    version: ^5.6.6  # <-- here
```

and after you call `rm dependencies.lock && idf.py build`, the released version will be picked

```
...
NOTICE: [1/1] idf (5.5.0)
NOTICE: Dependencies lock doesn't exist, solving dependencies.
..NOTICE: Updating lock file at /home/fuhanxi/esp/test_proj/dependencies.lock
NOTICE: Processing 2 dependencies:
NOTICE: [1/2] wolfssl/wolfssl (5.7.2)
NOTICE: [2/2] idf (5.5.0)
...
```

I also agree on the released version shall always be picked first, unless defined `prerelease: true` aside. Will deal with this issue internally.

gojimmypi, 2024-12-13:
Thanks @hfudev and @igrr - all good suggestions.

I've pinned the source code of the [examples](https://components.espressif.com/components/wolfssl/wolfssl/versions/5.7.4-preview1f/examples/wolfssl_benchmark?language=en) to a specific published version:

```
$ mkdir wow
gojimmypi:/mnt/c/workspace/wolfssl_scripts-gojimmypi/espressif
$ cd wow
gojimmypi:/mnt/c/workspace/wolfssl_scripts-gojimmypi/espressif/wow
$ idf.py create-project-from-example "wolfssl/wolfssl^5.7.4-preview1f:wolfssl_benchmark"
Executing action: create-project-from-example
Example "wolfssl_benchmark" successfully downloaded to /mnt/c/workspace/wolfssl_scripts-gojimmypi/espressif/wow/wolfssl_benchmark
Done
gojimmypi:/mnt/c/workspace/wolfssl_scripts-gojimmypi/espressif/wow
$ cd wolfssl_benchmark
gojimmypi:/mnt/c/workspace/wolfssl_scripts-gojimmypi/espressif/wow/wolfssl_benchmark
$ cat ./main/idf_component.yml
dependencies:
  idf:
    version: '>=4.1.0'
  wolfssl/wolfssl:
    version: 5.7.4-preview1f
gojimmypi:/mnt/c/workspace/wolfssl_scripts-gojimmypi/espressif/wow/wolfssl_benchmark
```

The text on the web site would still need to be edited manually, as it has the leading `^`. It would be great if the web site would mirror the file contents, instead:

[image]

```
$ idf.py --version
ESP-IDF v5.3.1-5-g041cd42c47-dirty
gojimmypi:/mnt/c/workspace/wolfssl_scripts-gojimmypi/espressif/wow/wolfssl_benchmark
$ python -m idf_component_manager --help
usage: __main__.py [-h] [--path PATH] [--namespace NAMESPACE] [--service-profile SERVICE_PROFILE] [--name NAME]
                   [--archive ARCHIVE] [--job JOB] [--version VERSION] [--skip-pre-release] [--check-only]
                   [--allow-existing] [--example EXAMPLE]
                   {pack-component,upload-component,upload-component-status,create-project-from-example,delete-version}

IDF component manager v1.5.2
```

## Component Publish Date

Here's another Component Manager curiosity, probably not worthy of an issue of its own: incorrect "updated [n] days ago" text.

Just now (lunchtime, Dec 13) I published the [wolfSSL 5.7.4-preview1f](https://components.espressif.com/components/wolfssl/wolfssl/versions/5.7.4-preview1f?language=en). However, when visiting that page, there's an indication that it was published _6 days ago_:

[image]

Note the correct date stamp on the tgz file.

I _think_ that prior publish events had an accurate `updated [n] days ago` text.

Surely I cannot control that date. A local web server date problem?

## Binary Size Difference with ESP-IDF v5.3

This latest release, I also used ESP-IDF v5.3 instead of 5.2. Note there's a substantial increase in binary size:

[image]

kumekay, 2024-12-15:
Hi @gojimmypi, thank you for bringing our attention

> Component Publish Date

This is a regression, since refactoring 4 months ago, the date of component creation was frozen on app load, not when a component is created. Because we usually update code quite often, it stayed unnoticed for some time. Fix is made, should be deployed soon.

> as it has the leading ^

As the first step to improving the situation, we will change `^` to `=` for the `create-project-from-example` command.

gojimmypi, 2024-12-16:
Hi @kumekay - 

> Fix is made, should be deployed soon ...  will change ^ to = for the create-project-from-example command.

sounds great, thank you!

How's the staging site doing? I _thought_ my most recent version of wolfSSH preview was working. 

https://components-staging.espressif.com/components/gojimmypi/mywolfssh/versions/1.4.19-preview1a

Now I cannot even add another library such as "esp_jpeg", they all claim to be not found:

```
$ idf.py add-dependency "jason-mao/esp_jpeg^0.1.0"
Executing action: add-dependency
NOTICE: Successfully added dependency "jason-mao/esp_jpeg": "^0.1.0" to component "main"
NOTICE: If you want to make additional changes to the manifest file at path /mnt/c/test/component/wolfssh_template/main/idf_component.yml manually, please refer to the documentation: https://docs.espressif.com/projects/idf-component-manager/en/latest/reference/manifest_file.html
Done
gojimmypi:/mnt/c/test/component/wolfssh_template
$ idf.py build
Executing action: all (aliases: build)
Running cmake in directory /mnt/c/test/component/wolfssh_template/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=/home/gojimmypi/.espressif/python_env/idf5.2_py3.10_env/bin/python -DESP_PLATFORM=1 -DCCACHE_ENABLE=0 /mnt/c/test/component/wolfssh_template"...
-- Detected UNIX
-- Detected WSL
-- Detected Linux
-- No conflicting wolfSSL components found.
-- Found PROTOCOL_EXAMPLES_DIR=/mnt/c/SysGCC/esp32/esp-idf/v5.2/examples/common_components/protocol_examples_common
-- IDF_TARGET not set, using default target: esp32
-- Found Git: /usr/bin/git (found version "2.34.1")
-- The C compiler identification is GNU 13.2.0
-- The CXX compiler identification is GNU 13.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/gojimmypi/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /home/gojimmypi/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /home/gojimmypi/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- git rev-parse returned 'fatal: not a git repository (or any parent up to mount point /mnt)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).'
-- Could not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32
NOTICE: Dependencies lock doesn't exist, solving dependencies.
.HINT: Please check manifest file of the following component(s): main
CMake Error at /mnt/c/SysGCC/esp32/esp-idf/v5.2/tools/cmake/build.cmake:544 (message):
  WARNING: Component "gojimmypi/mywolfssh" not found

  WARNING: Component "gojimmypi/mywolfssl" not found

  WARNING: Component "jason-mao/esp_jpeg" not found

  ERROR: Because project depends on jason-mao/esp_jpeg (^0.1.0) which doesn't
  match any versions, version solving failed.

Call Stack (most recent call first):
  /mnt/c/SysGCC/esp32/esp-idf/v5.2/tools/cmake/project.cmake:605 (idf_build_process)
  CMakeLists.txt:122 (project)

-- Configuring incomplete, errors occurred!
See also "/mnt/c/test/component/wolfssh_template/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1, output of the command is in the /mnt/c/test/component/wolfssh_template/build/log/idf_py_stderr_output_24540 and /mnt/c/test/component/wolfssh_template/build/log/idf_py_stdout_output_24540
```

This is with:

```
export IDF_COMPONENT_REGISTRY_URL="https://components-staging.espressif.com"

idf.py --version
ESP-IDF v5.2-dev-3903-g66992aca7a-dirty

python -m idf_component_manager --help
Usage: python -m idf_component_manager [OPTIONS] COMMAND [ARGS]...

Options:
  -W, --warnings-as-errors  Treat warnings as errors.
  -h, --help                Show this message and exit.

Commands:
  autocomplete  Generate tab-completion scripts for the specified shell.
  cache         Group of commands to manage cache of the IDF Component...
  component     Group of commands to interact with components.
  manifest      Group of commands to manage manifest of the project.
  project       Group of project related commands
  registry      Group of commands to work with component registry
  version       Print version of the IDF Component Manager.
```

gojimmypi, 2025-02-04:
I've discovered why the staging site no longer finds components. See https://github.com/espressif/idf-component-manager/issues/81

TL;DR: needs explicit `registry_url` setting in `main/idf_component.yml`.

```
dependencies:
  gojimmypi/mywolfssh:
    version: ^1.4.19-preview1c
    registry_url: https://components-staging.espressif.com
  gojimmypi/mywolfssl:
    version: ^5.7.4-preview1i
    registry_url: https://components-staging.espressif.com
  idf:
    version: '>=4.1.0'
```