espressif/idf-component-manager #79: Use relative paths for local paths in dependencies.lock

bryghtlabs-richard, 2024-12-17:
### Related area

commit churn

### Is your feature request related to a problem?

The dependencies.lock file encodes the local path to one of my components. Every time dependencies are re-evaluated, this creates silly changes in the repository, and makes it difficult to reproduce building the project from scratch. I believe this only happens to a component I've forked from Espressif, probably because I haven't added idf_component.yml to my other local components.

### Describe the solution you'd like

Instead of:

    path: C:\Projects\SuperCoolProject\components\espressif__esp_websocket_client

Please consider:

    path: .\components\espressif__esp_websocket_client

### Describe alternatives you've considered

I could remove idf_component.yml from my forked component, is that what I should do?

### I have checked existing list of Feature requests and I have provided enough information.

- [X] I confirm I have checked existing list of Feature requests and I have provided enough information.

bryghtlabs-richard, 2025-03-24:
I did find [this documentation link](https://docs.espressif.com/projects/idf-component-manager/en/latest/reference/dependencies_lock.html), but we'd prefer to use a few local components and ensure all developers use the same component versions.

kumekay, 2025-03-24:
Hello @bryghtlabs-richard 
Thank you for opening the ticket,
In general, it will be difficult to do for all the local components and keep the compatibility, but probably we can make different rules for components in the `components` directory in the project.