espressif/idf-component-manager #80: Support for snapshots/nightlies on component registry / recommended way?

tronical, 2025-01-31:
### Related area

Components, registry

### Is your feature request related to a problem?

This is perhaps more of a question than a feature request: For Slint we're publishing a [component](https://components.espressif.com/components/slint/slint) and it's been a great success for us. The registry rocks :). It's so good that we'd love to also use it to publish nightly snapshots of development versions of Slint, to make it easier for our users to try out features before the release and without having to go through a alpha/beta release.

I'm curious if you have a recommended way to achieve this.

### Describe the solution you'd like

We could publish snapshots with a pre-release version scheme (such as `1.10.0-nightly+gitXYZ`), but I'm slightly concerned that it would "blow up" the existing combobox on https://components.espressif.com/components/slint/slint that shows all the versions.

It seems thought that the combobox is sorted (yay) and pre-releases are always at the end.

Or would you recommend publishing a separate package instead, so `slint/slint` for official releases and `slint/slint-nightly` as a separate nightly package?

### I have checked existing list of Feature requests and I have provided enough information.

- [x] I confirm I have checked existing list of Feature requests and I have provided enough information.

kumekay, 2025-02-10:
Hello @tronical 
Please excuse us for the delay with the answer.

We are have plans to redesign the version selector, to make it better suitable for often releases. However, I have another concern about having too many releases.

Every time the dependencies are solved, the component manager downloads a JSON file https://components-file.espressif.com/components/slint/slint.json with all the versions of a given component. Currently, this file is 40kb for slint, but it can grow big with nightly versions. So at this point `slint/slint-nightly` sounds like a better approach.

Currently, this JSON is sent without compression, one straightforward solution is to implement gzip for the response, which should help up to some scale.

tronical, 2025-02-10:
That's an excellent point - thanks! We'll go with `slint/slint-nightly` then.