espressif/idf-component-manager #82: Sub components in managed lib

malachib, 2025-02-10:
### The Component Manager version

2.1.1

### ESP-IDF Version

v5.3.2

### python Version

3.11.2

### Operating System

Debian x64

### Browser (for https://components.espressif.com  Issues)

Chrome

### Description

I have a use case, say 3 libraries:

- shared
- lib1
- lib2

`lib1` and `lib2` both consume `shared` (as if `shared` were an sdk)
I would like to consume all 3 of these under the banner of one managed component.

A crude comparison would be the usage of `esp_lcd` component as contained in `esp-idf` repository.

If I do this locally via a `path` specified in `idf_component.yml`, I am able to do so
If I do this remotely (as we typically do with esp components) via `git`: I get this error:

```
ERROR: The "path" field in the manifest file
"/tmp/tmp8jpiou7b/src/PGESP-74/managed/components/lib1/idf_component.yml"
  does not point to a directory.  You can safely remove this field from the
  manifest if this project is an example copied from a component repository.
```

### To Reproduce

```cpp
Issue is reproducible and observable in my proof of concept area here:

https://bitbucket.org/malachib/playground.esp/src/master/src/PGESP-74/

In particular, `app_remote` exhibits the issue while `app_local` does not.

If more exposition is required, please let me know.
```

### Expected behaviour

I expect to consume a managed component and gain access to contained components also present in its repository.

### Additional info

I am experimenting with the notion of a "meta" component which grabs everything.  I am not married to that approach, it just felt the most viable.

The broader context for this is I am making a graphics library and it has a core and then support drivers.  I want to pull in the core and also the (optional) support drivers in one fell swoop using esp component manager.

### I have checked existing issues and online Documentation

- [x] I confirm I have checked existing issues and online Documentation.

kumekay, 2025-02-18:
Hello @malachib I was able to reproduce your issue, however, the `git` dependency with local dependencies is the case we didn't cover well through the initial design. We will discuss possible solutions and I'll update this issue.

malachib, 2025-02-21:
Yes please share your thinking, and great work on the component system in general by the way!

XDanielPaul, 2026-09-01:
Hi @malachib, we've released a new IDF Component Manager version, which should fix your error

```
ERROR: The "path" field in the manifest file
"/tmp/tmp8jpiou7b/src/PGESP-74/managed/components/lib1/idf_component.yml"
  does not point to a directory.  You can safely remove this field from the
  manifest if this project is an example copied from a component repository.
```

Could you please verify?

malachib, 2026-09-01:
Hi @XDanielPaul OK I gave it a try.  I installed compote 3.1.1 into IDF 5.4.4 and it didn't like it.  I then tried IDF 6.1 and he was happy.

I tried out the core test and `path` is indeed working as you suggest.  It appears to work identically to `override_path`, and we get copies of `lib1`, `lib2` and `shared_lib`  all under `managed_components` folder.  So this does work!  Cool

Some observations:

- If I arrange things as a meta library (with no actual lib of its own) with callouts to `components/lib1` etc the entire content of meta library sits there in `managed_components` unused alongside `lib1`, `lib2` and `shared_lib`
- Relative pathing where `shared_lib` reaches directly to `../components/lib1` works just fine, like with `override_path`.  That's nice.  No duplicate like with above `meta` approach
- In these scenarios sometimes I'm fond of doing relative pathing inside Kconfigs.  i.e. `rsource "../Kconfig.shared"`.  Since libs all copy to top-level `managed_components`, this doesn't play nice.  I'd like a solution to this too, but feels to me like that ought to be a separate issue - unless you have a clever workaround for that too 😄

XDanielPaul, 2026-09-01:
Hi @malachib, great to hear your news!

Just to clarify, Component Manager 3.x requires ESP-IDF 6.0+. For ESP-IDF 5.4.4 you can use 2.5.1, which contains the same fix - we maintain two major versions of Component Manager now.

Regarding your observations:

- Each dependency is installed as a separate managed component. Because your meta-component contains `lib1`, `lib2`, and `shared_lib`, those files are copied with the meta-component and then installed again as separate dependencies. A cleaner layout would keep the small meta-component along side the "real" components

```
components/
  meta/
  lib1/
  lib2/
  shared_lib/
```
The meta manifest can then reference `../lib1`, `../lib2`, and `../shared_lib`

-  Glad this works fine for you!

- As for the last point I am not sure, whether I grasped your scenario, but based on what I think... I believe `rsource` resolves relative to the current KConfig file (according to [docs](https://docs.espressif.com/projects/esp-idf-kconfig/en/latest/kconfiglib/language.html#the-source-entry)). Because of this `rsource "../Kconfig.shared"` depends on the file outside `lib1` and that original repository layout is not preserved when `lib1` is installed independently. I don't know if it fits your use case, but you can try to define shared options in `shared_lib/Kconfig`. ESP-IDF discovers that file when `shared_lib` is included, and `lib1` and `lib2` can reference its symbols directly

malachib, 2026-09-02:
Hi @XDanielPaul yeah I am pretty much rolling with that meta manifest with `../` paths as you suggest.  It does the job nicely.  Thank you!

The `Kconfig.shared` thing, let's talk about the use case that brought all this about.  I have a private graphics library and part of the Kconfig is SPI and Display Resolution settings.  Then, I have maybe a half dozen components for different display support (ILI9341, etc).  Each of these components reaches into a shared `../Kconfig.display` since pretty much everything has that.  Some of them reach into `../Kconfig.spi` also.  Full disclosure I may change all this around in the future, it has some downsides in general doing it this way.  But that is the use case.  Putting all that into one shared lib could work for display settings but is clunky for SPI settings

XDanielPaul, 2026-09-02:
Hi @malachib!

Hmm, that use case makes sense. In this case could the shared settings be split into small `display_config` and `spi_config` components? SPI displays would depend on both, while non-SPI displays would depend only on `display_config`. ESP-IDF will load each dependency's `KConfig` so drivers can reference its symbols directly

If preserving sibling includes such as `../Kconfig.spi` is important, that would be a separate issue because managed components are installed independently

malachib, 2026-09-02:
@XDanielPaul interesting idea.  It's an improvement over putting it into a top level shared lib, in this case it would be flipping it around into a kind of bottom level shared lib.

I'll ponder it, but it's definitely not as intuitive for the consumer of the lib.  Right now when the dev activates the ILI9341 lib, all the specific settings for ILI9341 are right in that one lib.

100% agreed that `../Kconfig.spi` is a separate issue.  To preserve the purity and readability of the original issue, let's pause this conversation and at a future time I'll open a new issue and reference this one.  Thank you for everything!