espressif/idf-component-manager #83: ERROR: Cannot establish a connection to the component registry. Are you connected to the internet?

fariscooool, 2025-02-14:
### The Component Manager version

2.0.4

### ESP-IDF Version

v5.3.2

### python Version

3.13.2

### Operating System

windows11

### Browser (for https://components.espressif.com  Issues)

edge

### Description

The issue arises whenever there is a need to fetch online components：
ERROR: Cannot establish a connection to the component registry. Are you connected to the internet?
I am not entirely sure whether this issue is related to my repeated installation of the ESP-IDF environment or if it is connected to my computer's environment configuration. This uncertainty arises because I did not encounter the same problem when I installed the identical IDF on another computer.

### To Reproduce

```cpp
I have installed the ESP-IDF plugin on VSCode and set up the relevant history and toolchain using the offline ESP-IDF installation tool. However, when I attempt to compile the corresponding project that includes Dependencies which need to be installed online, I encounter the following issue:

-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.47.1.windows.2")
-- The C compiler identification is GNU 13.2.0
-- The CXX compiler identification is GNU 13.2.0
-- The ASM compiler identification is GNU
-- Found assembler: D:/UserDev/UserSDK/Espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/xtensa-esp32-elf-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: D:/UserDev/UserSDK/Espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/xtensa-esp32-elf-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: D:/UserDev/UserSDK/Espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/xtensa-esp32-elf-g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- git rev-parse returned 'fatal: not a git repository (or any of the parent directories): .git'
-- Could not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32
Dependencies lock doesn't exist, solving dependencies.
CMake Error at D:/UserDev/UserSDK/Espressif/frameworks/esp-idf-v5.3.2/tools/cmake/build.cmake:552 (message):
  ERROR: Cannot establish a connection to the component registry.  Are you
  connected to the internet?

  URL:
  https://components-file.espressif.com/components/espressif/led_strip.json

Call Stack (most recent call first):
  D:/UserDev/UserSDK/Espressif/frameworks/esp-idf-v5.3.2/tools/cmake/project.cmake:710 (idf_build_process)
  CMakeLists.txt:6 (project)

-- Configuring incomplete, errors occurred!

Alternatively, I can use idf.py to download the corresponding components：

D:\UserDev\UserSDK\Espressif\frameworks\esp-idf-v5.3.2>idf.py add-dependency "espressif/mdns^1.6.0"
Executing action: add-dependency
ERROR: Cannot establish a connection to the component registry. Are you connected to the internet?
URL: https://components-file.espressif.com/components/espressif/mdns.json
```

### Expected behaviour

I hope to resolve the issue of fetching components over the network.

### I have checked existing issues and online Documentation

- [x] I confirm I have checked existing issues and online Documentation.

kumekay, 2025-02-15:
Hello @fariscooool 

Some of our customers experienced similar issues in the past,

In most cases, the root cause was the configuration of the firewall, either on the computer directly or on the corporate network level.

Can you access https://components-file.espressif.com/components/espressif/led_strip.json from your browser? 

If this works from the browser, then in some cases, the firewall can be configured to allow requests from a browser, but not from a CLI application.

Could you please check that these domains are not blocked by the firewall? And if it's the case, allow access to the required domains?

The ESP Component Registry API is hosted on AWS multiple virtual machines in the Asia Pacific (Hong Kong) Region, and the storage of files is exposed through AWS CloudFront.  It only tries to access a couple of domains:
 
The main registry:
https://components.espressif.com/ - API server and web interface
https://components-file.espressif.com - File storage CDN

Test deployment of the registry (can be used for development of new components, not needed for use of regular components)
https://components-staging.espressif.com/ - API
https://d30mc2df6nu4o1.cloudfront.net - File storage CDN

You can check it with one CLI command:

Open ESP-IDF terminal in VS Code and run: 

```sh
python -c "import requests; response = requests.get('https://components-file.espressif.com/components/example/cmp.json'); print(response.text)"
```

On Windows in some configuration, python may use a system set of TLS certificates, instead of the supplied one.

If the previous command fails, try to access something else through python, like:

```sh
python -c "import requests; response = requests.get('https://jsonplaceholder.typicode.com/posts/1'); print(response.text)"    
```

fariscooool, 2025-02-16:
hi @kumekay ,thank you for your replay.

I am unable to access [https://components-file.espressif.com/components/espressif/led_strip.json] from my browser.

I tried using the instructions you gave and got the following results:

PS D:\Espressif\frameworks\esp-idf-v5.4> python -c "import requests; response = requests.get('https://components-file.espressif.com/components/example/cmp.json'); print(response.text)"
Traceback (most recent call last):
  File "D:\Espressif\python_env\idf5.4_py3.11_env\Lib\site-packages\urllib3\connection.py", line 174, in _new_conn
    conn = connection.create_connection(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\Espressif\python_env\idf5.4_py3.11_env\Lib\site-packages\urllib3\util\connection.py", line 95, in create_connection
    raise err
  File "D:\Espressif\python_env\idf5.4_py3.11_env\Lib\site-packages\urllib3\util\connection.py", line 85, in create_connection
    sock.connect(sa)
TimeoutError: [WinError 10060] 由于连接方在一段时间后没有正确答复或连接的主机没有反应，连接尝试失败。

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "D:\Espressif\python_env\idf5.4_py3.11_env\Lib\site-packages\urllib3\connectionpool.py", line 716, in urlopen
    httplib_response = self._make_request(
                       ^^^^^^^^^^^^^^^^^^^
  File "D:\Espressif\python_env\idf5.4_py3.11_env\Lib\site-packages\urllib3\connectionpool.py", line 404, in _make_request
    self._validate_conn(conn)
  File "D:\Espressif\python_env\idf5.4_py3.11_env\Lib\site-packages\urllib3\connectionpool.py", line 1061, in _validate_conn
    conn.connect()
  File "D:\Espressif\python_env\idf5.4_py3.11_env\Lib\site-packages\urllib3\connection.py", line 363, in connect
    self.sock = conn = self._new_conn()
                       ^^^^^^^^^^^^^^^^
  File "D:\Espressif\python_env\idf5.4_py3.11_env\Lib\site-packages\urllib3\connection.py", line 179, in _new_conn
    raise ConnectTimeoutError(
urllib3.exceptions.ConnectTimeoutError: (<urllib3.connection.HTTPSConnection object at 0x000001FA8C4DBDD0>, 'Connection to components-file.espressif.com timed out. (connect timeout=None)')

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "D:\Espressif\python_env\idf5.4_py3.11_env\Lib\site-packages\requests\adapters.py", line 667, in send
    resp = conn.urlopen(
           ^^^^^^^^^^^^^
  File "D:\Espressif\python_env\idf5.4_py3.11_env\Lib\site-packages\urllib3\connectionpool.py", line 802, in urlopen
    retries = retries.increment(
              ^^^^^^^^^^^^^^^^^^
  File "D:\Espressif\python_env\idf5.4_py3.11_env\Lib\site-packages\urllib3\util\retry.py", line 594, in increment
    raise MaxRetryError(_pool, url, error or ResponseError(cause))
urllib3.exceptions.MaxRetryError: HTTPSConnectionPool(host='components-file.espressif.com', port=443): Max retries exceeded with url: /components/example/cmp.json (Caused by ConnectTimeoutError(<urllib3.connection.HTTPSConnection object at 0x000001FA8C4DBDD0>, 'Connection to components-file.espressif.com timed out. (connect timeout=None)'))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "D:\Espressif\python_env\idf5.4_py3.11_env\Lib\site-packages\requests\api.py", line 73, in get
    return request("get", url, params=params, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\Espressif\python_env\idf5.4_py3.11_env\Lib\site-packages\requests\api.py", line 59, in request
    return session.request(method=method, url=url, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\Espressif\python_env\idf5.4_py3.11_env\Lib\site-packages\requests\sessions.py", line 589, in request
    resp = self.send(prep, **send_kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\Espressif\python_env\idf5.4_py3.11_env\Lib\site-packages\requests\sessions.py", line 703, in send
    r = adapter.send(request, **kwargs)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\Espressif\python_env\idf5.4_py3.11_env\Lib\site-packages\requests\adapters.py", line 688, in send
    raise ConnectTimeout(e, request=request)
requests.exceptions.ConnectTimeout: HTTPSConnectionPool(host='components-file.espressif.com', port=443): Max retries exceeded with url: /components/example/cmp.json (Caused by ConnectTimeoutError(<urllib3.connection.HTTPSConnection object at 0x000001FA8C4DBDD0>, 'Connection to components-file.espressif.com timed out. (connect timeout=None)'))
PS D:\Espressif\frameworks\esp-idf-v5.4> python -c "import requests; response = requests.get('https://jsonplaceholder.typicode.com/posts/1'); print(response.text)"
{
  "userId": 1,
  "id": 1,
  "title": "sunt aut facere repellat provident occaecati excepturi optio reprehenderit",
  "body": "quia et suscipit\nsuscipit recusandae consequuntur expedita et cum\nreprehenderit molestiae ut ut quas totam\nnostrum rerum est autem sunt rem eveniet architecto"
}

Is this a problem with the firewall?

fariscooool, 2025-02-16:
Hello @kumekay , the issue has been resolved, but I'm not sure what the exact cause was. After I turned off the firewall, I still couldn't successfully ping components-file.espressif.com. Then, I checked the network adapter settings on my other computer to see what the differences were between the two. I noticed that on this computer, the "Internet Protocol Version 6 (TCP/IPv6)" option was not checked. After I enabled this option, I was able to ping successfully... However, I don't understand why this worked. It seems that this issue wasn't related to the firewall after all...

kumekay, 2025-02-18:
@fariscooool Thank you for the kind response. 
I'm not sure whether IPv6 check-box can affect the functionality because the storage is accessible through IPv4. 

Just to double-check, are you in mainland China? If so, we have a local mirror there: 

you can try it with:

```
$env:IDF_COMPONENT_STORAGE_URL="https://components-file.espressif.cn"
```

Or set the environment variable permanently.

kumekay, 2025-02-24:
@fariscooool I'm closing the issue now, if you still have an issue feel free to reopen

closed 2025-02-24 by kumekay

fariscooool, 2025-02-26:
Yes, I am from Chinese Mainland. Thank you for your reply and your help. At present, there is no problem in using IDF.

riuson, 2025-03-25:
**Same problem.**

> Open ESP-IDF terminal in VS Code and run:

> python -c "import requests; response = requests.get('https://components-file.espressif.com/components/example/cmp.json'); print(response.text)"

```
...
  File "ssl.py", line 517, in wrap_socket
  File "ssl.py", line 1075, in _create
  File "ssl.py", line 1346, in do_handshake
ssl.SSLCertVerificationError: [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:992)

During handling of the above exception, another exception occurred:
...
```

Specified file is successfully opened in browser.

This caused by using of **self-signed certificates** in company.

**Solution**

(OS WIndows)
Open manager of certificates (Start- Run - certmgr.msc).
Find in tree a "Trusted Root Certification Authrotities". Find root certificate from your company.
Right click on certificate and export it into x.509 .CER base-64 encoded file.
Open Visual Studio Code user preferences.
Find option "Custom Extra Vars" from Idf.
Add `REQUESTS_CA_BUNDLE = <path to cer file>` (path in format `c:/directory/company.cer` )
Save settings
Rebuild.

riuson, 2025-03-27:
@kumekay , yes, development version is working ok.