espressif/idf-component-manager #84: fails to run idf-component-manager with pydantic-2.11.0a2

stkw0, 2025-03-06:
### The Component Manager version

2.1.1

### ESP-IDF Version

5.4

### python Version

3.12

### Operating System

Gentoo

### Description

The latest version of pydantic-core seems to break with idf-component-manager. May or may not be related with idf-component-manager. The version I was using of pydantic is in alpha state. Since I lack the knowledge to know if it's a pydnatic bug or a idf-component-manager one I thought at least would be good to report it here. 

The output running idf.py build:
```
Executing action: update-dependencies
Executing action: reconfigure
Running cmake in directory /tmp/iperf/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=/usr/lib/python-exec/python3.12/python -DESP_PLATFORM=1 -DCCACHE_ENABLE=0 /tmp/iperf"...
-- IDF_TARGET is not set, guessed 'esp32' from sdkconfig '/tmp/iperf/sdkconfig'
-- Found Git: /usr/bin/git (found version "2.48.1")
-- Component directory /usr/share/esp-idf/components/asio does not contain a CMakeLists.txt file. No component will be added
-- Component directory /usr/share/esp-idf/components/tinyusb does not contain a CMakeLists.txt file. No component will be added
-- git rev-parse returned 'fatal: not a git repository (or any of the parent directories): .git'
fatal: not a git repository (or any of the parent directories): .git
-- The C compiler identification is GNU 14.2.0
-- The CXX compiler identification is GNU 14.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /opt/bin/xtensa-esp32-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /opt/bin/xtensa-esp32-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /opt/bin/xtensa-esp32-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- git rev-parse returned 'fatal: not a git repository (or any of the parent directories): .git'
-- Could not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32
NOTICE: Dependencies lock doesn't exist, solving dependencies.
NOTICE: Skipping optional dependency: espressif/esp-extconn
NOTICE: Skipping optional dependency: espressif/esp-extconn
NOTICE: Skipping optional dependency: espressif/esp-extconn
NOTICE: Using component placed at /usr/share/esp-idf/examples/system/console/advanced/components/cmd_system for dependency "cmd_system", specified in /tmp/iperf/main/idf_component.yml
........NOTICE: Skipping optional dependency: espressif/esp-extconn
CMake Error at /usr/share/esp-idf/tools/cmake/build.cmake:610 (message):
  Traceback (most recent call last):

    File "<frozen runpy>", line 198, in _run_module_as_main
    File "<frozen runpy>", line 88, in _run_code
    File "/usr/lib/python3.12/site-packages/idf_component_manager/prepare_components/__main__.py", line 6, in <module>
      main()
    File "/usr/lib/python3.12/site-packages/idf_component_manager/prepare_components/prepare.py", line 130, in main
      args.func(args)
    File "/usr/lib/python3.12/site-packages/idf_component_manager/prepare_components/prepare.py", line 28, in prepare_dep_dirs
      ).prepare_dep_dirs(
        ^^^^^^^^^^^^^^^^^
    File "/usr/lib/python3.12/site-packages/idf_component_manager/core.py", line 99, in wrapper
      return func(self, *args, **kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^
    File "/usr/lib/python3.12/site-packages/idf_component_manager/core.py", line 717, in prepare_dep_dirs
      downloaded_components = download_project_dependencies(
                              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    File "/usr/lib/python3.12/site-packages/idf_component_manager/dependencies.py", line 373, in download_project_dependencies
      lock_manager.dump(solution)
    File "/usr/lib/python3.12/site-packages/idf_component_tools/lock/manager.py", line 66, in dump
      lock = LockFile(**solution.model_dump())
                        ^^^^^^^^^^^^^^^^^^^^^
    File "/usr/lib/python3.12/site-packages/idf_component_tools/utils.py", line 220, in model_dump
      return super().model_dump(
             ^^^^^^^^^^^^^^^^^^^
    File "/usr/lib/python3.12/site-packages/pydantic/main.py", line 435, in model_dump
      return self.__pydantic_serializer__.to_python(
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

  pydantic_core._pydantic_core.PydanticSerializationError: Error calling
  function `serialize_model`: KeyError: 'name'

Call Stack (most recent call first):
  /usr/share/esp-idf/tools/cmake/project.cmake:710 (idf_build_process)
  CMakeLists.txt:8 (project)

-- Configuring incomplete, errors occurred!
```

### To Reproduce

```cpp
It should be trivial to reproduce. Installing the latest pydantic release and building the official ethernet/iperf example should trigger the issue.
```

### Expected behaviour

Builds the firmware

### Additional info

Downgrading to pydantic-core 2.10.6 solves the issue.

### I have checked existing issues and online Documentation

- [x] I confirm I have checked existing issues and online Documentation.

kumekay, 2025-03-07:
Hello @stkw0,
thank you for opening the issue.  
However, we only support installations of ESP-IDF made with the officially provided tools (install.sh script in ESP-IDF repo, IDE extensions, EIM), none of these tools install pre-release versions of dependencies. What was the way you used to set up your environment?

But regarding the issue. I can confirm that the problem is reproducible with pydantic-2.11.0a2 
However, there is a new version released today: https://pypi.org/project/pydantic/2.11.0b1/ And it doesn't have this bug, so I would close this issue. You are welcome to reopen if you think something else can be done from the component manager side.

closed 2025-03-07 by kumekay

stkw0, 2025-03-07:
Thanks, so it was really a pydantic bug. 

I installed it from the distribution package manager and since I maintain also the packages from my distribution I use the latest available versions whenever possible to catch potential bugs before it hits other users.

kumekay, 2025-03-07:
@stkw0 Thank you for your efforts catching the bugs!