espressif/idf-component-manager #85: build fail, cannot find "driver/gpio.h" (IDFGH-9411)

edword01, 2025-03-12:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

IDF5.0

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

build success.

### What is the actual behavior?

fatal error: driver/gpio.h: No such file or directory

### Steps to reproduce.

1. Step
    idf.py command to create new project, and copy the gpio example code.
2. Step
   build success,  then i use the idf component manager to add led strip component, and build success also.
3. Step
  then try to modify the component in "managed_components"，add one line to test, just :  "#define XXXX  1"
4. Step
  then idf.py reconfigure fail for sure,  then, using "mv" cmd to move the component to "components" dir in PROJECT DIR,
  so, build always erro now, and occurs this error.

hfudev, 2025-03-13:
Hi @edword01 

Could you check if the issue still persists? Alternatively, could you share the projects you're building so I can try reproducing it locally?

Indastri, 2025-05-19:
Since there was no response from the user - closing this issue. If the issue persists please reopen this or create a new issue.

Cheers!

closed 2025-05-19 by Indastri